kr0ner/OneESP32ToRuleThemAll #2: Error compiling/installing via ESPHome

sg-media-dev, 2024-03-15:
With reference to the discussion on https://community.home-assistant.io/t/configured-my-esphome-with-mcp2515-can-bus-for-stiebel-eltron-heating-pump/366053/245

I have included the mapper.cpp in the main YAML file and then get the following error:

`Linking .pioenvs/thz404/firmware.elf
/config/.esphome/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pioenvs/thz404/src/main.cpp.o:(.literal._Z17processCanMessageRKSt6vectorIhSaIhEE+0x10): undefined reference to 'GetValueByType(unsigned short, Type)'
/config/.esphome/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pioenvs/thz404/src/main.cpp.o: in function 'processCanMessage(std::vector<unsigned char, std::allocator<unsigned char> > const&)':
/config/.esphome/build/thz404/src/communication.h:80: undefined reference to 'GetValueByType(unsigned short, Type)'
collect2: error: ld returned 1 exit status
*** [.pioenvs/thz404/firmware.elf] Error 1`

kr0ner, 2024-03-15:
Ahoi @sg-media-dev. Looks like you have an old version of the communication.h ... you are missing this commit https://github.com/kr0ner/OneESP32ToRuleThemAll/commit/4518df79287e0e739f4224eea7ded1affa9da1c4 and this specific line https://github.com/kr0ner/OneESP32ToRuleThemAll/commit/4518df79287e0e739f4224eea7ded1affa9da1c4#diff-a9feca18e4e5e1d64cdb69f4059440139f417a331be7dcea63397e454f16c1b1R80. Pull the latest master and you should be fine ;)

sg-media-dev, 2024-03-15:
That looks great! Thanks :-)

closed 2024-03-15 by sg-media-dev