kr0ner/OneESP32ToRuleThemAll #3: Change operating mode

sg-media-dev, 2024-03-20:
Switching the operating mode does not work with my THZ404. The correct operating mode is read out and displayed in the selection field. If I change the operating mode via the heat pump, the new mode is also displayed after a short time. If I change the operating mode via ESPHome, it is not changed on the heat pump and after a short time the original operating mode is displayed again in the selection field. 

Any ideas for troubleshooting? Special Thanks :-)

kr0ner, 2024-03-20:
That means the send command fails. Since it is set to optimistic, it will assume that setting the new value worked. After some time the WP sends the currently set value again and this will be reflected in HA ;)
I will check later, if it behaves correctly on my THZ 504 and provide a fix.

kr0ner, 2024-03-20:
At the time when on_value is being called, the state has already been set to whatever is passed via x to the lambda. So checking for inequality here prevented sending the message to the WP.

sg-media-dev, 2024-03-21:
After the change, the "Programmschalter" floods my CAN bus

https://github.com/kr0ner/OneESP32ToRuleThemAll/assets/2487085/ad2cf86f-2c46-421a-a482-6fc652b9a307

...
`07:15:27 | [I] | [Communication:080] | Message received: Read/Write ID 0xd0 0x3c for property 0x0112 with raw value: 2816
07:15:27 | [I] | [CallbackHandler:024] | Callback not found for Kessel 0x0112
07:15:27 | [D] | [select:015] | 'PROGRAMMSCHALTER': Sending state Automatik (index 2)
07:15:27 | [D] | [select:015] | 'PROGRAMMSCHALTER': Sending state Automatik (index 2)
07:15:27 | [D] | [canbus:035] | send standard id=0x6a2 rtr=FALSE size=7
07:15:27 | [D] | [canbus:072] | received can message (#1) std can_id=0x180 size=7
07:15:27 | [I] | [Communication:080] | Message received: Read/Write ID 0xd0 0x3c for property 0x0112 with raw value: 2816
07:15:27 | [I] | [CallbackHandler:024] | Callback not found for Kessel 0x0112
07:15:27 | [D] | [select:015] | 'PROGRAMMSCHALTER': Sending state Automatik (index 2)
07:15:27 | [D] | [select:015] | 'PROGRAMMSCHALTER': Sending state Automatik (index 2)
07:15:27 | [D] | [canbus:035] | send standard id=0x6a2 rtr=FALSE size=7
07:15:27 | [D] | [canbus:072] | received can message (#1) std can_id=0x180 size=7
07:15:27 | [I] | [Communication:080] | Message received: Read/Write ID 0xd0 0x3c for property 0x0112 with raw value: 2816
07:15:27 | [I] | [CallbackHandler:024] | Callback not found for Kessel 0x0112
...`

I hope I'm not being annoying :-)

kr0ner, 2024-03-21:
Oh boi ... I was just checking, if it is possible to set the values from both sides :sweat_smile: 
Fix is on the way.

kr0ner, 2024-03-21:
so the strategy should be:
id(<sensor>).publish_state(<value>) in callback
and
set_action: lambda: ...
in the component :+1:

kr0ner, 2024-03-21:
@sg-media-dev no worries ... I'm really happy that someone tests this. Report as many issues as you like :)

closed 2024-03-21 by kr0ner

sg-media-dev, 2024-03-21:
Great, now it looks good :-)