kr0ner/OneESP32ToRuleThemAll #6: branch for Tecalor THZ 5.5

croessi, 2024-05-06:
Hi  @kr0ner,

I like your coding style very well! I have already some changes for the THZ 5.5 and I added a Feature to decode the Messages on the CAN to make them better readable (see below). The latter is helped a lot to decode the THZ 5.5 messages. How would you like to review the changes? Shall I fork or create a branch?

[image]

kr0ner, 2024-05-06:
Hi @croessi. Thanks for offering your help. I really appreciate it 😎 
I'd prefer fork + PR for the beginning 😉 
Do you already have a proposal how we could manage different variants in one repo?
The verbose CAN messages look good 👍 you want to create a PR for that already?

croessi, 2024-05-09:
Hi @kr0ner, I started a pull request with the verbose output and one on top with already some register changes for THZ 5.5 eco. Most of it should be not conflicting, but some ventilation registers seems off by +1. Can you check if this is the same for 504, or if we have a conflict here? In the latter case maybe we need to have two property.h versions in the repo, but could keep for the rest the same codebase.

kr0ner, 2024-05-09:
Hi @croessi thanks for the PR 👍 Already had a look at it and left some comments. Let me know what you think 😉

sg-media-dev, 2024-05-12:
> Hi @kr0ner, I started a pull request with the verbose output and one on top with already some register changes for THZ 5.5 eco. Most of it should be not conflicting, but some ventilation registers [...]

Hey guys, I have a THZ 404 and I can confirm @croessi observation. Here are some examples:

`kZULUFT_SOLL = 0x0596,
kZULUFT_IST = 0x0597,
    kABLUFT_SOLL = 0x0598,
    kABLUFT_IST = 0x0599,
    kFORTLUFT_SOLL = 0x059a,
    kFORTLUFT_IST = 0x059b,
    kVERFLUESSIGER_TEMP = 0x059c,
    kLAUFZEIT_VERDICHTER_HEIZEN = 0x05a4,
    kLAUFZEIT_VERDICHTER_WW = 0x05a6,
    kLAUFZEIT_NHZ_HEIZEN = 0x05a8,
    kLAUFZEIT_NHZ_WW = 0x05a7,`

If I can help, please let me know :-)

kr0ner, 2024-05-12:
> Hey guys, I have a THZ 404 and I can confirm @croessi observation. Here are some examples:
>
> `kZULUFT_SOLL = 0x0596, kZULUFT_IST = 0x0597, kABLUFT_SOLL = 0x0598, kABLUFT_IST = 0x0599, kFORTLUFT_SOLL [...]

Thanks for double checking this 👍 
@croessi : could you maybe create a second PR containing only those changes, so that we can merge this already?

croessi, 2024-05-13:
@sg-media-dev Thanks for checking! So are we sure it will be the same for 504? Then we do not need to differentiate. I have found already some more of these, and it would be way easier if we do not need to implement variations in `properties.h`.
May I ask you to doublechek, as soon as I have a larger list?

kr0ner, 2024-09-26:
obsolete

closed 2024-09-26 by kr0ner