kr0ner/OneESP32ToRuleThemAll #8: PUMPENDREHZAHL_HEIZEN changes are not applied

sg-media-dev, 2024-05-12:
Hey kr0ner, the project is developing nicely - it's great. I have implemented the changes from #5 . The value RAUMEINFLUSS is displayed correctly and can be changed successfully. The correct value is also displayed for PUMPENDREHZAHL_HEIZEN, but the changes are not applied in the WP (THZ404). Do you have any ideas?

kr0ner, 2024-05-12:
Hi @sg-media-dev. Could be, that the target is wrong and has to be adapted for the 404 case. You could try for example 

> PUMPENDREHZAHL_HEIZEN:          !include { file: thz504_number.yaml, vars: { property: "PUMPENDREHZAHL_HEIZEN"      , step:   "0.1", target: "HK1" }}

Can you compile with verbose logging and look for the messages that are being sent by the WP, when you adjust the settings there?
Weird, that it only works for one of them and for the other not 🤷

sg-media-dev, 2024-05-14:
I no longer get a value with HK1. Well, that's not so important, but still strange ;)

closed 2024-05-14 by sg-media-dev