kr0ner/OneESP32ToRuleThemAll #12: ESP32 internal CAN Controller

kr0ner, 2024-05-17:
hey there, 

I am using the ESP32 internal CAN Controller (https://esphome.io/components/canbus.html?highlight=can#esp32-can-component). 

could you also "elegantly integrate" a variant for this?

btw. I have a Tecalor TTC07cool heatpump and still have some issues with the existing implementations. This is why I'm following your progress here with your "clean c++ code approach" with great enthusiasm :)

_Originally posted by @mkaiser in https://github.com/kr0ner/OneESP32ToRuleThemAll/issues/11#issuecomment-2117840238_

kr0ner, 2024-05-17:
Hi @mkaiser

shouldn't it be as easy as this 😉 ESP needs to be compatible though [docs](https://esphome.io/components/canbus.html?highlight=can#esp32-can-component)
```
 #########################################
+# The ESP32 has an integrated CAN controller and therefore doesn’t need an
+# external controller necessarily. You only need to specify the RX and TX pins.
+# Any GPIO will work.
 canbus:
-  - platform: mcp2515
+  - platform: esp32_can
     id: my_mcp2515
-    spi_id: McpSpi
-    cs_pin: GPIO15
     can_id: 680
     use_extended_id: false
     bit_rate: 20kbps
```

With regards to the missing features for Tecalor TTC07cool, don't hesitate to create issues, preferably with debug logs, so that we can try to fix them.

mkaiser, 2024-05-25:
I answered in the previous thread regarding the configuration. 

When I have some spare time I will check on the TTC07c and give some feedback. Tried this with the unmaintained integrations which are the base of this one, but did not have time to finish (and I really disliked the coding style).

kr0ner, 2024-09-26:
marked as stale

closed 2024-09-26 by kr0ner