kr0ner/OneESP32ToRuleThemAll #15: HA noob needs help

PhagoZ, 2024-07-11:
Hello there, really nice project and it lloks promising for my THZ 5.5 Eco.

I already tried 2 different other project (one iobroker and the bullit186 project) on both I just had a couple of values to read and nothing to control.

To be hornest, I´m a bit confused how to handle all the different yamls.

is the esp32-poe yaml the main code for the esp and all the others are packages?

is it a problem to use a esp32-s3 instead of poe, what has to be changed?

kr0ner, 2024-07-11:
Thanks :) In my current setup I can do almost everything the heat pump display can do. Cooling, heating, setting temps, ventilation, sensor readings, adjusting config that requires password. There is no limit ... Some stuff might just not be implemented

> is the esp32-poe yaml the main code for the esp and all the others are packages?

Yes, this ist the main config. The other yamls are there to structure it better and to be able to exclude stuff more easily. Just comment out the stuff you dont need.
You can use any ESP you want. Even a 8266 should work nowadays. You Just need to figure Out which pins to use. Should not be a big deal.

PhagoZ, 2024-07-11:
Wow .. lightning fast answer ^^

thanks for that.. while compiling I getting the error "Component time.sntp requires component network." From THZ504.yaml Line 205.
I looked into the wiki and didn´t find some obvious error.. any idea?

kr0ner, 2024-07-11:
You need to add WiFi and delete the Ethernet part. https://esphome.io/components/wifi.html

PhagoZ, 2024-07-11:
Okay yeah.. the wifi it was.

of cause I try to compile the esp32.yaml.

Right now I have errors from sml-reader.h

:(

In file included from src/main.cpp:301:
src/sml_reader.h: In member function 'void Readings::publishHA() const':
src/sml_reader.h:66:12: error: 'Instantaneous_Power' was not declared in this scope
         id(Instantaneous_Power).publish_state(power);
            ^~~~~~~~~~~~~~~~~~~
src/sml_reader.h:67:12: error: 'Instantaneous_Power_L1' was not declared in this scope
         id(Instantaneous_Power_L1).publish_state(power_l1);
            ^~~~~~~~~~~~~~~~~~~~~~
src/sml_reader.h:68:12: error: 'Instantaneous_Power_L2' was not declared in this scope
         id(Instantaneous_Power_L2).publish_state(power_l2);
            ^~~~~~~~~~~~~~~~~~~~~~
src/sml_reader.h:69:12: error: 'Instantaneous_Power_L3' was not declared in this scope
         id(Instantaneous_Power_L3).publish_state(power_l3);
            ^~~~~~~~~~~~~~~~~~~~~~
src/sml_reader.h:70:12: error: 'Instantaneous_Voltage_L1' was not declared in this scope
         id(Instantaneous_Voltage_L1).publish_state(voltage_l1);
            ^~~~~~~~~~~~~~~~~~~~~~~~
src/sml_reader.h:71:12: error: 'Instantaneous_Voltage_L2' was not declared in this scope
         id(Instantaneous_Voltage_L2).publish_state(voltage_l2);
            ^~~~~~~~~~~~~~~~~~~~~~~~
src/sml_reader.h:72:12: error: 'Instantaneous_Voltage_L3' was not declared in this scope
         id(Instantaneous_Voltage_L3).publish_state(voltage_l3);
            ^~~~~~~~~~~~~~~~~~~~~~~~
src/sml_reader.h:73:12: error: 'Instantaneous_Current_L1' was not declared in this scope
         id(Instantaneous_Current_L1).publish_state(current_l1);
            ^~~~~~~~~~~~~~~~~~~~~~~~
Compiling .pioenvs/esphome-web-9d8be0/FrameworkArduino/IPAddress.cpp.o
src/sml_reader.h:74:12: error: 'Instantaneous_Current_L2' was not declared in this scope
         id(Instantaneous_Current_L2).publish_state(current_l2);
            ^~~~~~~~~~~~~~~~~~~~~~~~
Compiling .pioenvs/esphome-web-9d8be0/FrameworkArduino/IPv6Address.cpp.o
Compiling .pioenvs/esphome-web-9d8be0/FrameworkArduino/MD5Builder.cpp.o
src/sml_reader.h:75:12: error: 'Instantaneous_Current_L3' was not declared in this scope
         id(Instantaneous_Current_L3).publish_state(current_l3);
            ^~~~~~~~~~~~~~~~~~~~~~~~
src/sml_reader.h:76:12: error: 'Frequency' was not declared in this scope
         id(Frequency).publish_state(frequency);
            ^~~~~~~~~
src/sml_reader.h:76:12: note: suggested alternative: 'frequency'
         id(Frequency).publish_state(frequency);
            ^~~~~~~~~
            frequency
src/sml_reader.h:77:12: error: 'Total_incoming' was not declared in this scope
         id(Total_incoming).publish_state(total_incoming);
            ^~~~~~~~~~~~~~
src/sml_reader.h:77:12: note: suggested alternative: 'total_incoming'
         id(Total_incoming).publish_state(total_incoming);
            ^~~~~~~~~~~~~~
            total_incoming
src/sml_reader.h:78:12: error: 'Total_Outgoing' was not declared in this scope
         id(Total_Outgoing).publish_state(total_outgoing);
            ^~~~~~~~~~~~~~
src/sml_reader.h:78:12: note: suggested alternative: 'total_outgoing'
         id(Total_Outgoing).publish_state(total_outgoing);
            ^~~~~~~~~~~~~~
            total_outgoing
src/sml_reader.h: At global scope:
src/sml_reader.h:82:55: error: expected class-name before '{' token
 class SMLReader : public Component, public UARTDevice {
                                                       ^
src/sml_reader.h:229:28: error: expected ')' before '*' token
     SMLReader(UARTComponent* parent) : UARTDevice(parent) {}
[... 34 lines cut ...]
src/sml_reader.h:254:17: note: suggested alternative: 'Readings'
                 publishReadings();
                 ^~~~~~~~~~~~~~~
                 Readings
src/sml_reader.h: In instantiation of 'void SMLReader::waitForSequence(const T&, F) [with T = std::array<unsigned char, 8>; F = SMLReader::loop()::<lambda()>]':
src/sml_reader.h:239:18:   required from here
src/sml_reader.h:90:25: error: 'available' was not declared in this scope
         while (available()) {
                ~~~~~~~~~^~
src/sml_reader.h:90:25: note: suggested alternative: 'dacDisable'
         while (available()) {
                ~~~~~~~~~^~
                dacDisable
src/sml_reader.h: In instantiation of 'void SMLReader::waitForSequence(const T&, F) [with T = std::array<unsigned char, 5>; F = SMLReader::loop()::<lambda()>]':
src/sml_reader.h:242:95:   required from here
src/sml_reader.h:90:25: error: 'available' was not declared in this scope
src/sml_reader.h:90:25: note: suggested alternative: 'dacDisable'
         while (available()) {
                ~~~~~~~~~^~
                dacDisable
Compiling .pioenvs/esphome-web-9d8be0/FrameworkArduino/Print.cpp.o
Compiling .pioenvs/esphome-web-9d8be0/FrameworkArduino/Stream.cpp.o
Compiling .pioenvs/esphome-web-9d8be0/FrameworkArduino/StreamString.cpp.o
Compiling .pioenvs/esphome-web-9d8be0/FrameworkArduino/Tone.cpp.o
*** [.pioenvs/esphome-web-9d8be0/src/main.cpp.o] Error 1

PhagoZ, 2024-07-11:
Wuuhuuu ich can see so much stuff.. as soon as I will be at home, I will try to control some stuff.

one question more: is there any chance of seeing the live data of power consuption? Now I have these data, I could get these data via MQTT in EVCC project for pv power control

kr0ner, 2024-07-11:
You mean power consumption of the Heat pump? 🤔

PhagoZ, 2024-07-11:
yeah.. normaly there should be a measuring dive, because it can calculate the kWh used....

PhagoZ, 2024-07-11:
el_aufnahmeleistung_heiz_summe_kwh is such a thing.. leistung is power and kWh is energy.. so is it energy or "live" data power?

kr0ner, 2024-07-11:
Its overall energy consumption in kWh. Good enough for hourly Update in Energy Dashboard 

[image]

PhagoZ, 2024-07-11:
Okay so I changed in energy.yaml to very_fast and will see if I therfor recalculate the power calue in kW.
I think I will do this in iobroker.. it´s way easier to programm ^^

kr0ner, 2024-07-11:
When updating the value you have the  old and the new one and the Intervall ;) so Delta/Interval and done 😎

PhagoZ, 2024-07-11:
Yeah I know.. that´s my plan.. thanks man.. 
Now I just have to design and print a nice case.

Do I remember right, that te cooling function is possible to activate, with the ISG? so the ESP is doin it´s job it should work as well right?

kr0ner, 2024-07-11:
Yep, it sends the avg. room temp and humidity and can set the cooling params ... So no need to pay 800€ for an ISG 😎

PhagoZ, 2024-07-11:
So cool. You´re my man :)

My actual room temp is tellin me, that it is 10°C... don´t know why.
But anyway how is it possible to put a helper value as the the Raum_ist?

I´m calculating my avg_Raumtemp in iobroker, because all my NSPanels are working in there as well as thermostats.

kr0ner, 2024-07-11:
10° RaumIstTemp is most likely your RaumSollTemp. If you don't send the temp frequently enough, it gets set to the heating, cooling target temp.

PhagoZ, 2024-07-12:
@kr0ner Yeah no problem, I will write it down and add some hardware photos as well for my setup, as soon as it´s all done.

did you use the elster list for ID reference? I still have hope to add the rel power consumption data ^^ maybe there is something possible

kr0ner, 2024-07-12:
thanks :)

With regards to the power readings. If you can't find those values somewhere in the display, it is unlikely that they exist. You can check that and if you find sth. meaningful we can have a closer look ... try 7777 to get the most out of it

closed 2024-07-12 by kr0ner