kr0ner/OneESP32ToRuleThemAll #17: Kuehl_Raumsoll_tag not writable

PhagoZ, 2024-07-16:
Hey it´s me again..

I tried last week to use the cooling function, but it´s not working. I think I found why.
The Kuehl_Raumsoll_tag temperatur is at 210°C as stock, how can I add another "SollTemp" input, or alternativly copy the value from the heating value?

kr0ner, 2024-07-16:
I see where you are trying to go. You can just use the "Heating Day" and/or "Heating Night" climates. If you look at https://github.com/kr0ner/OneESP32ToRuleThemAll/blob/master/src/custom_climate.h#L110 together with https://github.com/kr0ner/OneESP32ToRuleThemAll/blob/master/yaml/thz504.yaml#L424 you can see that you need to set either 
Property::kRAUMSOLLTEMP_NACHT or Property::kKUEHL_RAUMSOLL_NACHT
Property::kRAUMSOLLTEMP_I or Property::kKUEHL_RAUMSOLL_TAG 
this depends on whether your heat pump is in summer mode (cooling active) or not
also Property::kKUEHLMODE has to be set to true to enable cooling

PhagoZ, 2024-07-16:
my cooling is enabled
[image]

the mode ("nur Lüfter" on the picture) is always changing back, even if I activate cooling hier as well
[image]

At least the Kuehl_Raumsoll_tag should be correclty displayed as 21.0, right?
no it is displayed as 210°C maybe this is the problem, why the cooling is always deactivated. ^^
I saw, that the value is changing when I set the temperature.. I changed my RausSoll to 22°C and the KuehlSoll changed to 220°C. So just a multiplier error somewhere.

kr0ner, 2024-07-16:
Are you running latest master? These changes should be there
https://github.com/kr0ner/OneESP32ToRuleThemAll/commit/6521a009d1d1923e87b524c75f6ba1e2c6552c8c#diff-aa67afd076546b54264d045a7d88dc2d071365cab4440c8a0cd8c08832d60759R102-R106
This has to affect the values the same way as on my end 
[image]
Are you sending room temp and humidity to the heat pump?

PhagoZ, 2024-07-16:
No looks like, I did not use this verison.
I pasted the new versions of the files you compared above. will see if it works.

I´m sending the room temperature with a helper yes.
I calculate the average temp in iobroker from all my NSPanels in the house.

The humidity is not sent, because I don´t measure it. I just have one aquara temp sensor with humidity which I use for temperatur calibrating of the NSpanels

PhagoZ, 2024-07-16:
is there any way to contact each other beside these threads?
Would like to discuss a topic with you

kr0ner, 2024-07-16:
> No looks like, I did not use this verison. I pasted the new versions of the files you compared above. will see if it works.
> 
> I´m sending the room temperature with a helper yes. I calculate the average temp in iobroker from all my NSPanels in the house.
> 
> The humidity is not sent, because I don´t measure it. I just have one aquara temp sensor with humidity which I use for temperatur calibrating of the NSpanels
> 
> Okay, now It´s done. he doesn´t try to "cool" down to 210°C :D

maybe just create a dummy entity sending 60 % for now to check if it works. FEK would send both values periodically

kr0ner, 2024-07-16:
> is there any way to contact each other beside these threads? Would like to discuss a topic with you

we can talk via discord somewhen in the evening if you like

PhagoZ, 2024-07-16:
Good idea, what's your discord Servers Name?

kr0ner, 2024-07-16:
https://discord.gg/ZMmVwdx9

I'll be there at around 9 PM CEST

kr0ner, 2024-07-17:
closing this as 5.5 eco does not support cooling
requested sensors have been added https://github.com/kr0ner/OneESP32ToRuleThemAll/commit/2a6821ca1f86de18c7e8cc02cd6367ecb2ce4ad3

closed 2024-07-17 by kr0ner