kr0ner/OneESP32ToRuleThemAll #18: Switch off LUEFT_STUFE_TAG automatically

Nebnenut, 2024-07-19:
I currently have the phenomenon that the LUEFT_STUFE_TAG switches off automatically as soon as the entry "no longer available" is in the log.

The same thing happens when the ESP controller is disconnected from the power and powered again.currently only detectable with the LUEFT_STUFE_TAG sensor.

kr0ner, 2024-07-19:
Hi, I have a suspicion: After boot we ask the heat pump for the value of LUEFT_STUFE_TAG ... but we are somehow reading the request and not the response. Requests always have 0 as a value. On reception of zero we turn the fan off ... which might be the second problem. I will try to prepare a fix. 

Side question: why is your esp so often not reachable? :thinking:

Nebnenut, 2024-07-19:
the problem with the device losing connection has been going on since this morning. If necessary, I'll swap the ESP for another one, of course everyone has a few ESP controllers on reserve at home :-D

kr0ner, 2024-07-19:
> of course everyone has a few ESP controllers on reserve at home

Yes :) you never know :D

hm ... so it might be caused by my changes from yesterday evening? When did you do the last update?
So it's definitely a bug in the code ... but we should also tackle the resets. If it is reproducible it would be nice to have the log output so that I can check what the last command was and then guess what could've been the problem.

Nebnenut, 2024-07-19:
Ok, the ESP has problems with the wifi chip or with the WiFi reception. The Fritzbox says cancels registrations. I'll swap it out.

https://pastebin.com/AEt0RfVd

kr0ner, 2024-07-19:
ok, first thing that I have overlooked 😅 
https://esphome.io/components/fan/#base-fan-configuration
> restore_mode (Optional): Control how the fan attempts to restore state on boot.
> ALWAYS_OFF (Default) - Always initialize the fan as OFF on bootup.
That explains a lot ... but this is only half the story ... once this is fixed, fan speeds behave weird ... still investigating

commit cfe2537a (2024-07-19): Fix broken fan control

After reboot the fans were turned off. This was happening due to
a missing config option:

restore_mode -
ALWAYS_OFF (Default) - Always initialize the fan as OFF on bootup.

Fixing this revealed another bug in the code and in ESPHOME.
A workaround is in place until the issue is fixed. Fan state and
fan speed have to be sent in two consecutive calls.

Fixes: #18
commit 5447dc0f (2024-07-22): Fix fan turn_on()

Turn_on() cannot be left blank. Doing so will not send a message,
if the fan was only turned on and no speed was set. First time
the fan is turned on, the speed will be set to 100%. All other
calls should reset the speed to the value that was set before
it had been turned off.

Fixes: #18
closed 2024-07-22 by kr0ner