kr0ner/OneESP32ToRuleThemAll #48: Running on ESP8266

do-gooder, 2024-11-09:
How can I run your project on ESP8266?

[logs_esp32-poe-technik_compile.txt](https://github.com/user-attachments/files/17689205/logs_esp32-poe-technik_compile.txt)

My config:
```
esphome:
  name: esp32-poe-technik
  friendly_name: ESP32-PoE Technik
  platformio_options:
    build_flags:
      - "-std=gnu++17"
    build_unflags:
      - "-std=gnu++11"
      - "-fno-rtti"

esp8266:
  board: nodemcuv2
  # framework:
    # type: arduino

# Enable logging
logger:
  level: INFO

# Enable Home Assistant API
api:
  encryption:
    key: "XXX"

ota:
  platform: esphome
  password: "XXX"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Heatingpump-Esp8266"
    password: "XXX"

time:
  - platform: homeassistant
    id: homeassistant_time

substitutions:
  interval_very_fast: 15s
  interval_fast: 30s
  interval_medium: 60s
  interval_slow: 5min
  interval_very_slow: 15min
  interval_once_in_a_while: 1h

  entity_room_temperature: "sensor.durchschnittstemperatur_haus_ohne_keller"
  entity_humidity: "sensor.durchschnitt_luftfeuchtigkeit_haus"

#########################################
#                                       #
#   SPI configuration                   #
#                                       #
#########################################
spi:
  id: McpSpi
  clk_pin: D5
  mosi_pin: D7
  miso_pin: D6

#########################################
#                                       #
#   CAN configuration                   #
#                                       #
#########################################
canbus:
  - id: !extend wp_can
    platform: mcp2515
    spi_id: McpSpi
    cs_pin: D2

#########################################
#                                       #
#   Buttons                             #
#                                       #
#########################################
button:
  - platform: restart
    name: "Restart ESP32-PoE Technik"

#########################################
#                                       #
#   Ethernet Konfiguration              #
#                                       #
#########################################
#ethernet:
#  type: LAN8720
#  mdc_pin: GPIO23
#  mdio_pin: GPIO18
#  clk_mode: GPIO0_OUT
#  phy_addr: 0
#  power_pin: GPIO12

#########################################
#                                       #
#   Packages                            #
#                                       #
#########################################
packages:
# energymeter: !include yaml/energymeter.yaml
# watermeter: !include yaml/watermeter.yaml
# sensors: !include yaml/sensors.yaml
# thz404: !include yaml/thz404.yaml
# thz504: !include yaml/thz504.yaml
# thz55eco: !include yaml/thz5_5_eco.yaml
  ttf07: !include yaml/ttf07.yaml
# kondensatpumpe: !include yaml/kondensatpumpe.yaml

#########################################
#                                       #
#   EVU Sperre                          #
#                                       #
#########################################
#switch:
#  - platform: gpio
#    name: "EVU Sperre"
#    id: evu_sperre
#    pin: GPIO5

```

kr0ner, 2024-11-09:
@do-gooder my bad ... forgot to include some headers. Should work now :)

do-gooder, 2024-11-10:
The compilation worked.

After flashing, the ESP8266 does not seem to work properly. As soon as I add the package “ttf07: !include yaml/ttf07.yaml”, I don't even see a log.

Is there a simple reason for this?

kr0ner, 2024-11-10:
Most likely it crashes. Try commenting out most of the sensors in common.yaml

My assumption is that the esp8266 cannot handle that many entities.

You can confirm the crash by connecting the esp to your PC and look at the uart logs. You might need to adapt the log settings in the top level yaml

mkaiser, 2024-11-10:
And keep in mind that the ttf07 stuff ist still under construction :)

I will make you a minimal example next Weekend with only one Sensor

do-gooder, 2024-11-11:
Ok, thank you.

Commenting out most sensors leads to a dependency problem. It does not seem to be easy.

kr0ner, 2024-11-11:
> Ok, thank you.
> 
> Commenting out most sensors leads to a dependency problem. It does not seem to be easy.

commenting out those should not be a problem:
```
VERDICHTER_STARTS:               !include { file: wp_generic_combined.yaml, vars: { sensor_name: "VERDICHTER_STARTS"                  , scaled_property: "VERDICHTER_STARTS_K"          , property: "VERDICHTER_STARTS"            , unit: ""   , accuracy_decimals: "0", scaler: "1000", icon: "mdi:counter" }}
  WAERMEERTRAG_2WE_WW_SUMME_MWH:   !include { file: wp_generic_combined.yaml, vars: { sensor_name: "WAERMEERTRAG_2WE_WW_SUMME_MWH"      , scaled_property: "WAERMEERTRAG_2WE_WW_SUM_KWH"  , property: "WAERMEERTRAG_2WE_WW_SUM_MWH"  , unit: "MWh", accuracy_decimals: "3", icon: "mdi:fire"  }}
  WAERMEERTRAG_2WE_HEIZ_SUMME_MWH: !include { file: wp_generic_combined.yaml, vars: { sensor_name: "WAERMEERTRAG_2WE_HEIZ_SUMME_MWH"    , scaled_property: "WAERMEERTRAG_2WE_HEIZ_SUM_KWH", property: "WAERMEERTRAG_2WE_HEIZ_SUM_MWH", unit: "MWh", accuracy_decimals: "3", icon: "mdi:fire"  }}
  WAERMEERTRAG_WW_SUMME_MWH:       !include { file: wp_generic_combined.yaml, vars: { sensor_name: "WAERMEERTRAG_WW_SUMME_MWH"          , scaled_property: "WAERMEERTRAG_WW_SUM_KWH"      , property: "WAERMEERTRAG_WW_SUM_MWH"      , unit: "MWh", accuracy_decimals: "3", icon: "mdi:fire"  }}
  WAERMEERTRAG_HEIZ_SUMME_MWH:     !include { file: wp_generic_combined.yaml, vars: { sensor_name: "WAERMEERTRAG_HEIZ_SUMME_MWH"        , scaled_property: "WAERMEERTRAG_HEIZ_SUM_KWH"    , property: "WAERMEERTRAG_HEIZ_SUM_MWH"    , unit: "MWh", accuracy_decimals: "3", icon: "mdi:fire"  }}

  SPEICHERSOLLTEMP:               !include { file: wp_temperature.yaml, vars: { property: "SPEICHERSOLLTEMP"          }}
  EINSTELL_SPEICHERSOLLTEMP:      !include { file: wp_temperature.yaml, vars: { property: "EINSTELL_SPEICHERSOLLTEMP" }}
  SPEICHERISTTEMP:                !include { file: wp_temperature.yaml, vars: { property: "SPEICHERISTTEMP"          , interval: $interval_medium }}
  AUSSENTEMP:                     !include { file: wp_temperature.yaml, vars: { property: "AUSSENTEMP"       }}
  SAMMLERISTTEMP:                 !include { file: wp_temperature.yaml, vars: { property: "SAMMLERISTTEMP"           , interval: $interval_medium }}
  VERDAMPFERTEMP:                 !include { file: wp_temperature.yaml, vars: { property: "VERDAMPFERTEMP"   }}
  RUECKLAUFISTTEMP:               !include { file: wp_temperature.yaml, vars: { property: "RUECKLAUFISTTEMP"         , interval: $interval_medium }}
  VORLAUFISTTEMP:                 !include { file: wp_temperature.yaml, vars: { property: "VORLAUFISTTEMP"           , interval: $interval_medium, target: "HK1" }}
  TAUPUNKT_HK1:                   !include { file: wp_temperature.yaml, vars: { property: "TAUPUNKT_HK1"             , target: "HK1" }}
  RAUMISTTEMP:                    !include { file: wp_temperature.yaml, vars: { property: "RAUMISTTEMP"              , target: "HK1" }}
  VERSTELLTE_RAUMSOLLTEMP:        !include { file: wp_temperature.yaml, vars: { property: "VERSTELLTE_RAUMSOLLTEMP"  , target: "HK1" }}
  RAUMSOLLTEMP_I:                 !include { file: wp_temperature.yaml, vars: { property: "RAUMSOLLTEMP_I"           , target: "HK1" }}
  VORLAUFSOLLTEMP:                !include { file: wp_temperature.yaml, vars: { property: "VORLAUFSOLLTEMP"          , target: "HK1" }}
  KUEHL_RAUMSOLL_TAG:             !include { file: wp_temperature.yaml, vars: { property: "KUEHL_RAUMSOLL_TAG"       , target: "HK1" }}

  FEHLERMELDUNG:                  !include { file: wp_generic.yaml, vars: { property: "FEHLERMELDUNG"            , icon: "mdi:alert-circle" }}
  ANZEIGE_NIEDERDRUCK:            !include { file: wp_generic.yaml, vars: { property: "ANZEIGE_NIEDERDRUCK"      , accuracy_decimals: "2" }}
  PUMPENZYKLEN_MIN_AUSSENT:       !include { file: wp_generic.yaml, vars: { property: "PUMPENZYKLEN_MIN_AUSSENT" }}
  FEUCHTE:                        !include { file: wp_generic.yaml, vars: { property: "FEUCHTE"                  , interval: $interval_very_slow, unit: "%", icon: "mdi:water-percent", target: "HK1", accuracy_decimals: "1" }}

  LEISTUNG_AUSLEGUNG_HEIZUNG:     !include { file: wp_number.yaml, vars: { property: "LEISTUNG_AUSLEGUNG_HEIZUNG" }}
  LEISTUNG_AUSLEGUNG_KUEHLEN:     !include { file: wp_number.yaml, vars: { property: "LEISTUNG_AUSLEGUNG_KUEHLEN" }}
  PUMPENDREHZAHL_HEIZEN:          !include { file: wp_number.yaml, vars: { property: "PUMPENDREHZAHL_HEIZEN"      , step: "0.1" }}
  RAUMEINFLUSS:                   !include { file: wp_number.yaml, vars: { property: "RAUMEINFLUSS"               , target: "HK1" }}
  HYSTERESE_WW:                   !include { file: wp_number.yaml, vars: { property: "HYSTERESE_WW"               , min: "2.0", max: "10.0", step: "0.1", unit: "°C" }}

  EL_AUFNAHMELEISTUNG_WW_SUMME_KWH:    !include { file: wp_daily_energy_combined.yaml, vars: { sensor_name: "EL_AUFNAHMELEISTUNG_WW_SUMME_KWH"   , property_wh: "EL_AUFNAHMELEISTUNG_WW_TAG_WH"  , property_kwh: "EL_AUFNAHMELEISTUNG_WW_TAG_KWH"   }}
  EL_AUFNAHMELEISTUNG_HEIZ_SUMME_KWH:  !include { file: wp_daily_energy_combined.yaml, vars: { sensor_name: "EL_AUFNAHMELEISTUNG_HEIZ_SUMME_KWH" , property_wh: "EL_AUFNAHMELEISTUNG_HEIZ_TAG_WH", property_kwh: "EL_AUFNAHMELEISTUNG_HEIZ_TAG_KWH" }}
[... cut ...]
  WAERMEERTRAG_WW_TAG_SUMME_KWH:       !include { file: wp_daily_energy_combined.yaml, vars: { sensor_name: "WAERMEERTRAG_WW_TAG_SUMME_KWH"      , property_wh: "WAERMEERTRAG_WW_TAG_WH"         , property_kwh: "WAERMEERTRAG_WW_TAG_KWH"          }}
  WAERMEERTRAG_HEIZ_TAG_SUMME_KWH:     !include { file: wp_daily_energy_combined.yaml, vars: { sensor_name: "WAERMEERTRAG_HEIZ_TAG_SUMME_KWH"    , property_wh: "WAERMEERTRAG_HEIZ_TAG_WH"       , property_kwh: "WAERMEERTRAG_HEIZ_TAG_KWH"        }}

  HEATPUMP_DATETIME: !include { file: wp_datetime.yaml }
```

But please consider to use an ESP32 ... they are cheap on amazon and save you a lot of time :wink:

kr0ner, 2024-11-13:
@do-gooder any update? Did it work?

closed 2024-12-22 by github-actions[bot]

kr0ner, 2025-03-27:
@do-gooder this change might reduce the code to a level where it can run on an esp8266  https://github.com/kr0ner/OneESP32ToRuleThemAll/pull/154

commit ba7fa516 (2024-11-09): Fix compilation for ESP8266

closes: #48