kr0ner/OneESP32ToRuleThemAll #49: Using config for Tecalor 5.5. eco

skycryer, 2024-11-11:
So i adapted the config for ap,wifi, encryption and other, think I added all other stuff beside the watermetedr and other thing.

On compiling I get a lot off error like this 
```

src/property.h:183:5: note: in expansion of macro 'PROPERTY'
     PROPERTY(BETRIEBS_STATUS_2, 0xc356);
     ^~~~~~~~
src/property.h:29:47: note:   initializing argument 1 of 'constexpr detail::Property::Property(int, uint16_t)'
     constexpr Property(const std::string_view _name, const std::uint16_t _id) : name(_name), id(_id) {}
                        ~~~~~~~~~~~~~~~~~~~~~~~^~~~~
src/property.h:56:34: warning: inline variables are only available with -std=c++17 or -std=gnu++17
     static inline detail::Mapper k##NAME##_MAPPING {                        \
                                  ^
src/property.h:183:5: note: in expansion of macro 'PROPERTY'
     PROPERTY(BETRIEBS_STATUS_2, 0xc356);
     ^~~~~~~~
src/property.h: In copy constructor 'Property::Property(const Property&)':
src/property.h:61:54: error: 'const struct Property' has no member named 'name'
     Property(const Property& p) : detail::Property{p.name, p.id, p.type} {}
                                                      ^~~~
src/property.h:61:72: error: no matching function for call to 'detail::Property::Property(<brace-enclosed initializer list>)'
     Property(const Property& p) : detail::Property{p.name, p.id, p.type} {}
                                                                        ^
src/property.h:29:15: note: candidate: 'constexpr detail::Property::Property(int, uint16_t)'
     constexpr Property(const std::string_view _name, const std::uint16_t _id) : name(_name), id(_id) {}
               ^~~~~~~~
src/property.h:29:15: note:   candidate expects 2 arguments, 3 provided
src/property.h:27:15: note: candidate: 'constexpr detail::Property::Property(int, uint16_t, Type)'
     constexpr Property(const std::string_view _name, const std::uint16_t _id, const Type _type)
               ^~~~~~~~
src/property.h:27:15: note:   conversion of argument 1 would be ill-formed:
src/property.h:22:8: note: candidate: 'constexpr detail::Property::Property(const detail::Property&)'
 struct Property : public Mapper {
        ^~~~~~~~
src/property.h:22:8: note:   candidate expects 1 argument, 3 provided
src/property.h: In constructor 'Property::Property(const detail::Property&)':
src/property.h:62:62: error: 'const struct detail::Property' has no member named 'name'
     Property(const detail::Property& p) : detail::Property{p.name, p.id, p.type} {}
                                                              ^~~~
src/property.h:62:80: error: no matching function for call to 'detail::Property::Property(<brace-enclosed initializer list>)'
     Property(const detail::Property& p) : detail::Property{p.name, p.id, p.type} {}
```

Log file, not complete, aborted building after some time.
[logs_esphome-web-7d2a20_run.txt](https://github.com/user-attachments/files/17704218/logs_esphome-web-7d2a20_run.txt)

Do i. miss something while cleaning up the file for me? I use an ESP32-Wroom-32 device.

skycryer, 2024-11-11:
Nearly forgot my ESP Config 

```
esphome:
  name: heatingpump
  
  platform: ESP32
  board: nodemcu-32s

time:
  - platform: homeassistant
    id: homeassistant_time

api:
  encryption:
    key: 1cSivcBYHZBouiCgueZtgbwR0cBoKPVc48JNhvXK6JI=
logger:
  level: DEBUG
ota:
  - platform: esphome
    password: !secret ota_password
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Heatpump Fallback Hotspot"
    password: AsBl9n46jlOx

captive_portal: 

improv_serial:

substitutions:
  interval_very_fast: 15s
  interval_fast: 30s
  interval_medium: 60s
  interval_slow: 5min
  interval_very_slow: 15min
  interval_once_in_a_while: 1h

  entity_room_temperature: "sensor.durchschnittstemperatur_haus_ohne_keller"
  entity_humidity: "sensor.durchschnitt_luftfeuchtigkeit_haus"

#########################################
#                                       #
#   SPI configuration                   #
#                                       #
#########################################
spi:
  id: McpSpi
  clk_pin: GPIO18
  mosi_pin: GPIO23
  miso_pin: GPIO19

#########################################
#                                       #
#   CAN configuration                   #
#                                       #
#########################################
canbus:
  - id: !extend wp_can
    platform: mcp2515
    spi_id: McpSpi
    cs_pin: GPIO15

#########################################
#                                       #
#   Buttons                             #
#                                       #
#########################################
button:
  - platform: restart
    name: "Restart ESP32-PoE Technik"

#########################################
#                                       #
#   Packages                            #
#                                       #
#########################################
packages:
  sensors: !include yaml/sensors.yaml
# thz404: !include yaml/thz404.yaml
#  thz504: !include yaml/thz504.yaml
  thz55eco: !include yaml/thz5_5_eco.yaml
# ttf07: !include yaml/ttf07.yaml
  kondensatpumpe: !include yaml/kondensatpumpe.yaml

#########################################
#                                       #
#   EVU Sperre                          #
#                                       #
#########################################
switch:
  - platform: gpio
    name: "EVU Sperre"
    id: evu_sperre
    pin: GPIO5

```

kr0ner, 2024-11-11:
You forgot to copy the platformio options ;)

https://github.com/kr0ner/OneESP32ToRuleThemAll/blob/master/esp32-poe-technik.yaml#L4

skycryer, 2024-11-11:
Ok build works now and I got many values, now I need to check if Iit works to also set values and if I have what I need :-)

closed 2024-11-11 by skycryer

kr0ner, 2024-11-11:
Great 😀 have fun