kr0ner/OneESP32ToRuleThemAll #50: tecalor 5.5 eco heating night IST value wrong

skycryer, 2024-11-11:
So everything is working so far what I saw. Just the Heating Night value is not correct it looks like. I can set heating night and it shows up after some time on my heating pump settings. But the IST value stays the same as heating day is. Maybe wrong value used here?

kr0ner, 2024-11-11:
What is the exact name of the sensor? IST should tell you either the current room temperature, if you have an FEK or publish the temp via HA. If not it shows the SOLL value of the currently active day/night setting depending on the time of the day. Wait a few hours and it will be correct ;-)
The Day/Night switch is super weird ... it depends a lot on your settings in the heatpump ...
I run mine on "day-mode" all the time to be able to schedule everything via HA and not having to find the right SOLL/IST value for the current time.

skycryer, 2024-11-12:
The sensor name is **climate.heating_night**

[image]

So i find it really confusing. I was expecting that large number show the set value for night and the small value is what I set. So my night setting is 17, so it should show 17. And if I change the small number, sending it to heating pump and then when set it shows the value also in the large number.

But if I understand you right it will show the value different depending of the heating pump running in day or night mode? I can not see any changes after night mode is starting in the history data of home assistant.

skycryer, 2024-11-12:
> What is the exact name of the sensor? IST should tell you either the current room temperature, if you have an FEK or publish the temp via HA. If not it shows the SOLL value of the currently active [...]

How can I publish the room temperature from home assistant? And wich room should it be? I have temp values for every room in home assistant also an average value from all temp sensors of all rooms.

kr0ner, 2024-11-12:
> The sensor name is **climate.heating_night**
> 
> <img alt="Bildschirmfoto 2024-11-12 um 08 02 06" width="416" src="https://private-user-images.githubusercontent.com/3725699/385174487-a3eb95e3-764c-4f5e-af1c-f5fc757131e6.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzEzOTkwNjksIm5iZiI6MTczMTM5ODc2OSwicGF0aCI6Ii8zNzI1Njk5LzM4NTE3NDQ4Ny1hM2ViOTVlMy03NjRjLTRmNWUtYWYxYy1mNWZjNzU3MTMxZTYucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MTExMiUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDExMTJUMDgwNjA5WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9NzQzYzM3ZjcxYmI5NzM1MmY5YjBhZDY4ZWU1MmM1ODU3NGZjNzNkZDdmNzZmNjgwZDk0ZTViYWExMDdjMmRkYSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.7XIW3gTsLX2pFvPOKInikyUow8fAJxqQVMf95woEDuY">
> So i find it really confusing. I was expecting that large number show the set value for night and the small value is what I set. So my night setting is 17, so it should show 17. And if I change the small number, sending it to heating pump and then when set it shows the value also in the large number.
> 
> But if I understand you right it will show the value different depending of the heating pump running in day or night mode? I can not see any changes after night mode is starting in the history data of home assistant.

Ok, so if you spin the climate to e.g. 20 °C in HA, what does the heatpump do? Is it showing the correct value?
If you are not sending the temp, it will always show IST == SOLL
I remember there were some problems with reading back the value when you set it at the heatpump directly ... we might need to investigate further. I don't notice that, because I don't operate the heatpump via the display anymore. HA is just way more convenient ^^

> How can I publish the room temperature from home assistant? And wich room should it be? I have temp values for every room in home assistant also an average value from all temp sensors of all rooms.

In the main yaml there are two HA sensors defined. One for humidity one for temp. I'm sending the avg temperature of the house (excluding basement) to the heatpump

skycryer, 2024-11-12:
Sending the changes is working it is just confusing for me what infos should be shown. :D Also showing in heating night the set value for day is making me crazy. I can change the day and night temperature, but shown is night temp. Is there a sensor for the night temp so I can output it? I can not find one.

kr0ner, 2024-11-12:
```
                                      current_temp    target_temp               heating_sens cooling_sens vent_sens    which sensors to adjust when spinning  
auto heating = new HeatingDayNight(id(RAUMISTTEMP),id(VERSTELLTE_RAUMSOLLTEMP), id(HEIZEN), id(KUEHLEN), id(LUEFTUNG), Property::kRAUMSOLLTEMP_NACHT, Property::kKUEHL_RAUMSOLL_NACHT);
```
might even be sufficient to exchange VERSTELLTE_RAUMSOLLTEMP with RAUMSOLLTEMP_NACHT
and to add
```
  RAUMSOLLTEMP_NACHT:        !include { file: wp_temperature.yaml, vars: { property: "RAUMSOLLTEMP_NACHT"  , target: "HK1" }}
```
to common.yaml

kr0ner, 2024-11-12:
Like I said, I never tested this usecase because I see no reason for having the heatpump in day / night mode at all. I want to have one thermostat for heating that I can turn and it should set the temp without having to think if it is night and day. Same for ventilation and cooling.

skycryer, 2024-11-12:
> Like I said, I never tested this usecase because I see no reason for having the heatpump in day / night mode at all. I want to have one thermostat for heating that I can turn and it should set the temp without having to think if it is night or day. Same for ventilation and cooling.

Think I will just go with the same approach. 

I also added the average sensors for temp and humidity in esp code but not sure if that worked. Can not see the values anywhere. Heating card still shows the same values.

kr0ner, 2024-11-12:
> I also added the average sensors for temp and humidity in esp code but not sure if that worked. Can not see the values anywhere. Heating card still shows the same values.

It could be that FEK in your case needs to be emulated differently. Logs are always helpful :wink: 
There is no FEK connected, right?

> What mode do you run your heating pump on to set all manual? In handmode?
I'm running in automatic mode, but the schedules for heating and hot water are set from 12 to 00 or the other way round which makes it kind of "all day mode"

skycryer, 2024-11-12:
No there is no FEK, it just shows the average temp now. But that is even more confusing because I set the heating on day to 19 and do not expect it to go higher. Maybe I just need to keep it for now until I am used to it 
/
/
/

kr0ner, 2024-11-12:
But you get the readings, right?
[image]
and those values sound reasonable ... can you check at which time the inverter was actually running?
> I See on the heating pump under diagnose - energiebilianz values for heating, warm water and ventilation for last 24h and more. So I hoped I can get this also over modbus.

Again ... no modbus involved here. The values you see there are the same you see here, at least to ones from the mushroom cards below should match. Or what values are you expecting? You need to be a bit more precise :wink: otherwise we'll have to do days of ping-pong

> The values from screenshot are not enough to fill nearly the gaps.

For me it looks like this
[image]

You just need to spend more time on this topic an pinpoint the problem so that we can work on it. Like when was the inverter turned on? What values are shown in the display and what are obtained by HA? What is for example the peak at 2 a.m. ?

skycryer, 2024-11-12:
Sorry about that not really sure how to describe it.

I am looking for the energy usage values of heating, water and ventilation. If I use the 

el_aufnahmeleistung_heiz_summe_kwh
sensor.el_aufnahmeleistung_ww_summe_kwh

they seem to be to low. If I check for example 2am on my screenshot, there we had "legionelen" heating active. So warm water got heated up to 61 degree celsius. But if I check the energy dashboard the value for warm water is way smaller then expected.

Any you have like I saw negativ values in energy dashboard like I when I add "Wärmerückgewinnung" but not sure if that is a needed value even because it gets negativ.

kr0ner, 2024-11-12:
> they seem to be to low. If I check for example 2am on my screenshot, there we had "legionelen" heating active. So warm water got heated up to 61 degree celsius. But if I check the energy dashboard the value for warm water is way smaller then expected.

haha, ok now I get it ^^
First of all do yourself a favor and set the time for Anti-Legionellenmodus to around 1 p.m. there the outside temp is usually the max of the day saving you some € throughout the year. And then please complain to Tecalor/Stiebel Eltron, why they don't count that into hot water consumption :D you could in theory manually reproduce the values .. there is MOTORSTROM, -LEISTUNG and SPANNUNG ... with an integral sensor you could sum that up. Also you could detect when the ZUSATZHEIZUNG is active and while it is, assume that for each level it will consume 16A * 230V and also sum those values up. Also for ventilation you either have to make an educated guess or install some additional current sensors :wink: that is just how that thing works.

> Any you have like I saw negativ values in energy dashboard like I when I add "Wärmerückgewinnung" but not sure if that is a needed value even because it gets negativ.
Only the EL* values should be added to the energy dashboard ... the other ones are WÄRME ... which is basically EL_* times the COP of your heatpump. It only tells you how efficient the heatpump is in transferring heat. The factor should roughly match the ones in the spec like A2/W35 and so on

How are you obtaining the value for overall power consumption? Volkszähler on energymeter? Or impulse counter?

> So I have no nearly nothing unknown,

Are you really sure about that? There is stuff like fridge(100W peaks), stove & oven (1-2 kw per day), hairdryer, dryer (9W) washing machine(9W), dishwasher (2W), freezer (100W peaks), trafo for doorbell (10W), charging station for electrical toothbush (4W). "Kleinvieh macht auch Mist"
those are at least the values that I found out for our devices

skycryer, 2024-11-12:
Everything in my house is counted, only limit is the stove/hotplate and until now the heating pump. Fridge, oven, computer, tvs, server and other stuff are getting tracked. Only minimum value that are not even listed could show up for lights. But they are so low every day trough auto on/off that it does not even show up I think.

The house power consumption is from my e3dc home power plant. I also had to add a señor to calculate a rough power usage of that system for converting the solar power and the system itself.  

So i am sure I have some power not tracked but it does not show up. Maybe because the values from the heating pump are not exact (little to big maybe so they cover the other untracked energy). But as said that should be minimum values here.

Because of "legionellen mode". I was already thinking some month before to move that around 2pm also I could use solar power for that in summer. But the service technician meant that is not ideal because of risk of burning because of hot water. But I think I will give it a try. Most external power came while summer because of that mode from outside and cost us money.

mkaiser, 2024-11-12:
I experienced some too high values with the "EL_AUFNAHMELEISTUNG*"-sensors, too. The sensors reported sometimes more energy /kWh usage than the house load, measured by my inverter. It happened in some early tests of this integration.... let's keep an eye on that ;)

btw. slightly off-topic: 

Im am wondering: What does the "**2WE**" stand for?

```
    PROPERTY(WAERMEERTRAG_2WE_WW_TAG_WH, 0x0922);
    PROPERTY(WAERMEERTRAG_2WE_WW_TAG_KWH, 0x0923, Type::et_double_val);
    PROPERTY(WAERMEERTRAG_2WE_WW_SUM_KWH, 0x0924);
    PROPERTY(WAERMEERTRAG_2WE_WW_SUM_MWH, 0x0925, Type::et_double_val);
    PROPERTY(WAERMEERTRAG_2WE_HEIZ_TAG_WH, 0x0926);
    PROPERTY(WAERMEERTRAG_2WE_HEIZ_TAG_KWH, 0x0927, Type::et_double_val);
    PROPERTY(WAERMEERTRAG_2WE_HEIZ_SUM_KWH, 0x0928);
    PROPERTY(WAERMEERTRAG_2WE_HEIZ_SUM_MWH, 0x0929, Type::et_double_val);
```

kr0ner, 2024-11-12:
> I experienced some too high values with the "EL_AUFNAHMELEISTUNG*"-sensors, too. The sensors reported sometimes more energy /kWh usage than the house load, measured by my inverter. It happened in [...]

I think it is "zweiter Wärmeerzeuger" means solar panels for hot water.
You can always double check via MOTORLEISTUNG ;)

kr0ner, 2024-11-12:
> Everything in my house is counted, only limit is the stove/hotplate and until now the heating pump. Fridge, oven, computer, tvs, server and other stuff are getting tracked. Only minimum value that [...]

This is how I estimate the power consumption for the ventilation

```
{% if is_state('fan.esp32_poe_technik_lueft_stufe_tag', 'on') %}
  {% if state_attr('fan.esp32_poe_technik_lueft_stufe_tag', 'percentage') == 33.0 -%}
    30.0
  {%- elif state_attr('fan.esp32_poe_technik_lueft_stufe_tag', 'percentage') == 66.0 -%}
    50.0
  {%- elif state_attr('fan.esp32_poe_technik_lueft_stufe_tag', 'percentage') == 100.0 -%}
    80.0
  {%- else -%}
    0.0
  {%- endif %}
{%- else -%}
  0.0
{%- endif %}
```
plus an additional integral sum sensor to be able to add to energy dashboard

skycryer, 2024-11-12:
> TBH this is just bullshit and those technicians have no clue what they are talking about. Yes ... the water is hotter that day ... perfect time to fill up a bathtub 😉

My wife will love to hear that 

> My heatpump was just setup with default values and was using twice as much energy as needed ... after tweaking I'm down do 4-5 Verdichterstarts per day and never over 25kWh

Any recommendations what to optimize? We had now 1 year with that heating pump and around 3mwh this year came just from heating and hot water. I did now set heating on night 2-3 degree lower then on day for test. But beside setting lower ventilation I have no idea what could be optimized. Ok now with this integration I can lower values when we are longer not home for example but anything else?

kr0ner, 2024-11-12:
This is my setup:
[image]
1, reduce the Heizleistung so that the heatpump is continuously running (~5 Verdichterstarts per day).
2, Open up all valves in all rooms and for e.g. bedroom reduce temperature in rooms by reducing the amount of water that flows through them. Opposite for bathroom. (Hydraulischer Abgleich)
3, Send avg room temperature and set RAUMEINFLUSS to 100%
4, Reduce Heizkreispumpe (how fast water flows through the FBH) to have a spread of 4° between VOR- and RÜCKLAUF
5, monitor VERDICHTERSTARTS daily to see, if the settings are still good under the current outside temp conditions
[image]

kr0ner, 2024-11-12:
> I did now set heating on night 2-3 degree lower then on day for test. 
@skycryer don't do that ... IMO this consumes even more energy and is counterproductive ... best with heat pump is to drive at a constant rate and avoid peaks at all cost. 
[image]

vs 

[image]

skycryer, 2024-11-13:
@kr0ner for understanding, I use now the average room temp sensor and see this on the heating for IST temp

[image]

I set "Raumeinfluss" to 100%, now the heating is running on Automatic. I think it now tries to match the room temp IST with the Raumsolltemp? Does that mean the heater will only start if the average room temp goes under Raumsolltemp?

skycryer, 2024-11-13:
[image]
[image]

SO, I set also heating program from 00:00 to 24:00 so all day daytime mode. But I can not set consistent the temperature only with the climate.heating_day. It resets sometimes, just did it few minutes ago.

But if I then set it with climate.heating_night, it changes day and night mode.

https://www.youtube.com/watch?v=90ZGW-X6nY0

But I know that yesterday I was able to change the daytime setting. Is there any setting I miss? 

And every time I change heating to "Heizmodus" it changes back to Auto. I tried now setting heating to 22 degree, as we have an average of 21.3 degree shown I thought it should start heating again. But not yet. Maybe this is related to the "Raumeinfluss" setting.

kr0ner, 2024-11-13:
> I set "Raumeinfluss" to 100%, now the heating is running on Automatic. I think it now tries to match the room temp IST with the Raumsolltemp? Does that mean the heater will only start if the average room temp goes under Raumsolltemp?

More or less :wink: 
The heatpump does not immediately apply the changes. It needs to run the Heizkreispumpe to get and update on the Rücklauf-Temperature. For me this happens every hour. You can check binary_sensor.esp32_poe_technik_heizkreispumpe for that info.
It'll then decide whether to heat or not. Leave the climate in auto as the mode is just the feedback from the heatpump, but can't be used to tell the heatpump what to do. This is solely done via target temp.
Even Raumeinfluss 100 will take other values into account (e.g. outside temp, Vorlauf-/Rücklauf temp) ... but it will react faster to other heat sources like sun or and additional fireplace. Overall it will try to keep the target temp ... but it is a slow system ... give it some time to react. That is why it should be run all the time without Nachtabsenkung

kr0ner, 2024-11-13:
> SO, I set also heating program from 00:00 to 24:00 so all day daytime mode. But I can not set consistent the temperature only with the climate.heating_day. It resets sometimes, just did it few minutes ago.
But if I then set it with climate.heating_night, it changes day and night mode.
https://www.youtube.com/watch?v=90ZGW-X6nY0

This is what i meant with 24 - 0 or 0 - 24 :D
I think you have "all night" mode now

skycryer, 2024-11-13:
> > SO, I set also heating program from 00:00 to 24:00 so all day daytime mode. But I can not set consistent the temperature only with the climate.heating_day. It resets sometimes, just did it few minutes ago.
> > But if I then set it with climate.heating_night, it changes day and night mode.
> > https://www.youtube.com/watch?v=90ZGW-X6nY0
> 
> This is what i meant with 24 - 0 or 0 - 24 :D I think you have "all night" mode now Try flipping the values and you should have all-day mode

So that does not work, I can not set daytime from 24:00 to 00:00 only other way. And now overtime I try to change the temp of day or night it is reseting not 19 degree. No sure if that is set somewhere and using it for any reason. I think that was before the set wanted room temp before I had added the average room temp sensor data.

kr0ner, 2024-11-13:
I'll check my settings in the evening.

For all values that can be set, there should be a sensor that reads it back to confirm that it works. Can you double check ?

skycryer, 2024-11-13:
[image]

Not sure wich one is the right one but this is the only that fit with the 19 degree showing. And I can see it showed the changes. I could also the on the display of the heating pump that it got the new values. But then short after it changed them back, no idea why

kr0ner, 2024-11-13:
> [image]
> 
> Not sure wich one is the right one but this is the only that fit with the 19 degree showing. And I can see it showed the changes. I could also the on the display of the heating pump that it got the new values. But then short after it changed them back, no idea why

Did you make any changes in common.yaml for the climates? Can you disable the Heizung Nacht climate?

What does VERSTELLTE_RAUMSOLLTEMP say?

skycryer, 2024-11-13:
> Did you make any changes in common.yaml for the climates? Can you disable the Heizung Nacht climate?

No changes made in common.yaml, where should I disable heating night? In Home Assistant or remove it anywhere?

> 
> What does VERSTELLTE_RAUMSOLLTEMP say?

[image]

kr0ner, 2024-11-13:
> > Did you make any changes in common.yaml for the climates? Can you disable the Heizung Nacht climate?
> 
> No changes made in common.yaml, where should I disable heating night? In Home Assistant or remove it anywhere?
> 
> > 
> > What does VERSTELLTE_RAUMSOLLTEMP say?
> 
> 
> 
> [image]
> 
> 

In HA in the Esp device itself. Definitely there is some competition going on 😅
One entity sets it... The other one resets it

skycryer, 2024-11-13:
Question is Woche one. I die only Change the Sensor for average rooms and Humidity.

kr0ner, 2024-11-13:
Either in common yaml or first via HA .. maybe that helps already
[image]

kr0ner, 2024-11-13:
Still weird behavior ... They should not compete at all ... might really be a problem of the day-night-mode ... let's see later. Hope it'll work for now

skycryer, 2024-11-13:
So temp is stable since over an hours now, heating works again. It is getting warmer slowly again :D

skycryer, 2024-11-14:
So, it is the next day now. Heating was working so far without heating_night removed. It heats more then I set like it set it to 22.5 and it heat to 23 (still heated before I set it down to 22 in the night). But maybe this is like for water where it heat some degree hotter then I set.

For me i do not need the night mode as I do set everything now over the day heating. But it is strange the reset each other.

kr0ner, 2024-11-14:
good to hear that it works. A bit deviation is always in there ... it gets better, if you leave the temp as is for the whole day :wink: 

yeah ... it is very strange and I will investigate further

skycryer, 2024-11-14:
> good to hear that it works. A bit deviation is always in there ... it gets better, if you leave the temp as is for the whole day 😉
> 
> yeah ... it is very strange and I will investigate further. Leaving this open until fixed.
> 
> @skycryer could you maybe reenable the heating_night to make it fail again and provide me the log output? I'd like to understand what is going on ;)

Wich log exactly do you need?

skycryer, 2024-11-14:
[image]
[image]
[image]
[image]

Could be that it is missing a sensor data for night. What I see in the heating_night code is it reference to this property if I am not wrong. But there is no sensor in my esp showing this señor data.
`
 PROPERTY(RAUMSOLLTEMP_NACHT, 0x0008, Type::et_dec_val);`

kr0ner, 2024-11-14:
> Wich log exactly do you need?

The ones from ESPHome -> ESP32 -> logs
[image]

[image]

kr0ner, 2024-11-14:
the missing sensor can be added like this:
```
  RAUMSOLLTEMP_NACHT:        !include { file: wp_temperature.yaml, vars: { property: "RAUMSOLLTEMP_NACHT"  , target: "HK1" }}
you can put it in wp_base.yaml next to the existing ones
```

skycryer, 2024-11-14:
[logs_esphome-web-7d2a20_run.txt](https://github.com/user-attachments/files/17746837/logs_esphome-web-7d2a20_run.txt)

Here the log, directly from starting and I also played with changing temp for day and night heating. Strongly, for now it works correct.

Maybe there was problems sending it? But when it was only day heating it worked flawless.

kr0ner, 2024-11-14:
I think now now the problem ... there is kRAUMSOLLTEMP_I, RAUMSOLLTEMP_NACHT and VERSTELLTE_RAUMSOLLTEMP.
kRAUMSOLLTEMP_I = temp for day mode
RAUMSOLLTEMP_NACHT = temp for night mode
VERSTELLTE_RAUMSOLLTEMP = either the value of kRAUMSOLLTEMP_I or  RAUMSOLLTEMP_NACHT depending on the time of day and the heating schedule

Means when you turn e.g. the night climate control, the CAN message for RAUMSOLLTEMP_NACHT will be sent setting the value in the heatpump. But day is active so VERSTELLTE_RAUMSOLLTEMP will be read and displayed as target temp also for the night climate ... which is confusing. I'll provide a fix

kr0ner, 2024-11-14:
this should fix it 
https://github.com/kr0ner/OneESP32ToRuleThemAll/pull/51/files#diff-59d82129f39e054312c9d9541a11dd0a5894539199c0132756356ba4c8840d47R201-R217

skycryer, 2024-11-14:
Is that merged into main branch? Should I just update the local files and rebuild?

kr0ner, 2024-11-14:
I'll merge it now ... expect some breaking changes ... check the README.md for instructions.
It is a necessary change to allow easier use in the future

PR #51 (2024-11-14): Restructure repo to allow easier setup
also fixes #50
closed 2024-11-14 by kr0ner