kr0ner/OneESP32ToRuleThemAll #52: Can't get any data with canbus: platform: esp32_can

giggo, 2024-11-12:
I am using an ESP32-S3-Devkitc-1

```
esphome:
  name: waermepumpe
  friendly_name: "Wärmepumpe CAN Interface"
  platformio_options:
    build_flags:
      - "-std=gnu++17"
    build_unflags:
      - "-std=gnu++11"
      - "-fno-rtti"

  platform: esp32
  board: esp32-s3-devkitc-1

```
I had to remove the SPI definition and modify the Canbus Config like this

```
canbus:
  - platform: esp32_can
    # TODO: Rename id
    id: !extend wp_can
    tx_pin: GPIO5
    rx_pin: GPIO4
    #can_id: 680
    use_extended_id: false
    bit_rate: 20kbps
```

This is a canbus config that worked for me with robertreiter's code but I can't received any data from my Stiebel Eltron WPL18 heatpump. DEBUG is enabled. 
I just get this:

```
...
[00:54:14][D][canbus:035]: send standard id=0x6a2 rtr=FALSE size=7
[00:54:14][D][canbus:035]: send standard id=0x6a2 rtr=FALSE size=7
[00:54:14][D][canbus:035]: send standard id=0x6a2 rtr=FALSE size=7
...

```
Any idea?

giggo, 2024-11-15:
The commented CAN ID was a test. I get the same results with the uncommented CAN ID.

Here are more logs:
```
INFO ESPHome 2024.10.3
INFO Reading configuration /config/esphome/waermepumpe.yaml...
WARNING GPIO3 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Detected timezone 'Europe/Berlin'
INFO Detected timezone 'Europe/Berlin'
INFO Detected timezone 'Europe/Berlin'
INFO Detected timezone 'Europe/Berlin'
INFO Detected timezone 'Europe/Berlin'
INFO Detected timezone 'Europe/Berlin'
INFO Detected timezone 'Europe/Berlin'
INFO Detected timezone 'Europe/Berlin'
INFO Generating C++ source...
INFO Compiling app...
Processing waermepumpe (board: esp32-s3-devkitc-1; framework: arduino; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32S3 240MHz, 320KB RAM, 8MB Flash
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5 
 - toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch5
Dependency Graph
|-- AsyncTCP-esphome @ 2.1.4
|-- libSML @ 1.1.4+sha.c166372
|-- WiFi @ 2.0.0
|-- FS @ 2.0.0
|-- Update @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 3.2.2
|-- DNSServer @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- ArduinoJson @ 6.18.5
RAM:   [==        ]  15.0% (used 49252 bytes from 327680 bytes)
Flash: [========  ]  77.9% (used 1429309 bytes from 1835008 bytes)
========================= [SUCCESS] Took 4.39 seconds =========================
INFO Successfully compiled program.
INFO Connecting to 192.168.1.42
INFO Uploading /data/build/waermepumpe/.pioenvs/waermepumpe/firmware.bin (1429696 bytes)
Uploading: [============================================================] 100% Done...

INFO Upload took 7.89 seconds, waiting for result...
INFO OTA successful
INFO Successfully uploaded program.
INFO Starting log output from 192.168.1.42 using esphome API
INFO Successfully connected to waermepumpe @ 192.168.1.42 in 7.205s
INFO Successful handshake with waermepumpe @ 192.168.1.42 in 0.042s
[22:14:00][I][app:100]: ESPHome version 2024.10.3 compiled on Nov 13 2024, 00:53:01
[22:14:00][C][wifi:600]: WiFi:
[22:14:00][C][wifi:428]:   Local MAC: C0:4E:30:3A:08:2C
[22:14:00][C][wifi:433]:   SSID: 'giggo.de.kati'[redacted]
[22:14:00][C][wifi:436]:   IP Address: 192.168.1.42
[22:14:00][C][wifi:440]:   BSSID: 1E:E8:29:9D:C2:D6[redacted]
[22:14:00][C][wifi:441]:   Hostname: 'waermepumpe'
[22:14:00][C][wifi:443]:   Signal strength: -70 dB ▂▄▆█
[22:14:00][C][wifi:447]:   Channel: 11
[22:14:00][C][wifi:448]:   Subnet: 255.255.255.0
[22:14:00][C][wifi:449]:   Gateway: 192.168.1.254
[22:14:00][C][wifi:450]:   DNS1: 192.168.1.254
[22:14:00][C][wifi:451]:   DNS2: 0.0.0.0
[22:14:00][C][logger:185]: Logger:
[22:14:00][C][logger:186]:   Level: DEBUG
[22:14:00][C][logger:188]:   Log Baud Rate: 115200
[22:14:00][C][logger:189]:   Hardware UART: USB_CDC
[22:14:00][D][canbus:035]: send standard id=0x6a2 rtr=FALSE size=7
[22:14:00][C][uart.arduino_esp32:151]: UART Bus 0:
[22:14:00][C][uart.arduino_esp32:153]:   RX Pin: GPIO2
[22:14:00][C][uart.arduino_esp32:155]:   RX Buffer Size: 256
[22:14:00][C][uart.arduino_esp32:157]:   Baud Rate: 9600 baud
[... 935 lines cut ...]
[22:15:03][D][canbus:035]: send standard id=0x6a2 rtr=FALSE size=7
[22:15:03][D][canbus:035]: send standard id=0x6a2 rtr=FALSE size=7
```

Where is this?

```
static const CanMember CanMembers[] =
    {
        //  Name              CanId     ReadId          WriteId         ConfirmationID
        {"ESPCLIENT", 0x680, {0x00, 0x00}, {0x00, 0x00}, {0xE2, 0x00}}, // The ESP Home Client, thus no valid read/write IDs
        {"PUMP", 0x180, {0x31, 0x00}, {0x30, 0x00}, {0x00, 0x00}},
        {"FE7X", 0x301, {0x61, 0x01}, {0x00, 0x00}, {0x00, 0x00}},
        {"FEK", 0x302, {0x61, 0x02}, {0x00, 0x00}, {0x00, 0x00}},
        {"MANAGER", 0x480, {0x91, 0x00}, {0x90, 0x00}, {0x00, 0x00}},
        {"FE7", 0x602, {0xC1, 0x02}, {0x00, 0x00}, {0x00, 0x00}}};

```
I can't find this anywhere.

```
grep -r CanMembers * | wc -l
       0

```

kr0ner, 2024-11-15:
> static const CanMember CanMembers[] =
    {
        //  Name              CanId     ReadId          WriteId         ConfirmationID
        {"ESPCLIENT", 0x680, {0x00, 0x00}, {0x00, 0x00}, {0xE2, 0x00}}, // The ESP Home Client, thus no valid read/write IDs
        {"PUMP", 0x180, {0x31, 0x00}, {0x30, 0x00}, {0x00, 0x00}},
        {"FE7X", 0x301, {0x61, 0x01}, {0x00, 0x00}, {0x00, 0x00}},
        {"FEK", 0x302, {0x61, 0x02}, {0x00, 0x00}, {0x00, 0x00}},
        {"MANAGER", 0x480, {0x91, 0x00}, {0x90, 0x00}, {0x00, 0x00}},
        {"FE7", 0x602, {0xC1, 0x02}, {0x00, 0x00}, {0x00, 0x00}}};

This ist from the original repo, since you said that that one was working for you ;)

giggo, 2024-11-29:
1 The logs from above had debug enabled
2 Cable is good. I can receive data if I use robertreiters code
3 I have a can ID defined in common.yaml but can't find can_id in your wp_base.yaml

kr0ner, 2024-11-29:
>1 The logs from above had debug enabled
>2 Cable is good. I can receive data if I use robertreiters code
>3 I have a can ID defined in common.yaml but can't find can_id in your wp_base.yaml

1, I think we need to go VERBOSE then to see what can messages are actually received.
2, ok, good to know
3, it should only be defined in one place ... they are extending each other and ultimately form a valid yaml. Entry point should be the top level esp32-poe-technik.yaml

closed 2025-01-06 by github-actions[bot]

kr0ner, 2025-02-02:
@giggo want to try it again?