kr0ner/OneESP32ToRuleThemAll #59: Stiebel WPC 10 Cool

Hotzenplotz1, 2024-11-20:
Hello everyone,

I first had to find my way around Github.

I have connected an ESP32 MCP25115 and connected it to the CAN bus. I have attached the yaml, as well as the logbook of the query.
Unfortunately I can't get any further because I don't have the basics. Perhaps you can give me some tips on how I can familiarize myself with it.

Thank you very much
HP
[yaml.txt](https://github.com/user-attachments/files/17826691/yaml.txt)
[logs_heatingpump_logs(2).txt](https://github.com/user-attachments/files/17826696/logs_heatingpump_logs.2.txt)

Hotzenplotz1, 2024-11-20:
Okay, got it.

My problem is to understand your how to...

kr0ner, 2024-11-20:
Can you be more specific?

Hotzenplotz1, 2024-11-20:
Yes, i dont know how to clone the repo or copy the sym link. 
I dont have the basics to solve the problems. Mostly trial and error...

mkaiser, 2024-11-20:
hey, 

I guess your WPC 10 Cool could be similar to my Tecalor TTF 07 cool. 

I am still in the debug-phase, but most of the sensors work - the control (set temperatures) does not work, yet

The code is currently completely messy. Just wait a week or so and I will have it mainlined again :)

https://github.com/mkaiser/OneESP32ToRuleThemAll/tree/dbg_msg

Hotzenplotz1, 2024-11-20:
yes. tecalor amd se should be the same.

kr0ner, 2024-11-20:
> Yes, i dont know how to clone the repo or copy the sym link. I dont have the basics to solve the problems. Mostly trial and error...
> 
> If you can show me how to geht the data from github to HA it will be nice!

Ok, here are soem hints:
1, Install VSCode addon for Home Assistant https://github.com/hassio-addons/addon-vscode
2, navigate to esphome folder, right click and open in terminal
3, execute : git clone https://github.com/kr0ner/OneESP32ToRuleThemAll.git
4, execute: cp OneESP32ToRuleThemAll/esp32-poe-technik.yaml esp32-poe-technik.yaml
5, comment out Thz504.yaml in esp32-poe-technik.yaml and remove # in front of ttf07
6, adopt the pins to your setup
7, build with esphome and do OTA

Hotzenplotz1, 2024-11-20:
How i clone 3.?

Hotzenplotz1, 2024-11-20:
easy :-) and done. What means 5 and 7?

6. understand :-)

kr0ner, 2024-11-20:
> easy :-) and done. What means 5 and 7?
> 
> 6. understand :-)

5. is exactly what is written there.
7. Is building the Software with ESPhome and installing it to the ESP

Hotzenplotz1, 2024-11-20:
Okay.
I got Nr. 5.
But I'm hanging on building the Software with ESPhome and installing it to the ESP.

Sorry for wasting your time. This projekt is a little bit to big for me.

kr0ner, 2024-11-20:
> Okay.
> I got Nr. 5.
> But I'm hanging on building the Software with ESPhome and installing it to the ESP.
> 
> Sorry for wasting your time. This projekt is a little bit to big for me.

But you have a running ESP already ... How did you install the the software there? It is fairly easy check esphome.io
And try to consolidate chatgpt and the Esphome discord or Forum. Don't give Up so easily

Hotzenplotz1, 2024-11-20:
Yes, ESP is running. And I also have the application running as posted, but without values.
For installation, I loaded the files from git hub into my home assistant via my desktop. 
Then I copied the text from the .yaml into the editor of the application using drag and drop.
Surely there is an easier way, as you have already shown me.

i am now missing the specific step of how to get your .yaml files onto the ESP32. simply pasting them is not an option, it doesn't work.

I am very sure that the solution is very simple, but unfortunately I don't have the wording to search for it in databases like ESP-Home or GPT.

Quotic, 2024-11-20:
But to be honest, you might want to familiarize yourself with esphome in general and implement other sensors with it before you tackle this complex topic.

Hotzenplotz1, 2024-11-20:
Never give up!

Thank you for your support.

I have carried out all the steps. when flashing ota I get an error message.

it is not clear to me whether I have to enter all the passwords, api etc. manually in the .yaml or whether these are taken over.

I have attached the script.
[esp.txt](https://github.com/user-attachments/files/17835978/esp.txt)

kr0ner, 2024-11-20:
> Never give up!
>
> Thank you for your support.
>
> I have carried out all the steps. when flashing ota I get an error message.
>
> it is not clear to me whether I have to enter all the passwords, api etc. [...]

I think your main problem is that you are not familiar with ESPHome. The initial yaml is for an esp32dev board and the one in the previous post still has the esp-poe from my config. Yours has WiFi mine has Ethernet. You need to work on the base config and make sure you understand what each section does. Also the yaml without logs or compiler errors does not help much, because I need to compile it in my head than.

Hotzenplotz1, 2024-12-01:
Hello everyone.

I have set up the ESP according to your instructions and am now getting the first values.
And also the following error message:
[W][component:237]: Component canbus took a long time for an operation (58 ms).
[W][component:238]: Components should block for at most 30 ms.
That should be related to the bit_rate, right? Actual: 20kbps

A few values from the heating also fit (room temperature, outside temperature, return temperature ....), others still have errors in the display, others are not found or are not logically linked.

As I have understood, I now check which values from the log match my values on the heating system and then adjust the address in the property h or adjust their addresses.
Is this the correct procedure?

Hotzenplotz1, 2024-12-01:
I use TTF07.  It would be very nice if you can share your result!

Thanks!

Hotzenplotz1, 2025-01-05:
> Hi,
> 
> > [W][component:237]: Component canbus took a long time for an operation (58 ms).
> > [W][component:238]: Components should block for at most 30 ms.
> > That should be related to the bit_rate, right? Actual: 20kbps
> 
> The Bit rate is fixed at 20kbit/s. The Long taking operations propably result by not answered requests / wrong register mappings
> 
> > A few values from the heating also fit (room temperature, outside temperature, return temperature ....), others still have errors in the display, others are not found or are not logically linked.
> > As I have understood, I now check which values from the log match my values on the heating system and then adjust the address in the property h or adjust their addresses.
> 
> Did you use the TTF07 define?
> 
> The Heatpump of the repos owner differs very much from my TTF07 ( and your WPC).
> 
> IT took me some weeks to get most of the register adresses working. But my findings are not upstreamed yet. The easy way for you would be to wait 2-3 weeks until I find some time to finish that.

Good morning and a happy new year,

have you already had the opportunity to collect new values?
Or could you be so kind as to share your current status?

Thank you very much!

mkaiser, 2025-01-10:
Please Check https://github.com/kr0ner/OneESP32ToRuleThemAll/issues/47#issuecomment-2573921300

kr0ner, 2025-01-22:
please have a look if https://github.com/kr0ner/OneESP32ToRuleThemAll/pull/93 is sufficient

kr0ner, 2025-01-28:
Without logs I can't help. Please post the logs and in case you changed the yamls I'd also need the changes there in order to be able to see whats going wrong

kr0ner, 2025-01-29:
Did you really comment out the thz504 and remove the comment from ttf07?

[image]

If yes, which state is your repo at? Please pull latest master an try again.
Those sensors should not appear for TTF07
```
[18:00:25][C][template.binary_sensor:028]: Template Binary Sensor 'SCHALTPROGRAMM_AKTIV'
[18:00:25][C][template.binary_sensor:028]: Template Binary Sensor 'VERDICHTER'
[18:00:25][C][template.binary_sensor:028]: Template Binary Sensor 'WARMWASSERBEREITUNG'
[18:00:25][C][template.binary_sensor:028]: Template Binary Sensor 'ELEKTRISCHE_NACHERWAERMUNG'
[18:00:25][C][template.binary_sensor:028]: Template Binary Sensor 'SERVICE'
[18:00:25][C][template.binary_sensor:028]: Template Binary Sensor 'HEIZEN'
[18:00:25][C][template.binary_sensor:028]: Template Binary Sensor 'KUEHLEN'
[18:00:25][C][template.binary_sensor:028]: Template Binary Sensor 'LUEFTUNG'
[18:00:25][C][template.binary_sensor:028]: Template Binary Sensor 'EVU_SPERRE'
[18:00:25][C][template.binary_sensor:028]: Template Binary Sensor 'FILTERWECHSEL_BEIDE'
[18:00:25][C][template.binary_sensor:028]: Template Binary Sensor 'HEIZKREISPUMPE'
[18:00:25][C][template.binary_sensor:028]: Template Binary Sensor 'ABTAUEN_VERDAMPFER'
[18:00:25][C][template.binary_sensor:028]: Template Binary Sensor 'FILTERWECHSEL_ABLUFT'
[18:00:25][C][template.binary_sensor:028]: Template Binary Sensor 'FILTERWECHSEL_ZULUFT'
[18:00:25][C][template.binary_sensor:028]: Template Binary Sensor 'AUFHEIZPROGRAMM_AKTIV'
[18:00:25][C][template.binary_sensor:028]: Template Binary Sensor 'SOMMERBETRIEB_AKTIV'
[18:00:25][C][template.binary_sensor:028]: Template Binary Sensor 'OFEN_KAMIN_AKTIV'
```

Hotzenplotz1, 2025-02-01:
I pulled the last repo and get the follow log:

[logs_esp32-poe-technik_logs(3).txt](https://github.com/user-attachments/files/18628375/logs_esp32-poe-technik_logs.3.txt)

kr0ner, 2025-02-01:
Change the 301 to 601 and the 302 to 602 and flash again. That should help

substitutions:
  kessel_can_id: "0x180"
  hk1_can_id: "0x301"
  hk2_can_id: "0x302"
  manager_can_id: "0x6a1"
  espclient_can_id: "0x6a2"

Hotzenplotz1, 2025-02-01:
I changed in 2 files:
- common.yaml and
- communication
all 301 to 601 and 302 to 602.

Log:

[logs_esp32-poe-technik_logs(5).txt](https://github.com/user-attachments/files/18628518/logs_esp32-poe-technik_logs.5.txt)

kr0ner, 2025-02-01:
There should be no need to change communication.h

static const CanMember ESPClient{ESPCLIENT_ID, "ESPClient"};
static const CanMember Manager{MANAGER_ID, "Manager"};
static const CanMember Kessel{KESSEL_ID, "Kessel"};
static const CanMember HK1{HK1_ID, "HK1"};
static const CanMember HK2{HK2_ID, "HK2"};

Does your file look like this? What does "git status" print?

Hotzenplotz1, 2025-02-01:
Can we write in dicord?

kr0ner, 2025-02-01:
Your branch is 28 commits behind. So it can't work like that.

Do a "git stash" followed by "git pull origin master --rebase"

Hotzenplotz1, 2025-02-01:
You are the best!

[image]

But no more results...

[logs_esp32-poe-technik_logs(7).txt](https://github.com/user-attachments/files/18629148/logs_esp32-poe-technik_logs.7.txt)

kr0ner, 2025-02-01:
I think you need to try the other can Bus on your heatpump

Hotzenplotz1, 2025-02-01:
What exactly do you mean by that?

kr0ner, 2025-02-01:
There are two CANs in the heatpump ... one is attached to the display .. the other one should be used for ISG (or our ESP)

[image]

https://www.stiebel-eltron.ch/content/dam/ste/cdbassets/current/bedienungs-_u_installationsanleitungen/instructionandinstallationmanual_doc-00068270.pdf

[image]

Please confirm that the ESP is connected there

Hotzenplotz1, 2025-02-02:
I have checked the connection of the MCP2515 and the FEK: The FEK remote control is already connected to terminal block X2 as described in the instructions.
The MCP2515 is connected in parallel to the remote control, I have no other way of connecting the MCP.

Hotzenplotz1, 2025-02-03:
I have just found an error: The FEK connections of the CAN bus were reversed, so I only got room temperature and room humidity. I swapped the connections and started the whole thing again.

I have adjusted the CAN ID as follows:

substitutions:
  kessel_can_id: "0x180"
  hk1_can_id: "0x601"
  hk2_can_id: "0x602"
  manager_can_id: "0x6a1"
  espclient_can_id: "0x6a2"

Now I also get values:

[logs_esp32-poe-technik_run.txt](https://github.com/user-attachments/files/18644036/logs_esp32-poe-technik_run.txt)

kr0ner, 2025-02-03:
There is some weird stuff going on. The same properties are received multiple times in a row, but with different values. Can you please compile with loglevel DEBUG instead of INFO? Would also be nice to have some pictures of the display on the heatpump + the logs that were recorded during that time. That helped alot with other issues. In the meantime I will have a look at the other posts about the WPC 10 Cool.

Hotzenplotz1, 2025-02-03:
Which pictures do you need?

kr0ner, 2025-02-03:
basically like this 

[image]

I need to match the values from the logs with the values seen in the display. Altough that wil be a tough one ... there are so many duplicates with different values ...

Hotzenplotz1, 2025-02-04:
Is it possible to display only the received messages with values in the log?

I basically know what I have to do: compare the values from the log with the values in the controller and find and name duplicates.

Unfortunately, my controller does not yet give me values as complex as yours because I have the oldest one from stiebel.

[image]

Hotzenplotz1, 2025-02-04:
Here we are:

[logs_esp32-poe-technik_logs(10).txt](https://github.com/user-attachments/files/18661753/logs_esp32-poe-technik_logs.10.txt)

[WP.pdf](https://github.com/user-attachments/files/18661975/WP.pdf)

kr0ner, 2025-02-04:
@Hotzenplotz1 can you please once compile with VERBOSE loglevel? I just need a small log to see the raw CAN data. Looks like for the old WPs the format is different.

kr0ner, 2025-02-07:
@Hotzenplotz1 I'm working on it. Some messages have the format that I've seen before. Others are completely different. I'm not even sure if there is some randomness in the data 🤷 
I'll try to come up with sth. for you to test and if the weird values still persist you would need to try a different MCP2515. Those sometimes cause problems. I will keep you updated

Hotzenplotz1, 2025-02-07:
Thank you very much!

Do you think it's my can MCP2515 adapter and it's damaged?

kr0ner, 2025-02-07:
I don't know yet :/
there is a certain format of ids (byte 0 and 1) which almost always looks good. But sometimes those ids show up as byte 5 or 6 or at random places ... which looks weird. That would explain why the data is so random if the bytes are shifted. But I'm still investigating. Having a spare MCP2515 is never a bad idea 😉

Hotzenplotz1, 2025-02-09:
New MCP2515's log:

[logs_esp32-poe-technik_logs(12).txt](https://github.com/user-attachments/files/18724917/logs_esp32-poe-technik_logs.12.txt)

kr0ner, 2025-02-16:
sorry @Hotzenplotz1 did not find the time to look at the new logs 🤧

Hotzenplotz1, 2025-02-16:
No Problem. Get well soon!

Hotzenplotz1, 2025-02-24:
How van i fix it?

kr0ner, 2025-02-25:
You would need to try different CAN Bus. Can you take a picture of where the MCP2515 is connected to the heat pump? Cause with the current values we can't work ... there are only like 5 usable values in the logs.

Quotic, 2025-02-25:
what is the red cable which is also connected to CAN High?
Ah you have also the FEK connected?

Hotzenplotz1, 2025-02-25:
Yes, FEK is connected

Hotzenplotz1, 2025-02-25:
FEK was disconnected. Here is the current data log.

[logs_esp32-poe-technik_logs(14).txt](https://github.com/user-attachments/files/18969409/logs_esp32-poe-technik_logs.14.txt)

kr0ner, 2025-02-25:
> FEK was disconnected. Here is the current data log.
> 
> [logs_esp32-poe-technik_logs(14).txt](https://github.com/user-attachments/files/18969409/logs_esp32-poe-technik_logs.14.txt)

Did you now connect the MCP2515 to where the FEK was connected before?
Or did you just remove the FEK? It is possible to use FEK and the MCP at the same time.

Hotzenplotz1, 2025-02-25:
I have 2 possibilities to connect:
1. the mcp directly to the heater or
2. the mcp in series with the fek in the living room.

I had connected the mcp directly to the heating system, but because the wifi was poor here, I connected the mcp in the living room again and disconnected the 4 connections of the fek. the fek is currently dead.

when both devices were connected, we had the measured values that seemed confused to you.

maybe we can write in discord?

Hotzenplotz1, 2025-02-28:
I have adjusted the ANLAGENFROST value and can confirm the address. Does this help us?

[13:47:12][I][Communication:150]: Message received: Read/Write ID 0x60 0x79(0x339) for property ANLAGENFROST (0x0a00) with raw value: 40 --> Anlagenfrost OK

Does a comparison with the index from the ELSTER TABLE help?

kr0ner, 2025-03-05:
> I have adjusted the ANLAGENFROST value and can confirm the address. Does this help us?
>
> [13:47:12][I][Communication:150]: Message received: Read/Write ID 0x60 0x79(0x339) for property ANLAGENFROST [...]

Yes, that is exactly what we need
Can you do this for, let's say 5 values? Change it back and forth, capture the logs along with some pictures if the display where you change values. If the communication works reliably, mapping the entries is not a big deal.

Hotzenplotz1, 2025-03-06:
I have created a list based on the values from the log and the elster table.

I was able to extract the date from the values. 

I also recorded the time value, as this is automatically adjusted.

The question is, how can I adjust the other existing values to the exact time?

Is it possible to display further values that I can check?

[2025-03-06 Werte WPC.xlsx](https://github.com/user-attachments/files/19102732/2025-03-06.Werte.WPC.xlsx)

[2025-03-06 logs_esp32-poe-technik_logs.txt](https://github.com/user-attachments/files/19102736/2025-03-06.logs_esp32-poe-technik_logs.txt)

kr0ner, 2025-03-06:
after looking at the logs and consulting Deepseek I'm like : WTF?
looks like the date is the following
```
1539 -> hex -> 0603 -> 06.03
2053 -> hex -> 0805 -> 08.05
```

Hotzenplotz1, 2025-03-06:
Can I do something else?

kr0ner, 2025-03-06:
Still did not figure out how the time is encoded. There must be another property aside 0x0009 and 0x000a.

Can you take pictures of the display again? Can be the same menu, but i need at least 1 hours of logs

Hotzenplotz1, 2025-03-06:
You Need 1 Hour log?

[22 more replies]

closed 2025-03-20 by Hotzenplotz1