kr0ner/OneESP32ToRuleThemAll #68: Question about new sensors

skycryer, 2024-11-28:
I just saw your update for adding new power sensors.

Wich sensors are now for getting power for 

- fan - sensor.fan_power -> no kWh sensor correct? So we need to create one
- heating - sensor.motorleistung and sensor.motorleistung_kwh?
- warm water - is that included in heating or its there an extra sensor

Sorry for asking but there are several sensor showing energy that can not for like "wärmerückgewinnung". Having clarification wich sensors are showing live power for the parts would be great.

kr0ner, 2024-11-28:
> fan - sensor.fan_power -> no kWh sensor correct? So we need to create one
yes ...just create integral sensor in HA. Not sure it makes sense to do on the esp ... we can leave it as is for now and revisit that topic later

> heating - sensor.motorleistung and sensor.motorleistung_kwh?
Not really ... this is actually anything related to the verdichter (heating, ww, cooling) You can use the values to double check what the heatpump reports.

> warm water - is that included in heating or its there an extra sensor
This you get via EL_AUFNAHMELEISTUNG_WW_SUMME_KWH
The is the value reported by the heatpump ... as we could see in your initial screenshots, the electrical heating (in case of e.g. anti legionellen modus) seems to be excluded ... for this I created ELECTRIC_HEATING_POWER to keep track of that ... could not confirm yet, if this really works

> Sorry for asking but there are several sensor showing energy that can not for like "wärmerückgewinnung". 

What is wrong with that? It is the estimated heat regained from the air outflow ... this is not really a gain or consumption, but rather a "what did I save having the kreuzwärmetauscher"

skycryer, 2024-11-28:
And for sensor.el_aufnahmeleistung_heiz_summe_kwh, i think it does not update correct also. I just saw it stayed at 3kwh most day yesterday but the heating was running. If that is and integral sensor it maybe misses to update automatically even if the using source sensor is not changing?

[image]

kr0ner, 2024-11-28:
> I think there is a problem with the new fan power sensor. The values are updating wrong or to late I think.
> 
> <img alt="Bildschirmfoto 2024-11-28 um 09 59 45" width="2000" src="https://private-user-images.githubusercontent.com/3725699/390712384-52378985-427e-48f0-b2f2-73856350aab3.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzI3ODUxNTgsIm5iZiI6MTczMjc4NDg1OCwicGF0aCI6Ii8zNzI1Njk5LzM5MDcxMjM4NC01MjM3ODk4NS00MjdlLTQ4ZjAtYjJmMi03Mzg1NjM1MGFhYjMucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MTEyOCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDExMjhUMDkwNzM4WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9ZjA1Y2I4ZmZhMzRlYjA3ZWIzNzk5OTg1MDAzYTZkNGU2MTljNzJkZDM3ZmMwN2MwZTg5NDdjMjY3ZjM5ZDhkZSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.ce-llccrUKcMMrgEkP-4ubsK5Kw3aXdSU4BVxbQcE60">
> If I go from full power to 0, fan power should go also to 0 I think. But it goes to something different.

Don't change it so frequently ;) It updates on reception of ZULUFT_SOLL and ABLUFT_SOLL
so latest after 5 minutes it is up to date. If you need it faster you need to set the interval for those to a lower value

kr0ner, 2024-11-28:
> And for sensor.el_aufnahmeleistung_heiz_summe_kwh, i think it does not update correct also. I just saw it stayed at 3kwh most day yesterday but the heating was running. If that is and integral sensor it maybe misses to update automatically even if the using source sensor is not changing?
> 
> <img alt="Bildschirmfoto 2024-11-28 um 10 05 41" width="2000" src="https://private-user-images.githubusercontent.com/3725699/390714738-e5c7bcea-974f-48ee-8a60-52527792efc2.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzI3ODUxNTgsIm5iZiI6MTczMjc4NDg1OCwicGF0aCI6Ii8zNzI1Njk5LzM5MDcxNDczOC1lNWM3YmNlYS05NzRmLTQ4ZWUtOGE2MC01MjUyNzc5MmVmYzIucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MTEyOCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDExMjhUMDkwNzM4WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9ZTA0ZTU5ZmM1YWRkOGJhMzE0MmRmYTIyYTU3ZTBjYzMxZWZiNjZiMmViNjgzMTZhM2I3NWU4NmUwZDRjOTQwYSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.AHde_0os9gSsTlP5tzOaChBijC7Gs_bt00bzxQ22kFQ">

what does the motorleistung entity state for that time? or elektrische nacherwärmung?

skycryer, 2024-11-28:
[image]

kr0ner, 2024-11-28:
so sth. happened here 
[image]

this will be tricky to debug ... in general it seems to work but for whatever reason those high peaks in the morning and in the evening are not counted ... 
can you please provide the value for the elektrische nacherwärmung for that period?

skycryer, 2024-11-28:
you mean the value of this sensor - sensor.electric_heating_power?

There is no data, is that maybe new from the update today?

skycryer, 2024-11-28:
Here are all sensors together, Heizen, Verdichter and Elektrische Nacherwärmung are only binary sensors what I see.

[image]

kr0ner, 2024-11-28:
So at least we know that it did not use electric heating :laughing: 
hm ... but wait ... 
now show me WARMWASSERBEREITUNG
because I bet you have that at around 10 and 20:45

skycryer, 2024-11-28:
Yes I have 
[image]

kr0ner, 2024-11-28:
perfect ... all good :)
Making hot water is also filling the SAMMLER ... 
HEIZEN does not mean "heating up water" but rather circulating the water in the FBH
so it just used the remaining heat to keep the Vorlauftemp and there was no need to produce more hot water. So everything works as expected :)

skycryer, 2024-11-28:
So the energy sensor EL_AUFNAHMELEISTUNG_WW_SUMME_KWH is showing the used energy for warm water, correct?

So i f i want to track the power of my heating pump I need

sensor.motorleistung
sensor.fan_power
sensor.electric_heating_power

and if not available just create integral sensors for tracking the kwh to show them in my energy dashboard, right? Then I should have the ernergy used for heating, warm water and fan. I think then I only need to get the value of the heating pump running itself. Hm maybe I have to add an external sensor here, wanted to avoid that.

skycryer, 2024-11-28:
Ok will will let the sensor run a bit and then check if the collected data do match for me