kr0ner/OneESP32ToRuleThemAll #73: Manual Fanspeed changes to 3 without user input

zimbo86, 2024-12-07:
I noticed that my 504 starts an unscheduled vent stage 3 without any input since the canbus is connected.
I dont have logs atm, but try to get them.
Fan stage day is 1,
Fan stage night is 0.
These settings are not changed.

[image]

zimbo86, 2024-12-07:
okay, i have to correct myself.
It seems that the fan stage itself has changed somehow, so there is no unscheduled vent starting (fan symbol on thz display). It 
The fan speed charts sensor is based on the Hz of fans.

in fhem it looks like this, the lower vcalues (about 15Hz) are normal for fan stage day (1), since i connected the canbus it goes up to 50 hz by default (which leads my sensor to show "3").

[image]

Somehow there is fan stage 3 enabled, because i configured it manually to have underpressure (inlet 300m³, outlet 80m³)
[image]

zimbo86, 2024-12-07:
okay, the fan stage day directly on the heatpump display shows 3, i changed back to 1. Just waiting if this will update the LUEFT_STUFE_TAG control:
[image]

zimbo86, 2024-12-07:
seems fixed. 
Dont know how Stufe was set to 3, however it stays on 1 after setting it to 1.

zimbo86, 2024-12-07:
it seems that when flashing the esp and it boots up the first time, the fan speed day is adjusted to 100% (3).

kr0ner, 2024-12-07:
@zimbo86 I tried reproducing the issue and for me it works as intended. I updated the firmware while the Fan was set to 1 and the fan speed never changed 🤷‍♂️ Can you please paste the logs if/when it occurs again.

kr0ner, 2024-12-08:
Ok, now I got it. If the ventilation is off and you do an update and turn it on it will switch to 3. We can try experimenting with the restore_mode of the fan entity

restore_mode (Optional): Control how the fan attempts to restore state on boot.
```
NO_RESTORE - Don’t restore any state.

RESTORE_DEFAULT_OFF - Attempt to restore state and default to OFF if not possible to restore.

RESTORE_DEFAULT_ON - Attempt to restore state and default to ON.

RESTORE_INVERTED_DEFAULT_OFF - Attempt to restore state inverted from the previous state and default to OFF.

RESTORE_INVERTED_DEFAULT_ON - Attempt to restore state inverted from the previous state and default to ON.

ALWAYS_OFF (Default) - Always initialize the fan as OFF on bootup.

ALWAYS_ON - Always initialize the fan as ON on bootup.
```

zimbo86, 2024-12-09:
nice catch, I will also go a bit deeper and test some scenarios.

It´s pretty hard to control heating, water and fan through a single control each, as when changing the setting it has to pre-check if on-time or off-time is currently active.

kr0ner, 2024-12-09:
For that reason I switched off the day/night handling of the heat pump so that I do not have to decide which knob to turn, depending on the time of day :wink: I just use the _TAG_ variant all the time. An alternative would be to use scheduler-card/component and add e.g. LUEFT_STUFE_TAG and LUEFT_STUFE_NACHT to be set simultaneously. Or create a group for those entities.

zimbo86, 2024-12-10:
> Ok, now I got it. If the ventilation is off and you do an update and turn it on it will switch to 3. We can try experimenting with the restore_mode of the fan entity
>
> ```
> restore_mode (Optional): [...]

in my case its a little bit different:
Fan is on and set to 33%
When i flash ESP, ESP reconnects. Fan is off, HA turns fan on (because it was On before flashing) => 100%

[image]

kr0ner, 2024-12-10:
can you try if ```RESTORE_DEFAULT_OFF``` fixes the problem?

zimbo86, 2024-12-10:
yepp, that worked!
[image]
[image]

closed 2024-12-10 by kr0ner