kr0ner/OneESP32ToRuleThemAll #79: Wrong decimal interpretation in VERDICHTER HEIZLEISTUNG (. and ,)

zimbo86, 2024-12-10:
When VERDICHTER HEIZLEISTUNG contains decimal places, "," and "." are not considered.
A value of 65,529 results in 65529 (65.529)

[image]

kr0ner, 2024-12-10:
That is a different issue. Values are supposed to be between 0 and 100 ... what you see is close to uint16_t::max().

kr0ner, 2024-12-10:
looks like this is happening when hot water is being produced:
[image]

For the rest of the time it is normal. Two ideas how to fix it ..
1, Ignore everything that is not a response .... https://github.com/kr0ner/OneESP32ToRuleThemAll/pull/77/commits/84f42e68dfb4df995257f4da51c60f1c2ed57a6e
2, Limit the value range of the sensor to [0,100] and drop values that are outside that range

I hope that 1, fixes it. Can you try that?

kr0ner, 2024-12-10:
You still use day/night mode, right? Maybe that has an effect :shrug:

zimbo86, 2024-12-10:
> You still use day/night mode, right? Maybe that has an effect 🤷

yes, but day for HC is 10-24:00, Hot Water 10-11:30 and 17:30-20:30

kr0ner, 2024-12-11:
Ok, got it ... it is not "Heizleistung" but "Heizleistung relativ" ... which already states that values can be negative, too.
Fix therefore is:
[image]

zimbo86, 2024-12-11:
perfect!
as value turns positiv at 18:00 Uhr this matches exactly.
[image]

closed 2024-12-11 by kr0ner

zimbo86, 2024-12-12:
works like a charm
[image]