kr0ner/OneESP32ToRuleThemAll #81: FUSSPUNKT and STEIGUNG / added parameters, must be pulled manually

zimbo86, 2024-12-10:
I added FUSSPUNKT and STEIGUNG HK1, however these parameters wont pull automatically and need to be pulled

`  
STEIGUNG_HK1:                   !include { file: wp_number.yaml, vars: { property: "STEIGUNG_HK1"               , min: "0", max: "5.0", step: "0.01", icon: "mdi:chart-bell-curve-cumulative"}}
FUSSPUNKT_HK1:                  !include { file: wp_number.yaml, vars: { property: "FUSSPUNKT_HK1"               , min: "0", max: "20", step: "0.1", icon: "mdi:download-outline"}}`

`
PROPERTY(FUSSPUNKT_HK1, 0x059e, Type::et_dec_val);
PROPERTY(STEIGUNG_HK1, 0x010e, Type::et_dec_val);
`

kr0ner, 2024-12-10:
They are related to HK1 so they should also have ```target: "HK1"```.
Number values are requested once at boot and after that, since they need to actively be changed, on reception. Maybe that is related to 
```
if (!isResponse(msg))
```
and that was broken recently ... in this case the following should fix it
```
    if (isRequest(msg)) {
        return {Property::kINDEX_NOT_FOUND, value};
    }
```

kr0ner, 2024-12-10:
Did you add HK1 as target?
```
STEIGUNG_HK1:                   !include { file: wp_number.yaml, vars: { property: "STEIGUNG_HK1"               , min: "0", max: "5.0", step: "0.01", icon: "mdi:chart-bell-curve-cumulative", target: "HK1"}}
FUSSPUNKT_HK1:                  !include { file: wp_number.yaml, vars: { property: "FUSSPUNKT_HK1"               , min: "0", max: "20", step: "0.1", icon: "mdi:download-outline", target: "HK1"}}
```

zimbo86, 2024-12-10:
nope, I didnt. Your response time is pure madness :D

zimbo86, 2024-12-10:
that was the missing point, works!

[image]
(I use differnet names)

kr0ner, 2024-12-10:
try ```Type::et_cent_val``` instead of ```Type::et_dec_val```
```
PROPERTY(STEIGUNG_HK1, 0x010e, Type::et_cent_val);
```

zimbo86, 2024-12-10:
that worked!

[image]

closed 2024-12-10 by zimbo86