kr0ner/OneESP32ToRuleThemAll #83: WPF 7 Cool

Icare2000, 2024-12-17:
Hi

Thanks for your great job, I am able to connect but if have some errors, with unknown can's

I have setup ttf07.yaml as it seems more related to my equipment which is a geothermal equipment.

substitutions:
  kessel_can_id: "0x180"
  hk1_can_id: "0x301"
  hk2_can_id: "0x302"
  manager_can_id: "0x6a1"
  espclient_can_id: "0x6a2"

I have a can id which is unknown, 0x700, 

and i have several errors for abnormal values.
[21:24:49][D][canbus:072]: received can message (#1) std can_id=0x301 size=7
[21:24:49][D][CAN:177]: Can message received with CANId 0x0301
[21:24:49][I][Communication:109]: Message received: Read/Write ID 0xc0 0x01(0x0601) for property VORLAUFSOLLTEMP (0x0004) with raw value: 303
[21:24:49][D][Communication:111]: Message is not a response. Dropping it!
[21:24:49][D][canbus:072]: received can message (#1) std can_id=0x100 size=7
[21:24:49][D][CAN:177]: Can message received with CANId 0x0100
[21:24:49][I][Communication:109]: Message received: Read/Write ID 0x31 0x00(0x0180) for property AUSSENTEMP (0x000c) with raw value: 0
[21:24:49][D][Communication:111]: Message is not a response. Dropping it!
[21:24:49][D][CAN:184]: No CANMember with CANId 0x0100 available.
[21:24:50][D][canbus:072]: received can message (#1) std can_id=0x480 size=7
[21:24:50][D][CAN:177]: Can message received with CANId 0x0480
[21:24:50][I][Communication:109]: Message received: Read/Write ID 0xe1 0x00(0x0700) for property UNKNOWN (0xfdf4) with raw value: 0
[21:24:50][D][Communication:111]: Message is not a response. Dropping it!
[21:24:50][D][CAN:184]: No CANMember with CANId 0x0480 available.
[21:24:50][D][canbus:072]: received can message (#2) std can_id=0x700 size=7
[21:24:50][D][CAN:177]: Can message received with CANId 0x0700
[21:24:50][I][Communication:109]: Message received: Read/Write ID 0x92 0x00(0x0480) for property UNKNOWN (0xfdf4) with raw value: 274
[21:24:50][D][Communication:111]: Message is not a response. Dropping it!
[21:24:50][D][CAN:184]: No CANMember with CANId 0x0700 available.
[21:24:50][D][canbus:072]: received can message (#3) std can_id=0x700 size=7
[21:24:50][D][CAN:177]: Can message received with CANId 0x0700

If you want we can get in touch with "discord"

Best regards

Icare

mkaiser, 2024-12-17:
Hi,

I have a TTF07cool, which is kind of identical to yours. 

Do you have the FEK installed? Maybe this is the Adress 0x700.

Icare2000, 2024-12-17:
Hi

Yes I have a fek ( is it the external remote controller )

If
Yes what should we do in the configuration

Should we create another specific template ?

Philippe Pache

kr0ner, 2024-12-18:
@Icare2000 I would not focus too much on that ID right now :wink: Even if it is the FEK you can't do anything with that info. It reports room temp and humidity .. but that one you can also get from the heat pump.
During christmas time I will focus on porting @mkaiser changes

mkaiser, 2024-12-18:
I am still in an middle stage of fully understanding my heatpump :)

- When the FEK is attached, it controls the heating (curve, setpoints etc.) and it seems that there is no way to "remote control" the FEK itsetlf. 
- I can only change the running mode / Betriebszustand (eco, comfort, standby...) of the heatpump, when the FEK is NOT attached. 
- For TTF*/WPC* the room temp and humdity are meaasured by the FEK and periodically transmitted to the heatpump manager. If the values are missing, the heatpump will not allow the "cooling" feature

--> Bottom line: If you want to test stuff, remove the FEK first. 

> During christmas time I will focus on porting @mkaiser changes

sorry, I did not have any spare time in the last weeks. I still have locally multiple uncommitted changes. But I need to merge with the current main branch first. Maybe it is the easiest, if you wait some time until I finally made my homework :)

Icare2000, 2024-12-18:
> @Icare2000 I would not focus too much on that ID right now 😉 Even if it is the FEK you can't do anything with that info. It reports room temp and humidity .. but that one you can also get from the heat pump. During christmas time I will focus on porting @mkaiser changes

Hi , perfect, I agree that if the FEK is just a sensor providing info to the Heatpump it can be "ignored". Then why not adding in the code something to ignore all message from this equipment.

Icare2000, 2024-12-18:
> I am still in an middle stage of fully understanding my heatpump :)
>
> * When the FEK is attached, it controls the heating (curve, setpoints etc.) and it seems that there is no way to "remote control" [...]

Do you mean remove, is physically disconnect the equipment ?

kr0ner, 2024-12-18:
> Hi , perfect, I agree that if the FEK is just a sensor providing info to the Heatpump it can be "ignored". Then why not adding in the code something to ignore all message from this equipment.

In the end that's what is happening. There is no callback for 0x700 so the result is going to the void. Filtering the CAN messages earlier might be a thing, but I need to investigate how to couple this with the debug logs. Cause for new devices and tracing bugs this info is very helpful.

mkaiser, 2024-12-18:
> Do you mean remove, is physically disconnect the equipment ?

Yes, physically remove it :)

Icare2000, 2024-12-19:
Hi, Another question, Because i have the FEK, finally the Fek Blocked, and i had no heating at all during my workday. Can i Comment out this ?

[image]

in order not to send data which i do not have

What is your advice?

kr0ner, 2024-12-20:
Sorry to hear that 🥴
Yes, you can comment that part out. Those lines are basically replacing the FEK in my case. I calculate the AVG temperature and humidity in the house and send it to the heatpump. I'd suggest to also go with this solution cause you have the whole house as an input, not just one spot in one room

mkaiser, 2025-01-06:
@Icare2000 

As we have basically the same heatpump (Tecalor vs. Stiebel Eltron): Could you test, if my branch is working?
https://github.com/mkaiser/OneESP32ToRuleThemAll

kr0ner, 2025-01-22:
@Icare2000 the backport is ready to be tested https://github.com/kr0ner/OneESP32ToRuleThemAll/pull/93

Icare2000, 2025-02-09:
I tested!

kr0ner, 2025-02-16:
> I tested!

any findings? Or does it work for you as expected?

Icare2000, 2025-02-16:
@kr0ner  no still problems

[image]

[image]

kr0ner, 2025-03-26:
@Icare2000 can you provide some logs that capture like 2 hours? Looking at the logs from @mkaiser I can't see this behaviour https://github.com/kr0ner/OneESP32ToRuleThemAll/pull/144#issuecomment-2750426665

closed 2025-05-03 by github-actions[bot]