kr0ner/OneESP32ToRuleThemAll #84: Request for Guidance on Decoding CAN IDs and Values for THZ 5.5 ECO

eugenpirogoff, 2025-01-10:
Hi folks,

Thank you for the great project and all the hard work you've put into it! I'm currently working on a setup involving:

HomeAssistant with ESPHome
ESP32 with MCP2515
THZ 5.5 ECO attached to the CAN bus
Using the THZ 5.5 YAML configuration

I’ve commented out all other sensors and GPIO-attached devices to focus on the CAN communication. Many values appear correct, but others seem off. From what I understand, we may need to extend the mapping specific to the THZ 5.5 ECO in [property.h](https://github.com/kr0ner/OneESP32ToRuleThemAll/blob/master/src/property.h) to ensure proper functionality.

However, I'm struggling to figure out:

How to check the CAN IDs effectively: What’s the best way to identify which CAN IDs are relevant for decoding?
How to decode values and their encoding: Could you explain what to look for in the logs and how to determine the correct mappings for property.h?

Your detailed guidance or any resources on how to approach this would be greatly appreciated!

Thank you again for your fantastic work and for making this accessible to the community.

kr0ner, 2025-01-10:
> From what I understand, we may need to extend the mapping specific to the THZ 5.5 ECO in [property.h](https://github.com/kr0ner/OneESP32ToRuleThemAll/blob/master/src/property.h) to ensure proper functionality.

Yes, this might be required. Can you please provide some logs and screenshots so that it is easier to pin down the problems.

In the end adding / updating properties is trial and error. My approach usually is to take my laptop/tablet, go to the ESPHome dashboard and display the logs of the ESP. After that I navigate through the heatpump menu to the property I want to add or change and then try to match the value I see on the display to the already known IDs or look for UNKNOWN.

[image]

I hope this helps.

eugenpirogoff, 2025-01-10:
Sure thing. 

Some values are broken very subtly. Like "Verdichter Hz" and the "Verdichter"-Binary Sensor do not match. "WARMWASSERBEREITUNG" never turns on. And so on. 

Since fixing it, is too much to ask for. I ask for guidance. 

[logs_esp32-can_logs.txt](https://github.com/user-attachments/files/18375775/logs_esp32-can_logs.txt)

[image]
[image]

Additionally can [we refactor the those two HA sensors](https://github.com/kr0ner/OneESP32ToRuleThemAll/blob/master/yaml/common.yaml#L119) not to be mandatory to compile the integration?

kr0ner, 2025-01-10:
> Like "Verdichter Hz" and the "Verdichter"-Binary Sensor 

VERDICHTER and WARMWASSERBEREITUNG might be related, because they are different bit values in the BETRIEBSSTATUS
Can you plot these values together over the last day or so? 

> Additionally can [we refactor the those two HA sensors](https://github.com/kr0ner/OneESP32ToRuleThemAll/blob/master/yaml/common.yaml#L119) not to be mandatory to compile the integration?

Sure we can do that ... although I think everyone should use it over the FEK :D

But overall does not look too bad :sweat_smile:

kr0ner, 2025-01-10:
I'd need logs during the time when hot water production is active to see the value of the BETRIEBSSTATUS and the same for when VERDICHTER is active. Even better would be logs + picture of the display that shows the symbols in home screen

eugenpirogoff, 2025-01-10:
Here are some additional logs of the last hour. 
[logs_esp32-can_logs-2.txt](https://github.com/user-attachments/files/18376571/logs_esp32-can_logs-2.txt)

And as an explanation to my configuration of the heatpump. It's configured to run smoothly with low electrical power consumption (about 40% now for heating, and more for warm water, so its about 700 watts and sometimes off) and about 5 to 20 compressor starts during the day. It once does heat up water in between from 24 to 72 hours. Depending on how much we use it and how cold it's. So this means, VORLAUF temp is usually low and only goes up for hot water.

I will provide the logs you need. But it will take some time. 

Thank you in advance.

kr0ner, 2025-01-10:
So from the logs VERDICHTER seems to work fine
```
[12:16:14][I][Communication:109]: Message received: Read/Write ID 0xd2 0x22(0x06a2) for property MOTORLEISTUNG (0x06a0) with raw value: 87
[12:16:35][I][Communication:109]: Message received: Read/Write ID 0xd0 0x3c(0x06bc) for property BETRIEBS_STATUS (0x0176) with raw value: 1031
```
[image]
lowest 3 bits are Schaltprogramm_Aktiv, Verdichter, Heizkreispumpe

> about 40% now for heating, and more for warm water, so its about 700 watts and sometimes off)

Same for me ... you can reduce PUMPENDREHZAHL_HEIZEN (65% for me) to reduce the amount of VERDICHTERSTARTS

Why is your SPEICHERISTTEMP so high ? :scream: 
> SPEICHERISTTEMP (0x000e) with raw value: 521
> It once does heat up water in between from 24 to 72 hours.

I think it would be better to lower the temp and to heat up the water more often but only to around 45°. This is what i got from the forums

eugenpirogoff, 2025-01-10:
> ```
> [12:16:14][I][Communication:109]: Message received: Read/Write ID 0xd2 0x22(0x06a2) for property MOTORLEISTUNG (0x06a0) with raw value: 87
> [12:16:35][I][Communication:109]: Message received: [...]

Where do you have the binary representation from ? 

> 
> Same for me ... you can reduce PUMPENDREHZAHL_HEIZEN (65% for me) to reduce the amount of VERDICHTERSTARTS

Because i am not perfect ;-D and thank you for the hint. I just did the reduction to 70 (visible in the attached logs). 

[logs_esp32-can_logs-4.txt](https://github.com/user-attachments/files/18376903/logs_esp32-can_logs-4.txt)

> Why is your SPEICHERISTTEMP so high ? 😱

Because my kids take a bath from time to time and if i can not trigger this from the phone in a very short amount of time before, or configure a different max temp for the weekend (not sure if this is possible on the device itself, maybe it is) then its high as this. Also if i set it much lower, then it will not be enough to fill the bathtub with the correct temperature.  

> I think it would be better to lower the temp and to heat up the water more often but only to around 45°. This is what i got from the forums

I am sure this is correct. Efficiency sinks with the temperature.

Do you think i should resolder the ping that esphome is complainig about(strapping pin) and is the "lag"(reponse time of can component) related to this ?

kr0ner, 2025-01-10:
> Where do you have the values and binary representation from ?

The initial mapping I got from a Modbus pdf from Stiebel Eltron ^^ the rest was a bit trial and error and matching the pictures on the display to the correct bit

> Because i am not perfect ;-D and thank you for the hint. I just did the reduction to 70 (visible in the attached logs).

Me neither ^^... it takes time and constantly monitoring the graphs to tweak it. I usually only play with the HEIZLEISTUNG between 35% (+10°) (won't accept lower values) and 70% (-10°) ... most probably my Heating curve is wrong and that is why I need it. 

> Because my kids take a bath from time to time and if i can not trigger this from the phone in a very short amount of time before, or configure a different max temp for the weekend (not sure if this [...]

I'd recommend to use https://github.com/nielsfaber/scheduler-card und scheduler component ;) Via HA it is so easy an convenient to set schedules this way
[image]

> Also if i set it much lower, then it will not be enough to fill the bathtub with the correct temperature.

Are you trying to cook them? :D For me it is set so 43° and it is enough for a 180 x 80 bathtub to be filled once with 39°. I just have to crank it all the way open and after that the buffer is almost down to 30°

eugenpirogoff, 2025-01-10:
> > Where do you have the values and binary representation from ?
> 
> The initial mapping I got from a Modbus pdf from Stiebel Eltron ^^ the rest was a bit trial and error and matching the pictures on the display to the correct bit

Speaking of mapping: there is this other person on HA forum [that dropped this](https://community.home-assistant.io/t/integration-of-heat-pump-thz-5-5-eco-from-tecalor-stiebel-eltron-into-home-assistant/447298/18). It MIGHT be helpful. 

> Me neither ^^... it takes time and constantly monitoring the graphs to tweak it. I usually only play with the HEIZLEISTUNG between 35% (+10°) (won't accept lower values) and 70% (-10°) ... most [...]

Same here. I have no clue what all the details in there are. Sadly it was already broken multiple times. So i am a real pro disassembling it by now. (Did you know there is a filter behind the heat exchanger in the back of the device?)
[image]

> I'd recommend to use https://github.com/nielsfaber/scheduler-card und scheduler component ;) Via HA it is so easy an convenient to set schedules this way [...]

Good point: i will adjust this and also automate things, once i have gained trust that all of this ESP32/CAN/HA-Stuff really works.

kr0ner, 2025-01-10:
> Speaking of mapping: there is this other person on HA forum [that dropped this](https://community.home-assistant.io/t/integration-of-heat-pump-thz-5-5-eco-from-tecalor-stiebel-eltron-into-home-assistant/447298/18). It MIGHT be helpful.

Nice, that might be really helpful. Also left a message with regards to the room sensor.

> Same here. I have no clue what all the details in there are. Sadly it was already broken multiple times. So i am a real pro disassembling it by now. (Did you know there is a filter behind the heat exchanger in the back of the device?)

Yes ... the Stiebel Eltron support cut a hole in the back to be able to access it :cry: 

> Good point: i will adjust this and also automate things, once i have gained trust that all of this ESP32/CAN/HA-Stuff really works.

If you don't trust it (yet) just use it for monitoring and disable all the entities that can set stuff :wink: 
Logging and tracing the values over time enables you to debug problems better than the Tecalor Kundendienst

eugenpirogoff, 2025-01-10:
> Nice, that might be really helpful. Also left a message with regards to the room sensor.
👍 

> Yes ... the Stiebel Eltron support cut a hole in the back to be able to access it 😢
:-( 

> If you don't trust it (yet) just use it for monitoring and disable all the entities that can set stuff 😉 Logging and tracing the values over time enables you to debug problems better than the Tecalor Kundendienst
That is how i realised that it was even broken the last two times and could make a diagnosis. I had SOMETHING running (i think it was a earlier version of this repo. not sure), but it was not sending anything back, i removed the sending capability back then.

Will post logs next week.

eugenpirogoff, 2025-01-13:
Here are some logs and screenshots. 
[logs_esp32-can_logs-6.txt](https://github.com/user-attachments/files/18393688/logs_esp32-can_logs-6.txt)

[image]
[image]
[image]

The WARMWASSERBEREITUNG and VERDICHTER binary sensor where never turned on at all.

kr0ner, 2025-01-13:
Going to have a look at this today.

Exezz78, 2025-01-19:
Thanks a lot @eugenpirogoff  and @kr0ner. Really interested in this improvement, too as I run the exact same setup.

eugenpirogoff, 2025-01-19:
One interesting feature could be to check if the Hz of Abluft and Zuluft stay in relation to each other. 

Currently Zuluft has an ice crust on it. And could have a "Sensor in HA" for this case that looks at the Hz of the ventilation. 

[image]
[image]

[image]
[image]

eugenpirogoff, 2025-01-19:
After removing the ice. 

[image]

[image]

kr0ner, 2025-01-19:
That i'm going to check right now 😅 btw... I was looking at the logs and at least the Verdichter sensor should be working :/ can you check the value of EVU Sperre binary sensor while hot water is made?

eugenpirogoff, 2025-01-19:
I guess it was my fault here. But i am not sure. 
I have commented out this config from the default yaml:

```
#########################################
#                                       #
#   EVU Sperre                          #
#                                       #
#########################################
switch:
  - platform: gpio
    name: "EVU Sperre"
    id: evu_sperre
    pin: GPIO5
```

because it was interfering with my MCP2515 GPIO pin i have used. And by looking at it i thought that it was used to signed EVU to the outer world. So commented it out.

Placing different GPIO pin now and recompiling.

kr0ner, 2025-01-19:
No need to do that ;) I intended to use that to hard switch off the heat pump when e.g. there is water in the floor. But that one is to set the EVU Sperre ... The binary sensor is bit 10 of Betriebsstatus. Which seems to be on sometimes from what I see in the logs. Maybe the bits are used differently. Can you show me the binary sensor EVU sperre logs from the last 2 days and the speicheristtemp?

eugenpirogoff, 2025-01-19:
Sadly not the real log. 

[image]
[image]

Speaking of logs: is there a way to save the stream of logs from ESPHome to a file in HA? I currently need to have a browser window open and then copy paste it.

eugenpirogoff, 2025-01-19:
It was the master state from end of last year. Now it's the latest change from January. 

It's slightly inconvenient for me, to manually apply git stashes in a git submodule (because of the sensors in common.yaml that need adjustment). :-D Sorry for the complaint. 

So let's see how it will look tomorrow. 

The download button in esphome is is just giving you what you see on the screen, because the ESP32 has not enough space for the actual log. It would be beneficial to be able to stream them to HA. But that is a idea for a project on it's own.

eugenpirogoff, 2025-01-20:
Hey folks. It all magically works now. 

EVU Sperre, Verdichter Binary Sensor. 

Thank you for your patience. 

[image]

[image]

----

Even the error code is there because I set the warm temperature too high (was playing with the interface) and it tried to make very hot water all in the past day. All day long, with a limited power. 

[image]

[image]

[image]

So thank you very very much @kr0ner

kr0ner, 2025-01-20:
I think I screwed up the communication and fixed it here https://github.com/kr0ner/OneESP32ToRuleThemAll/pull/88
Sorry 🫣 It was not very obvious that it broke sth. 

But good to know that everything works for you now 🙂 Have fun and feel free to report bugs and request new Features. Closing this now

closed 2025-01-20 by kr0ner