kr0ner/OneESP32ToRuleThemAll #89: hot_tolerance not set but to high?

skycryer, 2025-01-17:
Hello, I saw that the heating is going way higher than what I set. I use a sensor for the average room temperature and thought the heating should go off when it reaches the temp.

Like when I set heating day to 22.0 it heats up to 22.8, when I check the documentation for generic thermostat config there is an option for hot_tolerance and cold tolerance to control when the heating goes on or off. But I can not see where this value is used in your config file (common.yaml or any other file), so how is that set? Or is it not set but Home assistant is using the thermostat wrong and we should just add a config for this values?

kr0ner, 2025-01-17:
You need to be aware that the thermostat is only an UI feature to set the preferred room/hot water temp. The actual controlling still happens in the heat pump. Even setting the room factor to 100% will not stop the heat pump from heating until it reached it's calculated value. I also can't find the formula again ... it basically just takes the room factor into the heating curve calculation. So to fix the problem you need to tweak your heating curve

kr0ner, 2025-01-17:
[image]

skycryer, 2025-01-17:
Good to got that clarified, thank you.

closed 2025-01-17 by skycryer