kr0ner/OneESP32ToRuleThemAll #100: VORLAUFISTTEMP broken for TTF07

mkaiser, 2025-01-24:
common.yaml: 
current:
`  VORLAUFISTTEMP:                 !include { file: wp_temperature.yaml, vars: { property: "VORLAUFISTTEMP"           , update_interval: $interval_medium, target: "HK1" }}
`

works on TTF07c:
`  VORLAUFISTTEMP:                 !include { file: wp_temperature.yaml, vars: { property: "VORLAUFISTTEMP"           , update_interval: $interval_medium }}
`

In wp_temperature.yaml, the default target is "Kessel". 

So in the TTF07 heatpump, the value is provided by Kessel and in the other heatpumps by HK1?

Any ideas how to deal with such a minor difference?

kr0ner, 2025-01-24:
I had similar problems with the WPL13 and ended up duplicating the sensor ... which I did not like ... but
I cannot use https://esphome.io/components/packages.html#extend because my packages contain way more stuff and cannot be addressed by an ID. Substitutions would affect all sensors :/

But since the target is only used in C++ code ... maybe it makes sense to map property <-> target somewhere else and not directly in the sensors. The mapping could be generic in the first place again and customized via the defines.

For now I'd just create a duplicate ... not the best option ... but works and is verbose 🤷

mkaiser, 2025-01-28:
Just another thought: 

Is there a reason (except work) to not put the propery-targets into property.h?

 e.g., 

PROPERTY(SPEICHERISTTEMP, 0x000e, Type::et_dec_val);
--> 
PROPERTY(SPEICHERISTTEMP, A_CAN_ID, 0x000e, Type::et_dec_val);

kr0ner, 2025-01-31:
The problem is that properties can be assigned to multiple targets :-/
I did not really figure out a nice way how to express this yet. An override on yaml level would be nice, but that only affects the esphome part and does not change anything on c++ level. Defines would be to hard to maintain and ugly to override. So in the end i think it is the easiest to just duplicate it

kr0ner, 2025-02-01:
@mkaiser  is VORLAUFISTTEMP the only one that needs to be fixed?

what about
  RAUMISTTEMP
  VERSTELLTE_RAUMSOLLTEMP
  VORLAUFSOLLTEMP

mkaiser, 2025-02-02:
attached the FEK again: 

**RAUMFEUCHTE**  39,3% (works as expected)

**RAUMISTTEMP** 22,7 °C toggles between 0 and the correct value: 
![Image](https://github.com/user-attachments/assets/c7aedb51-5851-4225-bf45-c233d73f2e42

**VORLAUFSOLLTEMP** toggles, too: 
[image]

**VERSTELLTE_RAUMSOLLTEMP** is quiet interesting, too: 

[image]

**All other values** show reasonable values

kr0ner, 2025-02-02:
@mkaiser did you remove the homeassistant sensors from the yamls?

common.yaml
```
  - interval: $interval_medium
    then:
      - lambda: |-
          // Send current RAUMISTTEMP
          const auto room_temp = id(gRAUMISTTEMP);
          if(room_temp > 0.0f) {
            ESP_LOGI("SET", "Sending cached room temperature %f to heatpump", room_temp);
            sendData(HK1, Property::kRAUMISTTEMP, static_cast<std::uint16_t>(room_temp * 10.0f));
          }
          // Send current RAUMFEUCHTE
          const auto humidity = id(gRAUMFEUCHTE);
          if(humidity > 0.0) {
            ESP_LOGI("SET", "Sending cached humidity %f to heatpump", humidity);
            sendData(HK1, Property::kRAUMFEUCHTE, static_cast<std::uint16_t>(humidity * 10.0f));
          }

......

#########################################
#                                       #
#   Home Assistant Sensors              #
#                                       #
#########################################
sensor:
  - platform: homeassistant
    name: "Temperature Sensor From Home Assistant"
    entity_id: $entity_room_temperature
    on_value:
      then:
        - lambda: |-
            ESP_LOGI("SET", "Updating room temperature to %f", x);
            id(gRAUMISTTEMP) = x;

  - platform: homeassistant
    name: "Humidity Sensor From Home Assistant"
    entity_id: $entity_humidity
    on_value:
      then:
        - lambda: |-
            ESP_LOGI("SET", "Updating humidity to %f", x);
            id(gRAUMFEUCHTE) = x;
```

kr0ner, 2025-02-02:
Not sure what exactly is happening ... but looks like the FEK competes against the ESP. That would not be good.
If just the ESP is confused and can't decide which value to display that would only be a minor issue
Can you check if those settings also toggle in the display of the heatpump?

Verstellte Raumsolltemp is the current setting the heat pump is trying to apply ... that one should not toggle ... heatpump might go on-off-on-off all day long

Icare2000, 2025-02-09:
I have the same issue with your latest code. I still have the FEK connected, but i have Can bus in ListenOnly 

[image]

i also removed the sensor & interval

to know, we have no hk2, shall we perhaps add a specific config for wpf07

kr0ner, 2025-02-16:
> I have the same issue with your latest code. I still have the FEK connected, but i have Can bus in ListenOnly
> 
> [image]
> 
> i also removed the sensor & interval
> 
> to know, we have no hk2, shall we perhaps add a specific config for wpf07

But that is really weird. Since no messages are sent, the toggling behaviour is caused by the heatpump and the FEK itself. Can any of you provide some detailled logs?

Icare2000, 2025-02-16:
Hi, yes, but you can also contact me on the discord ( Faravid)

this one week in listenmode... i have Zero outage..

kr0ner, 2025-02-17:
Ok, so you set it to listenonly and that fixed the toggling behaviour?

Icare2000, 2025-02-17:
> Ok, so you set it to listenonly and that fixed the toggling behaviour?

No it's still showing  data sometime with zero's

[image]

Icare2000, 2025-02-17:
> This is really weird. The only thing I can think of is that in your case the Request/Response/Update flags are different and that the implementation picks up the 0 value from the request. But without detailled logs I cannot look further into the topic

here is a file

Icare2000, 2025-02-17:
[logs_esphome-wth_logs.txt](https://github.com/user-attachments/files/18831322/logs_esphome-wth_logs.txt)

here is a file

kr0ner, 2025-02-28:
> [logs_esphome-wth_logs.txt](https://github.com/user-attachments/files/18831322/logs_esphome-wth_logs.txt)
> 
> here is a file

Sorry for the late reply :-/
The logs are too short. I can only see some good values for both VorlaufIST and SOLL.
Could you run it once again for and hour or two? Thanks.

closed 2025-08-21 by kr0ner

mkaiser, 2025-12-23:
Solved https://github.com/mkaiser/OneESP32ToRuleThemAll/commit/d877f9a33e7950c2d72f45bb4b57a7f1c088c158

upstream discussion here https://github.com/kr0ner/OneESP32ToRuleThemAll/issues/232

PR #190 (2025-08-21): Directly obtain message type from read/write/response id
Previously this was done by looking up the the ids of the existing CanMembers, which might lead to problems when an FEK is connected, because that one uses 0x1c01 instead of 0x8001 to request data from 0x601. Both are valid, but the code only handled 0x8001 correctly.

Fixes: #100, #188
commit 0c45809e (2025-08-21): Directly obtain message type from read/write/response id

Previously this was done by looking up the the ids of the existing
CanMembers, which might lead to problems when an FEK is connected,
because that one uses 0x1c01 instead of 0x8001 to request data from
0x601. Both are valid, but the code only handled 0x8001 correctly.

Fixes: #100, #188