kr0ner/OneESP32ToRuleThemAll #104: Wpl13 - WPE- HK: PROGRAMMSCHALTER not showing current state

adrai, 2025-01-25:
PROGRAMMSCHALTER does not show the current state => i changed multiple times the program manually, Komfort, Eco, Program

[image]

[logs and other infos](https://github.com/kr0ner/OneESP32ToRuleThemAll/issues/101)

kr0ner, 2025-02-01:
TBH from what is currently implemeted it should work 🤷 
I added a log so that we can see, if the mapping works.

adrai, 2025-02-03:
Changed from Programmbetrieb to Komfortbetrieb and back again:

[image]

[new_logs.txt](https://github.com/user-attachments/files/18648197/new_logs.txt)

adrai, 2025-02-11:
some additional infos....
- Changing the Programmschalter directly on the heating pump **is not** reflected in the UI (PROGRAMMSCHALTER state is Unbekannt)
- Changing the Programmschalter in the UI **is** reflected in the UI (PROGRAMMSCHALTER state is changed and keeps there until changed again via UI) but **is not** reflected on the heating pump display
- Changing the Programmschalter directly on the heating pump again **is still not** reflected in the UI (PROGRAMMSCHALTER state keeps the last set state via UI)

kr0ner, 2025-02-13:
I've added some logs to make it easier to identify the issue

adrai, 2025-02-13:
still the same...
changed to komfort mode directly on the heating pump
then changed to eco from the web ui
and then changed back to programm mode directly on the heating pump

[new_logs.txt](https://github.com/user-attachments/files/18788873/new_logs.txt)

kr0ner, 2025-02-13:
There was no functional change, because all I can do would be wild guessing. I need to look at the new logs and see if it makes more sense now

PR #124 (2025-02-13): Wpl13 improvements
Resolves: #102
Resolves: #103
Resolves: #104
closed 2025-02-17 by kr0ner