kr0ner/OneESP32ToRuleThemAll #117: support LWZ403_sol

Dobbiiey, 2025-01-25:
Hi krOner,
my heatpump is a Stiebel LWZ403sol. I'm currently migrate/copy my working ioBroker Implementation to ESP32 HomeAssistant.
Your Repo seems to be perfect to add also this old heatpump ;-).

I have installed an esp32-s3 with "platform: esp32_can" and I'm connected to the heatpump. (tried it with simple code -> receiving & sending works fine)

Now I tried lot things with your Repo and commented out lots but I can't get it installed correctly ...
for example: ethernet -> network and time Platform from sntp to host

Do you have an idea about my wrong config from this log ?

thanks
Tobi

`
INFO ESPHome 2024.12.4
INFO Reading configuration /config/esphome/esphome-stiebel.yaml...
INFO Detected timezone 'Europe/Berlin'
INFO Detected timezone 'Europe/Berlin'
INFO Detected timezone 'Europe/Berlin'
INFO Detected timezone 'Europe/Berlin'
INFO Detected timezone 'Europe/Berlin'
Traceback (most recent call last):
  File "/usr/local/bin/esphome", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/esphome/esphome/__main__.py", line 1036, in main
    return run_esphome(sys.argv)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 1023, in run_esphome
    rc = POST_CONFIG_ACTIONS[args.command](args, config)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 499, in command_logs
    return show_logs(config, args, port)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 390, in show_logs
    if get_port_type(port) == "SERIAL":
       ^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 128, in get_port_type
    if port.startswith("/") or port.startswith("COM"):
       ^^^^^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'startswith'
`

kr0ner, 2025-01-25:
Hi Tobi, can you please paste the whole config? The internal ESPHome logs could be anything ^^

Dobbiiey, 2025-01-25:
sorry ;-)

[esphome-stiebel.txt](https://github.com/user-attachments/files/18548291/esphome-stiebel.txt)

[lwz403.txt](https://github.com/user-attachments/files/18548292/lwz403.txt)
[common.txt](https://github.com/user-attachments/files/18548293/common.txt)

kr0ner, 2025-01-27:
I'd recommend to start with the thz504.yaml and follow the instructions in the README.md.
Let it connect to WiFi and if that works, strip down the yamls to match you expectations.

Any reason why you want to start from scratch?

Dobbiiey, 2025-01-27:
I tried it in this way but failed to compile and/or installation on the ESP32...
So I thought "strip down helps to find the problem" ;-)
Anyhow I found out, that I crashed complete ESPHome Addon in some cases ;-)...  Reinstalling the Addon helps - after it compiling works sometimes again...

So - in short words:  I will try again & hope that missing stuff like Sensors from HASS, KUEHLEN or MOTORSTROM will not cause issues...

Thanks  & I will let you know if there is progress ;-)

kr0ner, 2025-01-27:
How much RAM your machine has? I know that some people had problems with 4GB. I had 8, now 16 and never had issues. Did you comment out the packages in the top yaml? So that only thz504.yaml is still there? You can also add this option to esphome:

compile_process_limit (Optional, int): The maximum number of simultaneous compile processes to run. Defaults to the number of cores of the CPU which is also the maximum you can set.

Dobbiiey, 2025-01-27:
cool - it's done - compiled & lot is working out of the box - great!
Fan (day&night) / Clima (WarmWater-Reservoir) / display Interpretation (0x0176) / numbers (RAUMEINFLUSS) / lot of sensors...

@kr0ner Great Job! -> impressiv
 So I now think, it is easy to me adding the missing IDs and delete the not relevant things...

Maybe 2 naive questions:
- Do you get the info directly, when someone changes values at the standard display ? (this does currently not work...) or do you ignore such CAN messages? (maybe I misunderstand the manager_can_id -> should it be the display?)
- common.yaml line 82  interval: 250ms:
 Is this the general min. time sending messages out of the complete Repo ?
(I know my heatpump is slow & so I did it with 1s with ioBroker technic without issues ;-))

kr0ner, 2025-01-27:
Great News 😀 
Have fun. TBH I would recommend to keep the sensors and to just disable them in the HA UI. Will be easier to keep to codebase up to date since lot of stuff is WIP.

250ms seemed reasonable. In theory even 25ms should be doable @ 20kHz

kr0ner, 2025-01-27:
Should be fixed here: https://github.com/kr0ner/OneESP32ToRuleThemAll/pull/96

kr0ner, 2025-01-27:
Would also be nice if you could upstream your findings. If in the process you find out that you need to create separate config please try to reuse as much as possible from a common.yaml or wp_base to make it easier for me to maintain

kr0ner, 2025-01-29:
I screwed up in #96 

> Do you get the info directly, when someone changes values at the standard display ? (this does currently not work...) or do you ignore such CAN messages? (maybe I misunderstand the manager_can_id -> should it be the display?)

So setting numbers was no longer possible ... I've added some more functionality to the requestQueue to support those use cases

https://github.com/kr0ner/OneESP32ToRuleThemAll/pull/118

Dobbiiey, 2025-01-29:
Hello again with somehow a discussion, ;-)
up to know I added "*energy*daily*" and "*energy*sum*" to one "*energy*all" or total value (from maths and/or physic view this is ok).

Most important: Please don't take it as criticism! ;-)

`
example "E_HEIZ_ALL":
    lambda: |-
      if (id(EL_AUFNAHMELEISTUNG_HEIZ_TAG_WH).state && id(EL_AUFNAHMELEISTUNG_HEIZ_TAG_KWH).state && id(EL_AUFNAHMELEISTUNG_HEIZ_SUM_KWH).state && id(EL_AUFNAHMELEISTUNG_HEIZ_SUM_MWH).state) {
        id(gE_HEIZ_ALL) = (id(EL_AUFNAHMELEISTUNG_HEIZ_TAG_WH).state / 1000) + id(EL_AUFNAHMELEISTUNG_HEIZ_TAG_KWH).state + id(EL_AUFNAHMELEISTUNG_HEIZ_SUM_KWH).state + (id(EL_AUFNAHMELEISTUNG_HEIZ_SUM_MWH).state * 1000);
`

benefit
- Then I had one nice total_increasing energy-sensor for HASS-energy-dashboard with correct values the complete day...
- For daily consumtion I currently use "utility_meter" to calculate.

The current implementation with "Offset" in wp_daily_energy_combined.yaml has "I think" the issues:
- not exact or "slow" values compared to an "all / total"-energy
- I'm not able to read the values into a second sensor ;-)

Maybe a solution:
You could store the value of the "all / total"-energy at midnight and then use it to calculate the daily COP? So you might have both advantages?

I hope, I was able to express my thoughts clear enough.

And, for sure: It's a really nice project and it makes lot of fun !

kr0ner, 2025-01-29:
This was also brought up by @mkaiser and I changed it here #93 but reverted it here ac9c2a2ed5e62e31c477e7f9ed6b5c451878c80b, because if you update your device without renaming the entities beforehand, you will lose the whole history of your energy and it requires manually tweaking the database to get it back. So for the time being having it named incorrectly is the way to go to not break stuff for the people that are already using it 😉 
At some point I will rename it and figure out a way how to do that without breaking stuff for the existing users.
Until that happens I suggest ti just rename the entities in HA

kr0ner, 2025-01-29:
>Most important: Please don't take it as criticism! ;-)
>
>id(gE_WW_ALL) = (id(EL_AUFNAHMELEISTUNG_WW_TAG_WH).state / 1000) + >id(EL_AUFNAHMELEISTUNG_WW_TAG_KWH).state + id(EL_AUFNAHMELEISTUNG_WW_SUM_KWH).state +>(id(EL_AUFNAHMELEISTUNG_WW_SUM_MWH).state * 1000);
>
>benefit
>
>Then I had one nice total_increasing energy-sensor for HASS-energy-dashboard with correct values the complete day...
>For daily consumtion I currently use "utility_meter" to calculate.
>The current implementation with "Offset" in wp_daily_energy_combined.yaml has "I think" the issues:
>
>not exact or "slow" values compared to an "all / total"-energy
>I'm not able to read the values into a second sensor ;-)
>Maybe a solution:
>You could store the value of the "all / total"-energy at midnight and then use it to calculate the daily COP? So you might have both >advantages?

For HA it should not make any difference. You anyways need to poll the sensors, regardless of which approach you use. For energy dashboard it makes no difference if you have a monotonically increasing sensor or one that resets daily.

>not exact or "slow" values compared to an "all / total"-energy

this I don't get, because also in your case the 3 properties have to be polled and combined. So behind the curtain it is doing the same thing. If you want more fine grained (current default is 5 mins) you can increase the frequency.

>I'm not able to read the values into a second sensor ;-)

What second sensor? You are combining on ESP level then splitting it again on HA level ... looks the same as not splitting it and summing it up in HA xD

Although I agree that it might've been easier to just sum up 3 properties into 1 and scrap all that offset stuff in the first place.
If i remember correctly two years ago when I implemented the possibilities in energy dashboard were pretty limited

Dobbiiey, 2025-01-29:
> not exact or "slow" values compared to an "all / total"-energy

this I don't get, because also in your case the 3 properties have to be polled and combined. So behind the curtain it is doing the same thing. If you want more fine grained (current default is 5 mins) you can increase the frequency.

> 

yes - all correct.
I built 
SUM ( EL_AUFNAHMELEISTUNG_WW_SUMME_KWH  +   EL_AUFNAHMELEISTUNG_WW_SUM_KWH  + EL_AUFNAHMELEISTUNG_WW_SUM_MWH ) 
without concerning the gOFFSET_

I try to add a additional sensor to "my" *lwz403.yaml* like:

> 
#########################################
#                                       #
#   Global variables                    #
#                                       #
#########################################
globals:
  - id: gE_WW_ALL
    type: float
    initial_value: "0.0"

#########################################
#                                       #
#   Template Sensors                    #
#                                       #
#########################################
sensor:
  - platform: template
    name: "E_WW_ALL"
    id: "E_WW_ALL"
    update_interval: $interval_very_slow
    state_class: "total_increasing"
    unit_of_measurement: "kWh"
    accuracy_decimals: "3"   
    icon: "mdi:water-boiler"
    lambda: |-
      if (id(EL_AUFNAHMELEISTUNG_WW_TAG_WH).state && id(EL_AUFNAHMELEISTUNG_WW_TAG_KWH).state && id(EL_AUFNAHMELEISTUNG_WW_SUM_KWH).state && id(EL_AUFNAHMELEISTUNG_WW_SUM_MWH).state) {
        id(gE_WW_ALL) = (id(EL_AUFNAHMELEISTUNG_WW_TAG_WH).state / 1000) + id(EL_AUFNAHMELEISTUNG_WW_TAG_KWH).state + id(EL_AUFNAHMELEISTUNG_WW_SUM_KWH).state + #(id(EL_AUFNAHMELEISTUNG_WW_SUM_MWH).state * 1000);
      }
      return id(gE_WW_ALL);   
``

But I get compiling errors ;-)

[image]

kr0ner, 2025-01-29:
You need to change wp_daily_energy_combined.yaml

and read the raw sensor value of the Wh sensor
sensor:
  - platform: template
    name: ${property_wh}
    id: ${property_wh}
    state_class: "measurement"
    accuracy_decimals: 3
    internal: true
    update_interval: never
    filters:
      - multiply: 0.001
      - lambda: |-
          return x - id(gOFFSET_${property_wh});

mkaiser, 2025-01-29:
> So for the time being having it named incorrectly is the way to go to not break stuff for the people that are already using it 😉
> At some point I will rename it and figure out a way how to do that [...]

I have been at this point with my sungrow inverter integration... Turned out the more users used it, the higher the pressure not to change anything. There are some ugly names bothering me for years now, but I am too afraid of people complaining ...

Personally for me the risk of damaging something in the HA database is too high. For me it would be okay to loose history data. 

I vote for changing ASAP, because after supporting more heatpumps by your awesome integration, there will be a new users every day

kr0ner, 2025-01-29:
@mkaiser I agree that it should be done ASAP, which means as soon as I am sure that data loss can be prevented or easily recovered 😉 if you rename your entities already, you won't even notice 😄

kr0ner, 2025-01-30:
@Dobbiiey ... I'm evaluating your proposal, but I'm stuck ... 
Problem:
```
    PROPERTY(EL_AUFNAHMELEISTUNG_HEIZ_TAG_WH, 0x091e); <-- not reset daily
    PROPERTY(EL_AUFNAHMELEISTUNG_HEIZ_TAG_KWH, 0x091f, Type::et_double_val); <-- reset daily
    PROPERTY(EL_AUFNAHMELEISTUNG_HEIZ_SUM_KWH, 0x0920); <-- monotonically increasing but i have no clue when the HEIZ_TAG_KWH is added to the SUM_KWH
    PROPERTY(EL_AUFNAHMELEISTUNG_HEIZ_SUM_MWH, 0x0921, Type::et_double_val); <-- same
```
So we have SUM and TAG sensors ... they are independently updated ... I would need to implement the same logic again as I did for taking over the offset from one day to the next for the WH part. Cause if I just add TAG_KWH to SUM_MWH it will be wrong at some point. Only adding SUM_KWH to SUM_MWH is not precise enough. So the meter you implemented on HA side seems to be the better option to get monotonically increasing overall MWh value.

Dobbiiey, 2025-01-30:
@kr0ner : thanks for still thinking about my topics! ;-)

To be honest, I'm far from understanding the entire SW architecture.
For example, I briefly tried to create a 'wp_binary_sensor.yaml' without success. (wp_binary.yaml is without a queue)
 ;-)

I'm currently playing with 'wp_generic_combined.yaml' and/or templates...

On the subject:
Is there a hint for me in which order the 4-value querie is done (in case the interval is same), or is it random?
Otherwise I would probably have to think about a separate 'interval:' or 'Time: on_time:'...

Thank you

Dobbiiey, 2025-01-30:
> ```
>     PROPERTY(EL_AUFNAHMELEISTUNG_HEIZ_TAG_WH, 0x091e); <-- not reset daily
>     PROPERTY(EL_AUFNAHMELEISTUNG_HEIZ_TAG_KWH, 0x091f, Type::et_double_val); <-- reset daily [...]

This is my understanding:

    PROPERTY(EL_AUFNAHMELEISTUNG_HEIZ_TAG_WH, 0x091e); <-- not reset daily // yes, it's because Wh can't be added to the integer sensor in kWh
    PROPERTY(EL_AUFNAHMELEISTUNG_HEIZ_TAG_KWH, 0x091f, Type::et_double_val); <-- reset daily // in moment of reset it is added to the SUM ;-)
    PROPERTY(EL_AUFNAHMELEISTUNG_HEIZ_SUM_KWH, 0x0920); <-- monotonically increasing but i have no clue when the HEIZ_TAG_KWH is added to the SUM_KWH // As I remember, it was at midnight
    PROPERTY(EL_AUFNAHMELEISTUNG_HEIZ_SUM_MWH, 0x0921, Type::et_double_val); <-- same
	
	// so collecting all data in one moment (23:59:59 and 0:00:01) ;-) and make the TOATAL_SUM will show same result ;-)

kr0ner, 2025-01-30:
> // so collecting all data in one moment (23:59:59 and 0:00:01) ;-) and make the TOATAL_SUM will show same result ;-)

yes, but what do we gain? The current energy reporting works, the overall consumption is shown in kWh, which is precise enough to compare values. Can easily be fixed with a integral_sum sensor in HA ;)
I don't want to touch this on code level.

kr0ner, 2025-01-30:
> [@kr0ner](https://github.com/kr0ner) : thanks for still thinking about my topics! ;-)
>
> To be honest, I'm far from understanding the entire SW architecture. For example, I briefly tried to create a [...]

binary_sensors are just there to be filled from somewhere else in the code 😉 Currently they are used for representing each bit of the BETRIEBSTATUS to not be presented with 1367 and have to pick the bits manually. So you can create one but you need to add a callback in on_boot to publish the binary_sensor on reception of a property. Hope that helps

The problem with the energy values is that all this stuff requires exact timing. There is no straight forward way to check if the reset happened. TBH even to get to the current state took ages, because the tracking by the heatpump is so weird. In the end I even had to implement automatic time sync to get it working reliably :D It is also not guaranteed that the values are reported in the order in which they were requested. So in theory small jumps are possible ... this might be a thing to be fixed, so that during the day values have to be monotonic. (std::max(old,new)) easy fix. 

>  4-value querie is done (in case the interval is same), or is it random? 
It is not random. The TAG_SUM requests the WH_TAG and KWH_TAG at the same frequency and the overall sums are retrieved slower, but would be sufficient to be retrieved once per day

So to implement what you want ... take e.g.  WW_SUM (KWH_SUM + MWH_SUM) which will stay fixed for the whole day, add TAG_SUM (TAG_WH + TAG_KWH) . At midnight the WW_SUM will be lifted by yesterdays energy consumption and at the same time TAG_SUM will be reset ... so it works, but we will never be able to drop TAG_SUM 🤷‍♂

kr0ner, 2025-01-30:
@Dobbiiey are you fine with closing this issue and tracking everything else in new issues?

closed 2025-02-01 by kr0ner