kr0ner/OneESP32ToRuleThemAll #119: Breaking changes upcoming

kr0ner, 2025-01-30:
Energy sensors are going to be renamed from
```
EL_ENERGIEAUFNAHME_WW_SUMME_KWH
EL_ENERGIEAUFNAHME_HEIZ_SUMME_KWH
```
to
```
EL_AUFNAHMELEISTUNG_WW_SUMME_KWH
EL_AUFNAHMELEISTUNG_HEIZ_SUMME_KWH
```

You need to rename the sensors in Home Assistant !before! updating the ESP. If you miss this, you will have to live with the loss of heating / hot water energy consumption data in the energy dashboard or to manually fix the SQLite database.

sascha-egerer, 2025-02-02:
Isn't it the other way round?
From
```
EL_AUFNAHMELEISTUNG_WW_SUMME_KWH
EL_AUFNAHMELEISTUNG_HEIZ_SUMME_KWH
```
to
```
EL_ENERGIEAUFNAHME_WW_SUMME_KWH
EL_ENERGIEAUFNAHME_HEIZ_SUMME_KWH
```
?

kr0ner, 2025-02-02:
Oh boy ... You are totally right

kr0ner, 2025-03-24:
[image]

Entity has to be changed here. After that the energy dashboard most likely will show an error that the entity with the old name is not defined. Click on the existing entry and update the name and id. Dashboard and long term stats should be fine.

kr0ner, 2025-03-25:
If anyone needs help restoring data after renaming please let me know ... been there and know what to do 😅

kr0ner, 2025-03-26:
the fix https://github.com/kr0ner/OneESP32ToRuleThemAll/pull/153

Schmidtjanroman, 2025-04-20:
@kr0ner it looks like I've installed esphome and your git while those where in the code

`EL_AUFNAHMELEISTUNG_WW_SUMME_KWH
EL_AUFNAHMELEISTUNG_HEIZ_SUMME_KWH`

I recentyl did a pull because of the nice new feature of Kühlung switch and have now the new (old)

`EL_ENERGIEAUFNAHME_WW_SUMME_KWH
EL_ENERGIEAUFNAHME_HEIZ_SUMME_KWH`

but without any history because its in the EL_AUFNAHMELEISTUNG Data.

kr0ner, 2025-04-20:
There is a SQLite addon in the store. You need to install that and open the home assistant database that is in your config folder. create a backup before you touch anything! You need to go to the history metatable and find the entries that belong to the old and new sensor name. Rename the sensor to the old name, delete the entry of the new name, and rename the sensor again to the new name. Then go to the energy dashboard and add the new sensor. You might need to play around a bit. Its just from what I remember. Good luck 🫡👍🏼

closed 2025-07-11 by kr0ner