kr0ner/OneESP32ToRuleThemAll #129: A lot of messages in the log

sascha-egerer, 2025-02-02:
I've updated to master branch today and i have a lot of log messages now. Is that expected?
It's like 30 - 50 (identical?) messages per second.

```
18:59:56][I][Communication:150]: Message received: Read/Write ID 0xd2 0x22(0x6a2) for property MOTORLEISTUNG (0x06a0) with raw value: 0
[18:59:56][I][Communication:150]: Message received: Read/Write ID 0xd2 0x22(0x6a2) for property MOTORLEISTUNG (0x06a0) with raw value: 0
[18:59:56][I][Communication:150]: Message received: Read/Write ID 0xd2 0x22(0x6a2) for property MOTORLEISTUNG (0x06a0) with raw value: 0
[18:59:56][I][Communication:150]: Message received: Read/Write ID 0xd2 0x22(0x6a2) for property MOTORSPANNUNG (0x06a1) with raw value: 0
[18:59:56][I][Communication:150]: Message received: Read/Write ID 0xd2 0x22(0x6a2) for property MOTORSPANNUNG (0x06a1) with raw value: 0
[18:59:56][I][Communication:150]: Message received: Read/Write ID 0xd2 0x22(0x6a2) for property MOTORSPANNUNG (0x06a1) with raw value: 0
[18:59:56][I][Communication:150]: Message received: Read/Write ID 0xd2 0x22(0x6a2) for property MOTORSPANNUNG (0x06a1) with raw value: 0
[18:59:56][I][Communication:150]: Message received: Read/Write ID 0xd2 0x22(0x6a2) for property MOTORSPANNUNG (0x06a1) with raw value: 0
[18:59:56][I][Communication:150]: Message received: Read/Write ID 0xd2 0x22(0x6a2) for property MOTORSPANNUNG (0x06a1) with raw value: 0
[18:59:56][I][Communication:150]: Message received: Read/Write ID 0xd2 0x22(0x6a2) for property MOTORSPANNUNG (0x06a1) with raw value: 0
[18:59:56][I][Communication:150]: Message received: Read/Write ID 0xd2 0x22(0x6a2) for property MOTORSPANNUNG (0x06a1) with raw value: 0
[18:59:56][I][Communication:150]: Message received: Read/Write ID 0xd2 0x22(0x6a2) for property MOTORSPANNUNG (0x06a1) with raw value: 0
[18:59:56][I][Communication:150]: Message received: Read/Write ID 0xd2 0x22(0x6a2) for property MOTORSPANNUNG (0x06a1) with raw value: 0
[18:59:56][I][Communication:150]: Message received: Read/Write ID 0xd2 0x22(0x6a2) for property MOTORSPANNUNG (0x06a1) with raw value: 0
[18:59:56][I][Communication:150]: Message received: Read/Write ID 0xd2 0x22(0x6a2) for property DIFFERENZDRUCK (0xc11e) with raw value: 0
[18:59:56][I][Communication:150]: Message received: Read/Write ID 0xd2 0x22(0x6a2) for property DIFFERENZDRUCK (0xc11e) with raw value: 0
[18:59:56][I][Communication:150]: Message received: Read/Write ID 0xd2 0x22(0x6a2) for property DIFFERENZDRUCK (0xc11e) with raw value: 0
[18:59:56][I][Communication:150]: Message received: Read/Write ID 0xd2 0x22(0x6a2) for property DIFFERENZDRUCK (0xc11e) with raw value: 0
[18:59:56][I][Communication:150]: Message received: Read/Write ID 0xd2 0x22(0x6a2) for property DIFFERENZDRUCK (0xc11e) with raw value: 0
[18:59:56][I][Communication:150]: Message received: Read/Write ID 0xd2 0x22(0x6a2) for property DIFFERENZDRUCK (0xc11e) with raw value: 0
[18:59:56][I][Communication:150]: Message received: Read/Write ID 0xd2 0x22(0x6a2) for property DIFFERENZDRUCK (0xc11e) with raw value: 0
[18:59:56][I][Communication:150]: Message received: Read/Write ID 0xd2 0x22(0x6a2) for property DIFFERENZDRUCK (0xc11e) with raw value: 0
[18:59:56][I][Communication:150]: Message received: Read/Write ID 0xd2 0x22(0x6a2) for property DIFFERENZDRUCK (0xc11e) with raw value: 0
[18:59:56][I][Communication:150]: Message received: Read/Write ID 0xd2 0x22(0x6a2) for property DIFFERENZDRUCK (0xc11e) with raw value: 0
[18:59:56][I][Communication:150]: Message received: Read/Write ID 0xd2 0x22(0x6a2) for property DIFFERENZDRUCK (0xc11e) with raw value: 0
[18:59:56][I][Communication:150]: Message received: Read/Write ID 0xd2 0x22(0x6a2) for property DIFFERENZDRUCK (0xc11e) with raw value: 0
[18:59:56][I][Communication:150]: Message received: Read/Write ID 0xd2 0x22(0x6a2) for property DIFFERENZDRUCK (0xc11e) with raw value: 0
[18:59:56][I][Communication:150]: Message received: Read/Write ID 0xd2 0x22(0x6a2) for property DIFFERENZDRUCK (0xc11e) with raw value: 0
[18:59:56][I][Communication:150]: Message received: Read/Write ID 0xd2 0x22(0x6a2) for property WAERMEERTRAG_HEIZ_SUM_KWH (0x0930) with raw value: 115
[18:59:56][I][Communication:150]: Message received: Read/Write ID 0xd2 0x22(0x6a2) for property WAERMEERTRAG_HEIZ_SUM_KWH (0x0930) with raw value: 115
[18:59:56][I][Communication:150]: Message received: Read/Write ID 0xd2 0x22(0x6a2) for property WAERMEERTRAG_HEIZ_SUM_KWH (0x0930) with raw value: 115
[18:59:56][I][Communication:150]: Message received: Read/Write ID 0xd2 0x22(0x6a2) for property WAERMEERTRAG_HEIZ_SUM_KWH (0x0930) with raw value: 115
[18:59:56][I][Communication:150]: Message received: Read/Write ID 0xd2 0x22(0x6a2) for property WAERMEERTRAG_HEIZ_SUM_KWH (0x0930) with raw value: 115
[18:59:56][I][Communication:150]: Message received: Read/Write ID 0xd2 0x22(0x6a2) for property WAERMEERTRAG_HEIZ_SUM_KWH (0x0930) with raw value: 115
[18:59:56][I][Communication:150]: Message received: Read/Write ID 0xd2 0x22(0x6a2) for property WAERMEERTRAG_HEIZ_SUM_MWH (0x0931) with raw value: 5
[18:59:56][I][Communication:150]: Message received: Read/Write ID 0xd2 0x22(0x6a2) for property WAERMEERTRAG_HEIZ_SUM_MWH (0x0931) with raw value: 5
[18:59:56][I][Communication:150]: Message received: Read/Write ID 0xd2 0x22(0x6a2) for property WAERMEERTRAG_HEIZ_SUM_MWH (0x0931) with raw value: 5
[18:59:56][I][Communication:150]: Message received: Read/Write ID 0xd2 0x22(0x6a2) for property WAERMEERTRAG_HEIZ_SUM_MWH (0x0931) with raw value: 5
[18:59:56][I][Communication:150]: Message received: Read/Write ID 0xd2 0x22(0x6a2) for property WAERMEERTRAG_HEIZ_SUM_MWH (0x0931) with raw value: 5
[18:59:56][I][Communication:150]: Message received: Read/Write ID 0xd2 0x22(0x6a2) for property WAERMEERTRAG_HEIZ_SUM_MWH (0x0931) with raw value: 5
[18:59:56][I][Communication:150]: Message received: Read/Write ID 0xd2 0x22(0x6a2) for property WAERMEERTRAG_HEIZ_SUM_MWH (0x0931) with raw value: 5
[18:59:56][I][Communication:150]: Message received: Read/Write ID 0xd2 0x22(0x6a2) for property WAERMEERTRAG_HEIZ_SUM_MWH (0x0931) with raw value: 5
[18:59:56][I][Communication:150]: Message received: Read/Write ID 0xd2 0x22(0x6a2) for property WAERMEERTRAG_HEIZ_SUM_MWH (0x0931) with raw value: 5
[18:59:56][I][Communication:150]: Message received: Read/Write ID 0xd2 0x22(0x6a2) for property WAERMEERTRAG_HEIZ_SUM_MWH (0x0931) with raw value: 5
[18:59:56][I][Communication:150]: Message received: Read/Write ID 0xd2 0x22(0x6a2) for property WAERMEERTRAG_HEIZ_SUM_MWH (0x0931) with raw value: 5
[18:59:56][I][Communication:150]: Message received: Read/Write ID 0xd2 0x22(0x6a2) for property WAERMEERTRAG_HEIZ_SUM_MWH (0x0931) with raw value: 5
[18:59:56][I][Communication:150]: Message received: Read/Write ID 0xd2 0x22(0x6a2) for property WAERMEERTRAG_HEIZ_SUM_MWH (0x0931) with raw value: 5
[18:59:56][I][Communication:150]: Message received: Read/Write ID 0xd2 0x22(0x6a2) for property WAERMEERTRAG_HEIZ_SUM_MWH (0x0931) with raw value: 5
[18:59:56][I][Communication:150]: Message received: Read/Write ID 0xd2 0x22(0x6a2) for property WAERMEERTRAG_HEIZ_SUM_MWH (0x0931) with raw value: 5
[18:59:56][I][Communication:150]: Message received: Read/Write ID 0xd2 0x22(0x6a2) for property WAERMEERTRAG_HEIZ_SUM_MWH (0x0931) with raw value: 5
[18:59:56][I][Communication:150]: Message received: Read/Write ID 0xd2 0x22(0x6a2) for property WAERMEERTRAG_HEIZ_SUM_MWH (0x0931) with raw value: 5
[18:59:57][I][Communication:150]: Message received: Read/Write ID 0xd2 0x22(0x6a2) for property WAERMEERTRAG_HEIZ_SUM_MWH (0x0931) with raw value: 5
[18:59:57][I][Communication:150]: Message received: Read/Write ID 0xd2 0x22(0x6a2) for property WAERMEERTRAG_HEIZ_SUM_MWH (0x0931) with raw value: 5
[18:59:57][I][Communication:150]: Message received: Read/Write ID 0xd2 0x22(0x6a2) for property WAERMEERTRAG_HEIZ_SUM_MWH (0x0931) with raw value: 5
[18:59:57][I][Communication:150]: Message received: Read/Write ID 0xd2 0x22(0x6a2) for property WAERMEERTRAG_HEIZ_SUM_MWH (0x0931) with raw value: 5
[18:59:57][I][Communication:150]: Message received: Read/Write ID 0xd2 0x22(0x6a2) for property WAERMEERTRAG_HEIZ_SUM_MWH (0x0931) with raw value: 5
[18:59:57][I][Communication:150]: Message received: Read/Write ID 0xd2 0x22(0x6a2) for property WAERMEERTRAG_HEIZ_SUM_MWH (0x0931) with raw value: 5
[18:59:57][I][Communication:150]: Message received: Read/Write ID 0xd2 0x22(0x6a2) for property VORLAUFISTTEMP (0x000f) with raw value: 270
[... cut ...]
[18:59:57][I][Communication:150]: Message received: Read/Write ID 0xd2 0x22(0x6a2) for property VERDICHTER_STARTS_K (0xc0f5) with raw value: 4
[18:59:57][I][Communication:150]: Message received: Read/Write ID 0xd2 0x22(0x6a2) for property VERDICHTER_STARTS_K (0xc0f5) with raw value: 4
[18:59:57][I][Communication:150]: Message received: Read/Write ID 0xd2 0x22(0x6a2) for property VERDICHTER_STARTS_K (0xc0f5) with raw value: 4
[18:59:57][I][Communication:150]: Message received: Read/Write ID 0xd2 0x22(0x6a2) for property VERDICHTER_STARTS_K (0xc0f5) with raw value: 4
```

kr0ner, 2025-02-02:
Nope, that is not expected and does not happen in my setup. Please roll back now.

What are your local changes? Which of the yamls do you use?

sascha-egerer, 2025-02-02:
I think nothing related is changed in my setup
[changes.patch](https://github.com/user-attachments/files/18633407/changes.patch)

kr0ner, 2025-02-02:
I have no clue what is going on ... but if the ESP would request the data so often you should be able to tell from the logs e.g

ESP_LOGI("QUEUE", "Requesting data from %s for %s", member.name.c_str(), std::string(property.name).c_str());
    conditionalRequests.emplace_back(std::make_pair(member, property));

but there are no requests visible ... only responses. So if the heatpump constantly sends data, the ESP will process it 🤷 

now we need to find out what is happening ...
can you provide more logs and maybe try rebooting your esp and heat pump?

kr0ner, 2025-02-02:
Can you compile with log level DEBUG?

sascha-egerer, 2025-02-02:
Restarting did not help (restarting heatpump does also disconnect esp from power so both has been hard set)
```
WARNING esp32-poe-technik @ 192.168.1.221: Connection error occurred: [Errno 54] Connection reset by peer
INFO Processing unexpected disconnect from ESPHome API for esp32-poe-technik @ 192.168.1.221
WARNING Disconnected from API
INFO Successfully connected to esp32-poe-technik @ 192.168.1.221 in 0.006s
INFO Successful handshake with esp32-poe-technik @ 192.168.1.221 in 0.076s
[19:56:10][I][Communication:150]: Message received: Read/Write ID 0xd2 0x22(0x6a2) for property WAERMEERTRAG_WW_SUM_MWH (0x092d) with raw value: 5
[19:56:10][I][Communication:150]: Message received: Read/Write ID 0xd2 0x22(0x6a2) for property WAERMEERTRAG_WW_SUM_MWH (0x092d) with raw value: 5
[19:56:10][I][Communication:150]: Message received: Read/Write ID 0xd2 0x22(0x6a2) for property WAERMEERTRAG_WW_SUM_MWH (0x092d) with raw value: 5
[19:56:10][I][Communication:150]: Message received: Read/Write ID 0xd2 0x22(0x6a2) for property WAERMEERTRAG_HEIZ_SUM_KWH (0x0930) with raw value: 115
[19:56:10][I][Communication:150]: Message received: Read/Write ID 0xd2 0x22(0x6a2) for property WAERMEERTRAG_HEIZ_SUM_KWH (0x0930) with raw value: 115
[19:56:11][I][Communication:150]: Message received: Read/Write ID 0xd2 0x22(0x6a2) for property WAERMEERTRAG_HEIZ_SUM_KWH (0x0930) with raw value: 115
[19:56:11][I][Communication:150]: Message received: Read/Write ID 0xd2 0x22(0x6a2) for property WAERMEERTRAG_HEIZ_SUM_KWH (0x0930) with raw value: 115
[19:56:11][I][Communication:150]: Message received: Read/Write ID 0xd2 0x22(0x6a2) for property WAERMEERTRAG_HEIZ_SUM_KWH (0x0930) with raw value: 115
[19:56:11][I][Communication:150]: Message received: Read/Write ID 0xd2 0x22(0x6a2) for property WAERMEERTRAG_HEIZ_SUM_KWH (0x0930) with raw value: 115
[19:56:11][I][Communication:150]: Message received: Read/Write ID 0xd2 0x22(0x6a2) for property WAERMEERTRAG_HEIZ_SUM_KWH (0x0930) with raw value: 115
[19:56:11][I][Communication:150]: Message received: Read/Write ID 0xd2 0x22(0x6a2) for property WAERMEERTRAG_HEIZ_SUM_KWH (0x0930) with raw value: 115
[19:56:11][I][Communication:150]: Message received: Read/Write ID 0xd2 0x22(0x6a2) for property WAERMEERTRAG_HEIZ_SUM_KWH (0x0930) with raw value: 115
[19:56:11][I][Communication:150]: Message received: Read/Write ID 0xd2 0x22(0x6a2) for property WAERMEERTRAG_HEIZ_SUM_KWH (0x0930) with raw value: 115
[19:56:11][I][Communication:150]: Message received: Read/Write ID 0xd2 0x22(0x6a2) for property WAERMEERTRAG_HEIZ_SUM_MWH (0x0931) with raw value: 5
[19:56:11][I][Communication:150]: Message received: Read/Write ID 0xd2 0x22(0x6a2) for property WAERMEERTRAG_HEIZ_SUM_MWH (0x0931) with raw value: 5
[19:56:11][I][Communication:150]: Message received: Read/Write ID 0xd2 0x22(0x6a2) for property WAERMEERTRAG_HEIZ_SUM_MWH (0x0931) with raw value: 5
[19:56:11][I][Communication:150]: Message received: Read/Write ID 0xd2 0x22(0x6a2) for property WAERMEERTRAG_HEIZ_SUM_MWH (0x0931) with raw value: 5
[19:56:11][I][Communication:150]: Message received: Read/Write ID 0x31 0x00(0x180) for property PROGRAMMSCHALTER (0x0112) with raw value: 0
[19:56:11][I][Communication:150]: Message received: Read/Write ID 0x31 0x00(0x180) for property SAMMLERISTTEMP (0x000d) with raw value: 0
[19:56:11][I][Communication:150]: Message received: Read/Write ID 0x31 0x00(0x180) for property SAMMLERISTTEMP (0x000d) with raw value: 0
[19:56:11][I][Communication:150]: Message received: Read/Write ID 0xd2 0x21(0x6a1) for property PROGRAMMSCHALTER (0x0112) with raw value: 2816
[19:56:11][I][Communication:150]: Message received: Read/Write ID 0xd2 0x21(0x6a1) for property AUSSENTEMP (0x000c) with raw value: 65529
[19:56:11][I][Communication:150]: Message received: Read/Write ID 0xd2 0x21(0x6a1) for property AUSSENTEMP (0x000c) with raw value: 65529
[19:56:11][I][Communication:150]: Message received: Read/Write ID 0xd2 0x21(0x6a1) for property SAMMLERISTTEMP (0x000d) with raw value: 273
[19:56:11][I][Communication:150]: Message received: Read/Write ID 0x31 0x00(0x180) for property SPEICHERISTTEMP (0x000e) with raw value: 0
[19:56:11][I][Communication:150]: Message received: Read/Write ID 0xd2 0x21(0x6a1) for property SPEICHERISTTEMP (0x000e) with raw value: 457
[19:56:11][I][Communication:150]: Message received: Read/Write ID 0x31 0x00(0x180) for property AUSSENTEMP (0x000c) with raw value: 0
[19:56:11][I][Communication:150]: Message received: Read/Write ID 0xd2 0x21(0x6a1) for property SPEICHERISTTEMP (0x000e) with raw value: 457
[19:56:11][I][Communication:150]: Message received: Read/Write ID 0xd2 0x21(0x6a1) for property SPEICHERISTTEMP (0x000e) with raw value: 457
[19:56:11][I][Communication:150]: Message received: Read/Write ID 0xd2 0x21(0x6a1) for property SPEICHERISTTEMP (0x000e) with raw value: 457
[19:56:11][I][Communication:150]: Message received: Read/Write ID 0xd2 0x21(0x6a1) for property SPEICHERISTTEMP (0x000e) with raw value: 457
[19:56:11][I][Communication:150]: Message received: Read/Write ID 0xd2 0x21(0x6a1) for property SPEICHERISTTEMP (0x000e) with raw value: 457
[19:56:11][I][Communication:150]: Message received: Read/Write ID 0xd2 0x22(0x6a2) for property SPEICHERSOLLTEMP (0x0003) with raw value: 420
[19:56:11][I][Communication:150]: Message received: Read/Write ID 0xd2 0x22(0x6a2) for property SPEICHERSOLLTEMP (0x0003) with raw value: 420
[... cut ...]
[19:56:11][I][Communication:150]: Message received: Read/Write ID 0xd2 0x22(0x6a2) for property SPEICHERISTTEMP (0x000e) with raw value: 458
[19:56:11][I][Communication:150]: Message received: Read/Write ID 0xd2 0x22(0x6a2) for property SPEICHERISTTEMP (0x000e) with raw value: 458
[19:56:11][I][Communication:150]: Message received: Read/Write ID 0xd2 0x22(0x6a2) for property SPEICHERISTTEMP (0x000e) with raw value: 458
[19:56:11][I][Communication:150]: Message received: Read/Write ID 0xd2 0x22(0x6a2) for property SPEICHERISTTEMP (0x000e) with raw value: 458
```

kr0ner, 2025-02-02:
So I'm currently running the THZ504.yaml without these issues. Nevertheless I improved logging a bit to not spam so much. 
https://github.com/kr0ner/OneESP32ToRuleThemAll/pull/130
Still does not help for your findings. 

It is weird, why so many messages are being received. 
What ESPHome version do you use?
What entities are displayed in the home screen of your display?

Found this somewhere related to the CAN Controller
// - If you only have 2 devices on the CAN bus (the device running this sketch and some other device sending 
//   messages), you may find that duplicate messages are received when waking up. This is because when 
//   the MCP2515 wakes up it enters LISTENONLY mode where it does not send ACKs to messages, so the transmitter 
//   will retransmit the same message a few times. See below for a simple solution to filter duplicate messages out.

sascha-egerer, 2025-02-02:
I'm also using THZ504.yaml and did not notice this issue before. But now i'm also not sure if it did exist before i've updated today as i did not checked the logs before i've updated.
If now also tried do use an older version of this package but still the same spam.

> What ESPHome version do you use?
I've been using the latest 2024.12.4 and now also tried with 2024.10.3 and 2024.11.3 but still the same issue.

So... maybe it's an hardware issue. I don't know...

kr0ner, 2025-02-02:
another thing that comes to my mind would be termination. Did you terminate with 120Ohm? Ther should be a jumper on the board ... might work with and without ... but worth a try

sascha-egerer, 2025-02-02:
No i do not. But i actually did not touch anything since 6 month or so. Just do updates on the package from time to time. Maybe i can have a look tomorrow into the hardware.
I have to remove the hardware anyways in 8 days as i've a heat pump maintenance appointment and i'm not sure what he would say about a few cables and a circuit board that are just hanging there. ;)
So maybe a good time to recheck it and finally put it into a case or so.

kr0ner, 2025-02-02:
> I have to remove the hardware anyways in 8 days as i've a heat pump maintenance appointment and i'm not sure what he would say about a few cables and a circuit board that are just hanging there. ;)

haha 🤣  #me2 in summer

I bought the MCP2515 as a pack of 5 on Amazon and also use multiple ESPs in case something goes down the drain or I get weird behavior. The quality of these is sometimes really 💩 I spent a whole night trying to get a differential pressure sensor up and running just to find out the ESP was bugged and the same software was running perfectly fine on another one 🤦 
I keep my fingers crossed and hope it us just a small hardware problem. Please keep me posted so that I can benefit from your experience with the topic ;) Feel free to reach out at any time

sascha-egerer, 2025-02-02:
Maybe it's also MCP2515 again. It somehow died also the last time. Tried to figure out why it does not work anymore and then found out that the MCP2515 was just broken - at least it worked with another one.
Still looking for an esp32 with LAN and canbus on it and compatible with esphome for sure... Not sure if there are any news since the last year... but maybe i'll have a look. 

anyway, thank you very much for your great support for random strangers

kr0ner, 2025-02-02:
I can just recommend to buy a few more and off a decent brand ...
for ESP I can recommend this one:
https://www.olimex.com/Products/IoT/ESP32/ESP32-POE/open-source-hardware

flashed it like 5000 times while developing this stuff and it still works :D never re/disconnected randomly and is in general way more reliable than a WiFi based ESP.

kr0ner, 2025-02-02:
>anyway, thank you very much for your great support for random strangers

I'm happy to help and happy for all you "random not so random people" helping me find bugs for more than a year :D

mkaiser, 2025-02-02:
> another thing that comes to my mind would be termination. Did you terminate with 120Ohm? There should be a jumper on the board ... might work with and without ... but worth a try

The CAN installation is already terminated. So adding another 120Ohm resistor (i.e. having 2x 120Ohm in parallel) is not a good idea :)

sascha-egerer, 2025-02-03:
> I can just recommend to buy a few more and off a decent brand ...
> for ESP I can recommend this one:
> https://www.olimex.com/Products/IoT/ESP32/ESP32-POE/open-source-hardware
>
> flashed it like 5000 [...]

It has also POE 😱 thank you for pointing ne to this one.

sascha-egerer, 2025-02-03:
Just a short update:
1. I've removed the CAN connection to the heatpump -> log is still going crazy
2. I've removed the power form the MCP -> log is silent

So i think this MCP2515 is broken again... But i don't have one to test that is not broken ;)  So i have to wait until my order arrives.

kr0ner, 2025-02-03:
Thanks for the update. Was expecting sth. like that. Some buffer not being cleared or so. Cause there is no chance that the heatpump will spam messages like that... Hopefully 😅

sascha-egerer, 2025-02-03:
Ah and now i remember again why I've not found a ESP32 with LAN... i wanted one that supports 20KBPS with the integrated CAN controller (https://esphome.io/components/canbus/esp32_can). And that's why i didn't find one...

kr0ner, 2025-02-03:
You know that you are going to need a CAN-Transceiver IC when using the internal CAN, right? I got that initially wrong and thought I can just use the board as is with no other modules attached.

And in this case it makes no sense to use the internal CAN over MCP2515 board. Costs more or less the same and still requires some sort of wiring 🤷 

Having LAN and PoE was most important req. for me

sascha-egerer, 2025-02-03:
Yeah i just noticed that today when i scrolled down on that page ;-) 

[image]

sascha-egerer, 2025-02-10:
I've just replaced the MCP and i still get the same log message spam. I'll continue with replacing the ESP tomorrow...

kr0ner, 2025-02-10:
Those are not the news I was hoping for.

sascha-egerer, 2025-02-11:
New ESP32-POE is in place and now everything works as expected again. No CANBUS Log spamming anymore. 

```
[11:30:09][I][Communication:158]: Message received: Read/Write ID 0xd2 0x22(0x6a2) for property SAMMLERISTTEMP (0x000d) with raw value: 285
[11:30:09][I][QUEUE:094]: Requesting data from Kessel for MOTORLEISTUNG
[11:30:09][I][Communication:158]: Message received: Read/Write ID 0xd2 0x22(0x6a2) for property MOTORLEISTUNG (0x06a0) with raw value: 0
[11:30:09][I][QUEUE:094]: Requesting data from HK1 for VORLAUFISTTEMP
[11:30:10][I][Communication:158]: Message received: Read/Write ID 0xd2 0x22(0x6a2) for property VORLAUFISTTEMP (0x000f) with raw value: 294
[11:30:10][I][QUEUE:094]: Requesting data from Kessel for HEIZLEISTUNG_RELATIV
[11:30:10][I][Communication:158]: Message received: Read/Write ID 0xd2 0x22(0x6a2) for property HEIZLEISTUNG_RELATIV (0x069a) with raw value: 23
[11:30:11][I][QUEUE:094]: Requesting data from Kessel for SPEICHERISTTEMP
[11:30:12][I][Communication:158]: Message received: Read/Write ID 0xd2 0x22(0x6a2) for property SPEICHERISTTEMP (0x000e) with raw value: 498
[11:30:18][I][Communication:158]: Message received: Read/Write ID 0xd2 0x21(0x6a1) for property AUSSENTEMP (0x000c) with raw value: 63
[11:30:18][I][Communication:158]: Message received: Read/Write ID 0xd2 0x21(0x6a1) for property SAMMLERISTTEMP (0x000d) with raw value: 283
[11:30:18][I][Communication:158]: Message received: Read/Write ID 0xd2 0x21(0x6a1) for property SPEICHERISTTEMP (0x000e) with raw value: 497
[11:30:25][I][QUEUE:094]: Requesting data from Kessel for VOLUMENSTROM
[11:30:26][I][Communication:158]: Message received: Read/Write ID 0xd2 0x22(0x6a2) for property VOLUMENSTROM (0x01da) with raw value: 167
[11:30:28][I][Communication:158]: Message received: Read/Write ID 0xd2 0x21(0x6a1) for property AUSSENTEMP (0x000c) with raw value: 62
[11:30:28][I][Communication:158]: Message received: Read/Write ID 0xd2 0x21(0x6a1) for property SAMMLERISTTEMP (0x000d) with raw value: 281
[11:30:28][I][Communication:158]: Message received: Read/Write ID 0xd2 0x21(0x6a1) for property SPEICHERISTTEMP (0x000e) with raw value: 497
[11:30:32][I][Communication:158]: Message received: Read/Write ID 0xd0 0x3c(0x6bc) for property MINUTE (0x0126) with raw value: 7936
[11:30:35][I][QUEUE:094]: Requesting data from Kessel for ZULUFT_IST
[11:30:36][I][Communication:158]: Message received: Read/Write ID 0xd2 0x22(0x6a2) for property ZULUFT_IST (0x0597) with raw value: 26
[11:30:37][I][Communication:158]: Message received: Read/Write ID 0xd2 0x21(0x6a1) for property AUSSENTEMP (0x000c) with raw value: 63
[11:30:37][I][Communication:158]: Message received: Read/Write ID 0xd2 0x21(0x6a1) for property SAMMLERISTTEMP (0x000d) with raw value: 282
[11:30:37][I][Communication:158]: Message received: Read/Write ID 0xd2 0x21(0x6a1) for property SPEICHERISTTEMP (0x000e) with raw value: 497
[11:30:37][I][Communication:158]: Message received: Read/Write ID 0xd0 0x3c(0x6bc) for property MINUTE (0x0126) with raw value: 7936
[11:30:37][I][Communication:158]: Message received: Read/Write ID 0xd0 0x3c(0x6bc) for property FEHLERMELDUNG (0x0001) with raw value: 0
[11:30:37][I][Communication:158]: Message received: Read/Write ID 0xd0 0x3c(0x6bc) for property BETRIEBS_STATUS (0x0176) with raw value: 1029
[11:30:37][I][Communication:158]: Message received: Read/Write ID 0xd0 0x3c(0x6bc) for property PROGRAMMSCHALTER (0x0112) with raw value: 2816
[11:30:37][I][Communication:158]: Message received: Read/Write ID 0xd0 0x3c(0x6bc) for property WOCHENTAG (0x0121) with raw value: 256
[11:30:37][I][Communication:158]: Message received: Read/Write ID 0xd0 0x3c(0x6bc) for property TAG (0x0122) with raw value: 2816
[11:30:37][I][Communication:158]: Message received: Read/Write ID 0xd0 0x3c(0x6bc) for property MONAT (0x0123) with raw value: 512
[11:30:37][I][Communication:158]: Message received: Read/Write ID 0xd0 0x3c(0x6bc) for property JAHR (0x0124) with raw value: 6400
[11:30:37][I][Communication:158]: Message received: Read/Write ID 0xd0 0x3c(0x6bc) for property STUNDE (0x0125) with raw value: 2816
[11:30:48][I][Communication:158]: Message received: Read/Write ID 0xd2 0x21(0x6a1) for property AUSSENTEMP (0x000c) with raw value: 62
[11:30:48][I][Communication:158]: Message received: Read/Write ID 0xd2 0x21(0x6a1) for property SAMMLERISTTEMP (0x000d) with raw value: 281
[11:30:48][I][Communication:158]: Message received: Read/Write ID 0xd2 0x21(0x6a1) for property SPEICHERISTTEMP (0x000e) with raw value: 497
[11:30:51][I][QUEUE:094]: Requesting data from Kessel for ZULUFT_SOLL
[11:30:52][I][Communication:158]: Message received: Read/Write ID 0xd2 0x22(0x6a2) for property ZULUFT_SOLL (0x0596) with raw value: 170
[11:30:53][I][QUEUE:094]: Requesting data from Kessel for DIFFERENZDRUCK
[11:30:53][I][Communication:158]: Message received: Read/Write ID 0xd2 0x22(0x6a2) for property DIFFERENZDRUCK (0xc11e) with raw value: 0
```

Thank you for your support again! I'll close the issue now.

closed 2025-02-11 by sascha-egerer

kr0ner, 2025-02-11:
Great News 😀 although this is really weird 😅