kr0ner/OneESP32ToRuleThemAll #132: LWZ 5 CS COP WW TAG value shows 128

aLimp86, 2025-02-07:
Hi all, something seems to be off for the calculation of LWZ 5 CS COP WW TAG. 
The values are the following. 
[image]

Best regards,
Andy

kr0ner, 2025-02-07:
Hi Andy,

are you on latest master? Asking cause I recently added filters to the sensors 😉
Can you show me the history of COP WW Tag? Is it always off like that or just right after midnight and/or uploading a new firmware?

aLimp86, 2025-02-07:
Hi, 
I'm on the current master branch. Your adaption of these values made me update. 
Here you can see when it happend. 
[image]
[image]
[image]

So on Thursday there was no need for new hot water. Maybe it was divided 0 through 0? Because 128 seems to be a special value (0..127). 

Best regards,
Andy

kr0ner, 2025-02-10:
Division by 0 would most likely crash the ESP
```
if (id(WAERMEERTRAG_HEIZ_TAG_SUMME_KWH).state && id(EL_AUFNAHMELEISTUNG_HEIZ_SUMME_KWH).state && id(EL_AUFNAHMELEISTUNG_HEIZ_SUMME_KWH).state != 0.0f) {
  id(gCOP_HEIZ_TAG) = id(WAERMEERTRAG_HEIZ_TAG_SUMME_KWH).state / id(EL_AUFNAHMELEISTUNG_HEIZ_SUMME_KWH).state;
}
return id(gCOP_HEIZ_TAG);
```

So in theory it should just return the value from the day before (or 1.0 if never set) until the conditions are met so that a new CoP can be calculated 🤷 
Did it ever occur again?

aLimp86, 2025-02-14:
Now it is showing the value 4. But the input values are again 0.0 today. So normally nothing should have been calculated.

kr0ner, 2025-02-14:
This was intended, because otherwise you get too high values in the early hours or too small ones. But I get your point. Since there is TAG in the name. What what be the preferred way? Removing TAG from the name or having incorrect values in the early days?

aLimp86, 2025-02-16:
I would expect that the sensor resets at 0:00 and then ok, incorrect values in the early days I guess. Mhh

Quotic, 2025-02-16:
In my opinion, it would make the most sense to always look at the last 24 hours instead of the day. Not Sure that is possible.

kr0ner, 2025-02-16:
@aLimp86 I'll add a flag to support reset at midnight.
@Quotic I agree that this would be the most useful metric, but hard to calculate :-/ The energy sensors are reset every day so rolling avg over 24h would always get the dent. We'd require overall el. & heat power tracking to get there. I might add this in the far future if nothing else is left :D

kr0ner, 2025-02-16:
@aLimp86 or basically just add this to your config:

```
time:
  - platform: sntp
    id: sntp_time
    on_time:
      - seconds: 0
        minutes: 0
        hours: 0
        then:
          - globals.set:
            id: gCOP_WW_TAG
            value: '1.0'
          - globals.set:
            id: gCOP_HEIZ_TAG
            value: '1.0'
```

There is no straight forward way of having this configurable.
I will rename the values so that they no longer give the impression of tracking the COP values of the current day.

closed 2025-02-16 by kr0ner