kr0ner/OneESP32ToRuleThemAll #133: Unscheduled ventilation function // außerplanmäßige Lüftung

Dobbiiey, 2025-02-08:
Hello,
added to my lwz403.yaml following function. It's not nice but working code ;-)

I like using the function. 
- after a long shower
- campfire at the neighbor's house

Maybe someone else has a similar ventilation system and can use it.

[image]

> 

packages:
  LUEFTUNGSSTUFE:            !include { file: wp_generic.yaml, vars: { property: "LUEFTUNGSSTUFE", update_interval: $interval_slow, icon: "mdi:fan" }} #0x05dd (auch Sensor! Lüfterstufe IST)

#########################################
#                                       #
#   Global variables                    #
#                                       #
#########################################
globals:
  - id: gLUEFT_AUSS_STUFE
    type: int
    restore_value: yes
    initial_value: "0"
  - id: gLUEFT_AUSS_COUNTDOWN
    type: int
    restore_value: no
    initial_value: "0"
  - id: gLUEFT_AUSS_TIME
    type: int
    restore_value: yes
    initial_value: "15"    

#########################################
#                                       #
#   Selects                             #
#                                       #
#########################################
select:
  - platform: template                   # LUEFT_AUSS_TIME erhalten an alle 4 can_id diesen Wert schreiben
    name: "LUEFT_AUSS_TIME"
    id: LUEFT_AUSS_TIME
    options:
      - "15 min"
      - "30 min"
      - "1 h"
      - "2 h"
      - "5 h"
      - "15 h"
    #initial_option: "15 min"
    restore_value: true
    optimistic: true
    #setup_priority: 100
    on_value:
      then:
      - lambda: |-
          // ESP_LOGI("lueftung","LUEFT_AUSS_TIME wurde im SELECT geändert");
          if (id(LUEFT_AUSS_TIME).active_index() == 0) {
            id(gLUEFT_AUSS_TIME) = 15;
          } else if (id(LUEFT_AUSS_TIME).active_index() == 1) {
            id(gLUEFT_AUSS_TIME) = 30;
          } else if (id(LUEFT_AUSS_TIME).active_index() == 2) {
            id(gLUEFT_AUSS_TIME) = 60;
          } else if (id(LUEFT_AUSS_TIME).active_index() == 3) {
            id(gLUEFT_AUSS_TIME) = 120;
          } else if (id(LUEFT_AUSS_TIME).active_index() == 4) {
            id(gLUEFT_AUSS_TIME) = 240;
          } else if (id(LUEFT_AUSS_TIME).active_index() == 5) {
            id(gLUEFT_AUSS_TIME) = 900;
          }
          ESP_LOGI("lueftung","LUEFT_AUSS_TIME geändert auf: %d", id(gLUEFT_AUSS_TIME));
          sendData(Kessel, Property::kLUEFT_AUSS_ZEIT_STUFE_0, static_cast<std::uint16_t>(id(gLUEFT_AUSS_TIME)));
      - delay: 250ms
      - lambda: |-
          sendData(Kessel, Property::kLUEFT_AUSS_ZEIT_STUFE_1, static_cast<std::uint16_t>(id(gLUEFT_AUSS_TIME)));
      - delay: 250ms
      - lambda: |-
          sendData(Kessel, Property::kLUEFT_AUSS_ZEIT_STUFE_2, static_cast<std::uint16_t>(id(gLUEFT_AUSS_TIME)));
      - delay: 250ms
      - lambda: |-
          sendData(Kessel, Property::kLUEFT_AUSS_ZEIT_STUFE_3, static_cast<std::uint16_t>(id(gLUEFT_AUSS_TIME)));

#########################################
#                                       #
#   Number                              #
#                                       #
#########################################
number:
  - platform: template                   # LUEFT_AUSS_STUFE selektieren
    name: "LUEFT_AUSS_STUFE" 
    optimistic: true
    min_value: 0
    max_value: 3
    step: 1
    restore_value: true
    #initial_value: float(gLUEFT_AUSS_STUFE)
    set_action:
      - lambda: |-
          id(gLUEFT_AUSS_STUFE) = x;
          ESP_LOGI("lueftung","LUEFT_AUSS_STUFE geändert auf: %d", id(gLUEFT_AUSS_STUFE));

#########################################
#                                       #
#   Button                              #
#                                       #
#########################################
button:
  - platform: template                   # ausserplanmäßige Lüftung starten mit Stufe "gLUEFT_AUSS_STUFE" indem 0x05dd geschrieben wird
    name: "LUEFT_AUSS_START"
    id: LUEFT_AUSS_START
    on_press:
      - lambda: |-
          id(gLUEFT_AUSS_COUNTDOWN) = id(gLUEFT_AUSS_TIME)+1;   // +1 weil sofort subtrahiert wird...
          sendData(Kessel, Property::kLUEFTUNGSSTUFE, static_cast<std::uint16_t>(id(gLUEFT_AUSS_STUFE)));
          ESP_LOGI("lueftung","auss wurde gestartet mit Stufe: %d", id(gLUEFT_AUSS_STUFE));
          ESP_LOGI("lueftung","auss wurde gestartet mit Countdown: %d",  id(gLUEFT_AUSS_COUNTDOWN));          

#########################################
#                                       #
#   sensor                              #
#                                       #
#########################################
sensor:
  - platform: template
    name: LUEFT_AUSS_COUNTDOWN
    id: LUEFT_AUSS_COUNTDOWN
    accuracy_decimals: 0
    unit_of_measurement: "min"
    icon: mdi:clock-end
    lambda: |-
      if (id(gLUEFT_AUSS_COUNTDOWN) > 0) {
        id(gLUEFT_AUSS_COUNTDOWN) -= 1;
        ESP_LOGI("lueftung","auss countdown: %d", id(gLUEFT_AUSS_COUNTDOWN));
      }
      return id(gLUEFT_AUSS_COUNTDOWN);
    update_interval: 60s

>

kr0ner, 2025-02-10:
Nice one 👍 
I think at some point we should create a folder with yaml templates for HA for such things :)
This is the display that I've replicated in HA
[image]

But IMO stuff like that should not end up on the target. Not sure how I feel about adding this as a package to the repo.

Dobbiiey, 2025-03-09:
Currenly I think, that I will kick this implemantion on ESP32 side out - it's not working reliable (don't know why).
Additional checks (- did the command is exepted by heatpump or not -) could be done in HomeAssistant more easy and countdown on ESP32 side are maybe unneeded load...

I plan to integrate a "generic CAN send - action / script or similar" in HomeAssistant. (This would maybe help doing quick tests without "compiling and flashing new  ESP32 software"...)
Do you have an idea how to setup an interface to esp32 ?
Is there a nice possibility to send the 8x HEX Values (or combination) from HomeAssistant to ESP32 nicely ?

kr0ner, 2025-03-10:
> Do you have an idea how to setup an interface to esp32 ?
> Is there a nice possibility to send the 8x HEX Values (or combination) from HomeAssistant to ESP32 nicely ?

check this out: https://github.com/kr0ner/OneESP32ToRuleThemAll/pull/146

Dobbiiey, 2025-03-11:
both actions works nicely at my device.

Changing values (with mapping "out of the box" & "auto-querie after 10s")
[image]

Sending totally free stuff (no mapping, no querie):
[image]

"this is the way"

thanks !!

kr0ner, 2025-07-16:
> **Lüften** I'd like to understand this how it works, because my THZ 5.5 eco doesn't support cooling with "Kühlmodus", on very hot days I've set the Party Mode from 3:00 - 07:00 AM, but could to exchange the warm air with cold.
> 
> **Other topic:** I did the same with the Icons, they simply show up on my dashboard. This is very simple done as they only show up when the expectet status comes up
> 
> <img alt="Image" width="261" height="170" src="https://private-user-images.githubusercontent.com/77236661/467167733-67cec692-e7a3-4342-a966-87acfe0cf75e.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTI2OTU3MTYsIm5iZiI6MTc1MjY5NTQxNiwicGF0aCI6Ii83NzIzNjY2MS80NjcxNjc3MzMtNjdjZWM2OTItZTdhMy00MzQyLWE5NjYtODdhY2ZlMGNmNzVlLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTA3MTYlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwNzE2VDE5NTAxNlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWI2MzFiZWI4YzJjZWEzYjVkZTJiOGU4NzY2NjNlOGRiOWM0MDg3ZGY2ZDNhY2ZhMTdiZTlmOWEyYzFhZmQxNzEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.gI_EYdrNfsTrxG5IRd4J8cgzJc_-pnzWeqvDLvNvm_g">

You could also dynamically set LUEFT_ZULUFT_STUFE* to 80 (this is I think the lowest value) and LUEFT_ABLUFT_STUFE* to 240? That should create underpressure in the house and suck in air through the windows 🤷

Dobbiiey, 2025-10-22:
@kr0ner maybe I should close the item - it isn't an issue to me anymore ;-) Do you agree ?

kr0ner, 2025-10-23:
@Dobbiiey I agree it is not an issue ... but nevertheless might be helpful. so feel free to keep it open ;)