kr0ner/OneESP32ToRuleThemAll #134: added "wp_4_combined_4.yaml"

Dobbiiey, 2025-02-08:
I know, not needed @ standard - but it's working nicely...

[wp_4_combined_4.yaml.txt](https://github.com/user-attachments/files/18722120/wp_4_combined_4.yaml.txt)

kr0ner, 2025-02-10:
Can you please change the title to reflect the content? Since I will not add this the bot will mark it as stale and close it. If there is a good title, people eventually will find it again :)

zimbo86, 2025-04-07:
pretty good work, thanks @Dobbiiey 
Implemented it today, lets see what my weird time sensors will to with it.
You solution is very clever and i´m optimistic, that it will fullfill my needs.

Theres a lack of calculating the deltas at midnight, but as it is a total_increasing sensor this is OK for me.
As well as the sensor is only updated if the value is greater than before the difference should be pretty small.

Will report what happens the next days :)

zimbo86, 2025-04-08:
@Dobbiiey  works like a charm :)

[image]

My next step is to add a combined sensor, which sums WW + HC into a single value

zimbo86, 2025-04-09:
just added this sensor:
```
  - platform: template
    name: consumption_combined_total
    id: consumption_combined_total
    unit_of_measurement: kWh
    icon: "mdi:lightning-bolt"
    accuracy_decimals: 3
    state_class: "total_increasing"
    device_class: energy
    filters:
      - round: 4
    update_interval: $interval_very_slow
    lambda: |-
      ESP_LOGI("consumption_combined_total", "Initiate Update");
      if (isnan(id(consumption_hc_total).state) && isnan(id(consumption_dhw_total).state)){
        return id(consumption_hc_total).state + id(consumption_dhw_total).state;
      }
```

[image]

need it for a energy dashboard, which is limited to two "external" energy devices:

[image]

Dobbiiey, 2025-10-22:
@kr0ner I'm still using this file "wp_4_combined_4.yaml" but anyhow we could close this item. Do you agree ?

closed 2025-11-02 by Dobbiiey