kr0ner/OneESP32ToRuleThemAll #135: help implementing Programms (Heating, Fan, WarmWater)

Dobbiiey, 2025-02-08:
Next implementation I'd like to add for LWZ403 would be the Programs ( I only use Mo.-So. Prog1 )

Example:
         // '14a0' // HK1 Programm Mo.-So. // x=5: DayStart*15 [min]  x=6: NightStart*15[min]
and:
        // '17a0',    // Warmwasserprogramm Mo.-So.
	// '1da0',    // Lüfterprogramm Mo.-So.

Maybe et_time_domain is correct Type, but "how to work with it"?
    PROPERTY(HEIZPROG_HK1_MO_SO_TAG_ENDE, 14a0, Type::et_time_domain);   

Is there any example-code for et_time_domain ?

I just know, that this one address '14a0' handles both values StartDay & StartNight...
And writing to the address means always: "writing both values" ;-))

Are these 2 values controlable by 2 template - number in the end?

kr0ner, 2025-02-10:
Hi @Dobbiiey 
I never looked too closely at those settings. There are so many problems arising when you want to automate stuff in HA and the heatpump switches from day to night setting. So I ended up using https://github.com/nielsfaber/scheduler-card which is a great piece of software. So now I can set the schedules in HA and not having to use this weird heat pump display ^^
IMO this whole thing only makes sense once you can set the schedule of the heatpump with a nice UI in HA similar to the one from nielsfaber

kr0ner, 2025-02-10:
If you still want to have this supported I'd need let's say 10 samples of schedules set in the heatpump and the corresponding raw values on the bus 😉

zimbo86, 2025-03-07:
may I add some information that could be useful here:

This is the code from fhem, so 14a0 is correct
``` 
  "programHC1_Mo-Fr_0"		=> {cmd2=>"0B1480", argMin =>  "00:00",	argMax =>"24:00",	type =>"7prog",  unit =>""},
  "programHC1_Mo-Fr_1"		=> {cmd2=>"0B1481", argMin =>  "00:00",	argMax =>"24:00",	type =>"7prog",  unit =>""},
  "programHC1_Mo-Fr_2"		=> {cmd2=>"0B1482", argMin =>  "00:00",	argMax =>"24:00",	type =>"7prog",  unit =>""},
  "programHC1_Sa-So_0"		=> {cmd2=>"0B1490", argMin =>  "00:00",	argMax =>"24:00",	type =>"7prog",  unit =>""},
  "programHC1_Sa-So_1"		=> {cmd2=>"0B1491", argMin =>  "00:00",	argMax =>"24:00",	type =>"7prog",  unit =>""},
  "programHC1_Sa-So_2"		=> {cmd2=>"0B1492", argMin =>  "00:00",	argMax =>"24:00",	type =>"7prog",  unit =>""},
  "programHC1_Mo-So_0"		=> {cmd2=>"0B14A0", argMin =>  "00:00",	argMax =>"24:00",	type =>"7prog",  unit =>""},
  "programHC1_Mo-So_1"		=> {cmd2=>"0B14A1", argMin =>  "00:00",	argMax =>"24:00",	type =>"7prog",  unit =>""},
  "programHC1_Mo-So_2"		=> {cmd2=>"0B14A2", argMin =>  "00:00",	argMax =>"24:00",	type =>"7prog",  unit =>""},
```
There was a comment behind these lines:

`#1 is monday 0 is first prog; start and end; value 1Ch 28dec is 7 ; value 1Eh 30dec is 7:30`

> I just know, that this one address '14a0' handles both values StartDay & StartNight... And writing to the address means always: "writing both values" ;-))

In my opinion this are not StartDay and StartNight, its StartDay and EndDay.

So from the above examle 1C1E would be 7:30 Uhr

I will fetch some data later on the WP directly

zimbo86, 2025-03-07:
okay, its getting closer.
00:00 - 02:00 returns "8"
00:00 - 01:00 returns "4"
00:00 - 16:30 returns "66"

so with a beginning of 0:00 its pretty easy, as the values are quarters (like they can be set manual on the WP display)

2 hours = 8 quarters
16 hours = 64 quarters, 30 minutes = 2 Quarters = 66 quarters

zimbo86, 2025-03-07:
01:00 - 02:00 returns "1032"
1032 in hex is 0x0408  => 04 08 => 4 quarters start, 8 quarters end

kr0ner, 2025-03-08:
Good job finding this stuff out :)

How do you plan to visualize and schedule the date/times ?
I faced the same problem in the beginning and then decided to drop the idea and use https://github.com/nielsfaber/scheduler-card instead.
It could also be "hacked" to send the schedule to the heatpump instead of scheduling the stuff in HA

Dobbiiey, 2025-03-09:
> 01:00 - 02:00 returns "**1032**" `1032 bytewise is 0x0408 => 04 08 => Start: 4 quarters , End: 8 quarters` 02:00 - 03:00 returns "**2060**" `2060 bytewise is 0x080C => 08 0C => Start: 8 quarters , End: 12 quarters`
> 
> That works for "small" values.
> 
> However when I use 18:00 - 19:00 **17484** `0x445C => 44 5C => Start: 68 quarters , End: 92 quarters which makes no sense..` But correct would be 0x484C (dec 18508, 72 quarters, 76 quarters.
> 
> 17:15 - 18:15 **17481** `0x445C => 44 49=> Start: 68 quarters , End: 73 quarters..` But correct would be 0x4549 (dec 17.737, 69 quarters, 73 quarters.

This is strange - I was pretty sure, that it's working "reliable with the logic"... (but I had times like 7:00 - 18:00  // 9:00 - 15:00)
I will do tests similar to your examples in the next week...

@zimbo86 : you are right with: "StartDay and StartNight, its StartDay and EndDay" -> we are talking about the same. Sorry not beeing precise ;-)  
(StartDay = TimeAtDayStarts // EndDay = TimeAtDayEnds && TimeAtNightStarts)

Do you have progress to "write the programs with HomeAssistant/ESP32"?

zimbo86, 2025-03-10:
some more values....

[image]

zimbo86, 2025-03-11:
hmm I dont get the logic after 12am. Maybe someone else has a clue? It seems that the starting time always matches. But the second value differs in hours, minutues are fine.

[image]

Dobbiiey, 2025-03-11:
[image]

Hi @zimbo86 , could you please verify/check dec2hex calculation ? (My excel shows different values).
thanks

zimbo86, 2025-03-11:
> [image]
> 
> Hi [@zimbo86](https://github.com/zimbo86) , could you please verify/check dec2hex calculation ? (My excel shows different values). thanks

wtf, that costs me hours. Thank you!
i put all the data in chatgpt and it calculated the hex wrong starting at 12:00

[image]

So the solution is pretty simple...
Now we need a logic to write this to WP (and I think this was the basic question of this issue, sorry)

zimbo86, 2025-03-21:
hmm, it seems that a lot of this work is already done. As @Dobbiiey  mentioned, et_time_domain is already implemented and the correct type for this.

Just a quick test for a value brings up the correct time entered in fhem or heatpump display:

```
          // TEXT SENSOR TEST
          CallbackHandler::instance().addCallback(std::make_pair(Kessel,Property::kprogramDHW_Mo_So_0),[](const SimpleVariant& value){
            id(programm_test_sensor).publish_state(value);
          });

          // initial queueing of the Wh and kWh value
          queueRequest(Kessel, Property::kprogramDHW_Mo_So_0);

text_sensor:
  - platform: template
    id: programm_test_sensor
    name: programm_test_sensor
    icon: "mdi:alert-circle"
```

[image]

zimbo86, 2025-03-21:
So displaying these values is pretty easy, matches now my fhem values.
[image]

Next is setting these values..

zimbo86, 2025-03-21:
```
    PROPERTY(programDHW_Mo_So_0, 0x017A0, Type::et_time_domain);
    PROPERTY(programDHW_Mo_So_1, 0x017A1, Type::et_time_domain);
    PROPERTY(programDHW_Mo_So_2, 0x017A2, Type::et_time_domain);
    PROPERTY(programHC_Mo_So_0, 0x014A0, Type::et_time_domain);
    PROPERTY(programHC_Mo_So_1, 0x014A1, Type::et_time_domain);
    PROPERTY(programHC_Mo_So_2, 0x014A2, Type::et_time_domain);
    PROPERTY(programFan_Mo_So_0, 0x01DA0, Type::et_time_domain);
    PROPERTY(programFan_Mo_So_1, 0x01DA1, Type::et_time_domain);
    PROPERTY(programFan_Mo_So_2, 0x01DA2, Type::et_time_domain);
```

wp_program.yaml
```
esphome:
  on_boot:
    priority: -100
    then:
      - lambda: |-
          // TEXT SENSOR TEST
          CallbackHandler::instance().addCallback(std::make_pair(Kessel,Property::k${property}),[](const SimpleVariant& value){
            id(${property}).publish_state(value);
          });

          // initial queueing of the Wh and kWh value
          queueRequest(Kessel, Property::k${property});

text_sensor:
  - platform: template
    id: ${property}
    name: ${property}
    update_interval: never
    icon: "mdi:clock-time-eight-outline"
```

common.yaml
```
 programDHW_Mo_So_0:              !include { file: wp_program.yaml, vars: { property: "programDHW_Mo_So_0"}}   
  programDHW_Mo_So_1:              !include { file: wp_program.yaml, vars: { property: "programDHW_Mo_So_1"}} 
  programDHW_Mo_So_2:              !include { file: wp_program.yaml, vars: { property: "programDHW_Mo_So_2"}} 

  programHC_Mo_So_0:              !include { file: wp_program.yaml, vars: { property: "programHC_Mo_So_0"}}   
  programHC_Mo_So_1:              !include { file: wp_program.yaml, vars: { property: "programHC_Mo_So_1"}} 
  programHC_Mo_So_2:              !include { file: wp_program.yaml, vars: { property: "programHC_Mo_So_2"}} 

  programFan_Mo_So_0:              !include { file: wp_program.yaml, vars: { property: "programFan_Mo_So_0"}}   
  programFan_Mo_So_1:              !include { file: wp_program.yaml, vars: { property: "programFan_Mo_So_1"}} 
  programFan_Mo_So_2:              !include { file: wp_program.yaml, vars: { property: "programFan_Mo_So_2"}} 
```

however adding all these sensors brings my esp on its limit.

kr0ner, 2025-03-22:
> however adding all these sensors brings my esp on its limit.

What exactly is happening? Out of RAM ? Out of ROM? Issues during runtime?
Are you running on an ESP32?

zimbo86, 2025-03-24:
> > however adding all these sensors brings my esp on its limit.
> 
> What exactly is happening? Out of RAM ? Out of ROM? Issues during runtime? Are you running on an ESP32?

its not releated to this project, as I only use the Stiebel part ;)
Flashed it, never comes only. But was able to OTA again with some less senors and it came back online again. 
Will have to do it via USB to get the logs..

kr0ner, 2025-03-24:
Still would be interesting what setup you are using that runs into this kind of problems 😉 
There is still room for improvement. Maybe we can add some flags or optimize here and there

Dobbiiey, 2025-03-25:
> So displaying these values is pretty easy, matches now my fhem values.

cool  thanks @zimbo86  - I copied it & works fine.
I just modified my "heating" because it's "HK1" instead of "Kessel" (wp_program.yaml)

```
defaults:
  target: "Kessel"

esphome:
  on_boot:
    priority: -100
    then:
      - lambda: |-
          // TEXT SENSOR TEST
          CallbackHandler::instance().addCallback(std::make_pair(${target},Property::k${property}),[](const SimpleVariant& value){
            id(${property}).publish_state(value);
          });
          queueRequest(${target}, Property::k${property});

text_sensor:
  - platform: template
    id: ${property}
    name: ${property}
    update_interval: never
    icon: "mdi:clock-time-eight-outline"
    lambda: |-
        queueRequest(${target}, Property::k${property});
        return {};
```

kr0ner, 2025-03-25:
> > > however adding all these sensors brings my esp on its limit.
> > 
> > 
> > What exactly is happening? Out of RAM ? Out of ROM? Issues during runtime? Are you running on an ESP32?
> 
> Hello, again a stupid ESP newby question ;-) How to check these "ESP32 health values like RAM, ROM..." ? ( I have similar problems if I have >100 Entities --> after flashing the *.elf file ESP32 is "offline" / 100 works /101 crashes)

When you compile, what numbers are printed in your case?

[image]

Dobbiiey, 2025-03-25:
Hi, again a wrong assumption ;-) & to your information

[image]

( today, 101 entities works fine... So crashes are reasoned maybe by moon position )

zimbo86, 2025-03-26:
[image]

@kr0ner  Wow, your 73% Flash is massive, i read somewhere that values above 65% are critical?

kr0ner, 2025-03-26:
@zimbo86 I think you are confusing RAM and Flash ;) 
You can use 100% of the available flash.

kr0ner, 2025-03-27:
@zimbo86 @Dobbiiey check this out:
https://github.com/kr0ner/OneESP32ToRuleThemAll/pull/154

zimbo86, 2025-04-01:
In my case i´m in a reboot loop.
Tried to set -DCONFIG_ARDUINO_LOOP_STACK_SIZE=32768
however this doesnt change anything..

```
[10:50:18]***ERROR*** A stack overflow in task loopTask has been detected.
[10:50:18]
[10:50:18]***ERROR*** A stack overflow in task loopTask has been detected.
[10:50:18]
[10:50:18]
[10:50:18]Backtrace:0x601f7ffd:0x8011a8b6 |<-CORRUPTED
[10:50:18]
[10:50:18]
[10:50:18]
[10:50:18]
[10:50:18]ELF file SHA256: 0000000000000000
[10:50:18]
[10:50:18]Rebooting...
[10:50:18]ets Jun  8 2016 00:22:57
[10:50:18]
[10:50:18]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[10:50:18]configsip: 0, SPIWP:0xee
[10:50:18]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[10:50:18]mode:DIO, clock div:2
[10:50:18]load:0x3fff0018,len:4
[10:50:18]load:0x3fff001c,len:1044
[10:50:18]load:0x40078000,len:8896
[10:50:18]load:0x40080400,len:5828
[10:50:18]entry 0x400806ac
```

zimbo86, 2025-04-01:
> cool thanks [@zimbo86](https://github.com/zimbo86) - I copied it & works fine. I just modified my "heating" because it's "HK1" instead of "Kessel" (wp_program.yaml)
>
> ```
> defaults:
>   target: [...]

your correct, 
HC values are not Kessel but HK1.
Thanks!

[image]

Dont worry about the strange time spans, my heatpump is running another time.. (that makes calculating daily energy values for energy dashboard pretty hard....^^)

Dobbiiey, 2025-04-03:
currently we can't edit/set program values by ESP (reading is working)...
But with function: https://github.com/kr0ner/OneESP32ToRuleThemAll/issues/133#issuecomment-2715952757  I created a script in HomeAssistant to edit/set the values:

[image]

[image]

script:
```
sequence:
  - variables:
      adr1_write: |
        {%- if typ == "Heizung" %}
          96
        {%- elif typ == "Lüftung" %}
          48
        {%- elif typ == "Warmwasser" %}
          48
        {%- else %}
          error
        {%- endif %}
      adr1_read: |
        {%- if typ == "Heizung" %}
          97
        {%- elif typ == "Lüftung" %}
          49
        {%- elif typ == "Warmwasser" %}
          49
        {%- else %}
          error
        {%- endif %}
      adr2: |
        {%- if typ == "Heizung" %}
          1
        {%- elif typ == "Lüftung" %}
          0
        {%- elif typ == "Warmwasser" %}
          0
        {%- else %}
          error
        {%- endif %}
      el1: |
        {%- if typ == "Heizung" %}
          20
        {%- elif typ == "Lüftung" %}
          29
        {%- elif typ == "Warmwasser" %}
          23
        {%- else %}
          error
        {%- endif %}
      el2: |
        160
      t1sr: |
        {%- set t1 = start_time %}  
        {%- set t1s = t1.split(':') %}
        {{ ( ( t1s[0]|int(0)*60 + t1s[1]|int(0) )/15 )|int(0) }}
      t2sr: |
        {%- set t2 = end_time %}  
        {%- set t2s = t2.split(':') %}
        {{ ( ( t2s[0]|int(0)*60 + t2s[1]|int(0) )/15 )|int(0) }}
  - action: esphome.stiebel_can_send_raw_data
    metadata: {}
    data:
      data: >
        [{{ adr1_write }},{{ adr2 }}, 250, {{ el1 }},{{ el2 }}, {{ t1sr }},{{
        t2sr }}]
  - action: logbook.log
    metadata: {}
    data:
      name: program_set
      entity_id: script.program_set_v01
      message: >
        Nutzer Eingabe Zeit: {{ start_time }} bis {{ end_time }} to: {{ typ }}
        ... writing: [{{ adr1_write }},{{ adr2 }}, 250, {{ el1 }},{{ el2 }}, {{
        t1sr }},{{ t2sr }}]
  - delay:
      hours: 0
      minutes: 0
      seconds: 10
      milliseconds: 0
  - action: esphome.stiebel_can_send_raw_data
    metadata: {}
    data:
      data: |
        [{{ adr1_read }},{{ adr2 }}, 250, {{ el1 }},{{ el2 }}, 0, 0]
  - action: logbook.log
    metadata: {}
    data:
      name: program_set
      entity_id: script.program_set_v01
      message: >
        querie ... sending: [{{ adr1_read }},{{ adr2 }}, 250, {{ el1 }},{{ el2
        }}, 0, 0}]
fields:
  start_time:
    selector:
      time: {}
    name: start_time
    description: Uhrzeit zu der Tagbetrieb startet
    required: true
    default: "06:00:00"
  end_time:
    selector:
      time: {}
    name: end_time
    description: Uhrzeit zu der Tagbetrieb endet
    required: true
    default: "18:00:00"
  typ:
    selector:
      select:
        options:
          - Lüftung
          - Warmwasser
          - Heizung
        multiple: false
    name: typ
    description: Definition des Programmes
    required: true
    default: Lüftung
alias: program_set_v01
description: ""
icon: mdi:progress-check

```

Dobbiiey, 2025-04-03:
> (that makes calculating daily energy values for energy dashboard pretty hard....^^)

@zimbo86 :
Maybe check for a "total-sum-sensor" (total increasing && with daily) (daily stuff then could be done by "HomeAssistant"):
https://github.com/kr0ner/OneESP32ToRuleThemAll/issues/134#issue-2840251524

background: https://github.com/kr0ner/OneESP32ToRuleThemAll/issues/117#issuecomment-2621377503

zimbo86, 2025-04-03:
> currently we can't edit/set program values by ESP (reading is working)... But with function: [#133 (comment)](https://github.com/kr0ner/OneESP32ToRuleThemAll/issues/133#issuecomment-2715952757) I created a script in HomeAssistant to edit/set the values:
> 
> [image]
> 
> [image]
> 
> script:
> 
> ```
> sequence:
>   - variables:
>       adr1_write: |
>         {%- if typ == "Heizung" %}
>           96
>         {%- elif typ == "Lüftung" %}
>           48
>         {%- elif typ == "Warmwasser" %}
>           48
>         {%- else %}
>           error
>         {%- endif %}
>       adr1_read: |
>         {%- if typ == "Heizung" %}
>           97
>         {%- elif typ == "Lüftung" %}
>           49
>         {%- elif typ == "Warmwasser" %}
>           49
>         {%- else %}
>           error
>         {%- endif %}
>       adr2: |
>         {%- if typ == "Heizung" %}
>           1
>         {%- elif typ == "Lüftung" %}
>           0
>         {%- elif typ == "Warmwasser" %}
>           0
>         {%- else %}
>           error
>         {%- endif %}
>       el1: |
>         {%- if typ == "Heizung" %}
>           20
>         {%- elif typ == "Lüftung" %}
>           29
>         {%- elif typ == "Warmwasser" %}
>           23
>         {%- else %}
>           error
>         {%- endif %}
>       el2: |
>         160
>       t1sr: |
>         {%- set t1 = start_time %}  
>         {%- set t1s = t1.split(':') %}
>         {{ ( ( t1s[0]|int(0)*60 + t1s[1]|int(0) )/15 )|int(0) }}
>       t2sr: |
>         {%- set t2 = end_time %}  
>         {%- set t2s = t2.split(':') %}
>         {{ ( ( t2s[0]|int(0)*60 + t2s[1]|int(0) )/15 )|int(0) }}
>   - action: esphome.stiebel_can_send_raw_data
>     metadata: {}
>     data:
>       data: >
>         [{{ adr1_write }},{{ adr2 }}, 250, {{ el1 }},{{ el2 }}, {{ t1sr }},{{
>         t2sr }}]
>   - action: logbook.log
>     metadata: {}
>     data:
>       name: program_set
>       entity_id: script.program_set_v01
>       message: >
>         Nutzer Eingabe Zeit: {{ start_time }} bis {{ end_time }} to: {{ typ }}
>         ... writing: [{{ adr1_write }},{{ adr2 }}, 250, {{ el1 }},{{ el2 }}, {{
>         t1sr }},{{ t2sr }}]
>   - delay:
>       hours: 0
>       minutes: 0
>       seconds: 10
>       milliseconds: 0
>   - action: esphome.stiebel_can_send_raw_data
>     metadata: {}
>     data:
>       data: |
>         [{{ adr1_read }},{{ adr2 }}, 250, {{ el1 }},{{ el2 }}, 0, 0]
>   - action: logbook.log
>     metadata: {}
>     data:
>       name: program_set
>       entity_id: script.program_set_v01
>       message: >
>         querie ... sending: [{{ adr1_read }},{{ adr2 }}, 250, {{ el1 }},{{ el2
>         }}, 0, 0}]
> fields:
>   start_time:
>     selector:
>       time: {}
>     name: start_time
>     description: Uhrzeit zu der Tagbetrieb startet
>     required: true
>     default: "06:00:00"
>   end_time:
>     selector:
>       time: {}
>     name: end_time
>     description: Uhrzeit zu der Tagbetrieb endet
>     required: true
>     default: "18:00:00"
>   typ:
>     selector:
>       select:
>         options:
>           - Lüftung
>           - Warmwasser
>           - Heizung
>         multiple: false
>     name: typ
>     description: Definition des Programmes
>     required: true
>     default: Lüftung
> alias: program_set_v01
> description: ""
> icon: mdi:progress-check
> ```

nice work, will try the next days.
Just had i quick view, but it wont handle two timespans?

zimbo86, 2026-02-02:
> [@SvenJ76](https://github.com/SvenJ76) do you want to program schedules into your heatpump? If not, I'd suggest to use scheduler integration for Home Assistant and do it via the UI ... it is much more powerful ;)

i would not do that, as every dynamically change of the temps will write to EEPROM everytime.

> > [@SvenJ76](https://github.com/SvenJ76) do you want to program schedules into your heatpump? If not, I'd suggest to use scheduler integration for Home Assistant and do it via the UI ... it is much more powerful ;)
> 
> Actually I want to program the "party schedule" from Home Assistant - I want to use it as "on demand" increase for the ventilation based on too high humidity or too high CO2 values.

Youre just talking about the fan? 
Why not just use the trigger the manual fan to speed 3?
Duration of this can also be set in the menu..

[image]

[image]

(I use my own sensor names..)