kr0ner/OneESP32ToRuleThemAll #138: not getting any data anymore

skycryer, 2025-02-17:
i had the software running some month now, everything was working. Today I saw that I did not get any updates for outside temp and heating power usage anymore since 2 days. So I just restarted the esp like I did before but it did not get any updates anymore from the heating pump. SO I tried like before cutting power on the esp and heating pump and started one after the other again, with no change. 

I can still send changes like fan speed, program, room temp and humidity, so I think the mod bus is still connected correct, but no Idea why it does not getting any data. In the logs I see a lot of requesting data but no updates, any idea?

I already did an update also to esp and your project, without any changes.

flom89, 2025-02-17:
I think you mean CAN BUS. If there are no received messages, check your hardware. Might be an issue with your can or with the SPI.

kr0ner, 2025-02-17:
Yep, it is a CAN Bus ;) 
Usually those kind of errors were related to broken MCP2515 or bad cable.
Its weird that this now happens suddenly. Do you have any spare ESP32 or MCP2515?

But you also mentioned you've updated the code. Did you maybe include the wrong yaml during rebase?

closed 2025-02-17 by skycryer

kr0ner, 2025-02-18:
How did you fix it?

skycryer, 2025-02-18:
It fixed itself after I opened the box where I have my esp and Can Bus Module and closed it again It just worked again. Not sure what was the problem, the cables where all still connected.