kr0ner/OneESP32ToRuleThemAll #139: ESPHome custom component removed from support

eugenpirogoff, 2025-02-19:
Hello folks, 

sadly ESPHome removed custom components in the newest version of ESPHome (2025.2.0). So it will not work with this project.

More Info here: https://esphome.io/guides/contributing#a-note-about-custom-components

[image]

kr0ner, 2025-02-19:
that sucks big time 💩 
Will take a while until this stuff is ported ... at least the code should be the same.

eugenpirogoff, 2025-02-19:
Yep. I thought so too. Please verify in your installation. Maybe "i am holding it wrong"

kr0ner, 2025-02-19:
I checked the docs and there are no custom components left. So I need to adopt

closed 2025-02-21 by kr0ner

kr0ner, 2025-03-07:
It's only slow cause your settings are wrong. It takes one cycle of the Heizkreispumpe to apply the settings. During heating period this is almost always on so changes are instant

zimbo86, 2025-03-07:
> It's only slow cause your settings are wrong. It takes one cycle of the Heizkreispumpe to apply the settings. During heating period this is almost always on so changes are instant

hmm I think you get me wrong.
of cause the changes made via climate controller (or number) are instant.

My point is you dont change the current water temperature with these settings, you change the parameters when the heatpump starts a WW cycle. Its nice to be able to set these values remotley but if I got suiting parameters to a house these values are not touched for month..

kr0ner, 2025-03-07:
But why limit the functionalty just because you don't use it? 😅 There are plenty of people who change the values multiple times a day.

With this argumentation one can also ditch the day/night schedules cause they can be implemented via Home Assistant 😉

PR #140 (2025-02-21): Bring back custom components
Fixes: #139
commit 6ae5bba3 (2025-02-21): Bring back custom components

Fixes: #139