kr0ner/OneESP32ToRuleThemAll #141: THZ504 - missing parameters

zimbo86, 2025-02-27:
in fhem there is a key called sGlobal with the following values:

`outsideTemp: 3.7 flowTemp: 30.5 returnTemp: 26.9 hotGasTemp: 59 dhwTemp: 47.8 flowTempHC2: -60 evaporatorTemp: -0.3 condenserTemp: 30.4 mixerOpen: 0 mixerClosed: 0 heatPipeValve: 0 diverterValve: 0 dhwPump: 0 heatingCircuitPump: 1 solarPump: 0 compressor: 1 boosterStage3: 0 boosterStage2: 0 boosterStage1: 0 highPressureSensor: 0 lowPressureSensor: 0 evaporatorIceMonitor: 1 signalAnode: 0 evuRelease: 1 ovenFireplace: 0 STB: 0 outputVentilatorPower: 25 inputVentilatorPower: 25 mainVentilatorPower: 56 outputVentilatorSpeed: 16 inputVentilatorSpeed: 23 mainVentilatorSpeed: 20 outside_tempFiltered: 5.3 relHumidity: 40.6 dewPoint: 0 P_Nd: 7.15 P_Hd: 18.22 actualPower_Qc: 2808.639 actualPower_Pel: 566.709 collectorTemp: -60 insideTemp: -60 windowOpen: 0 quickAirVent: 0 flowRate: 16.77 p_HCw: 1.4 humidityAirOut: 34.66`

the matching perl code is:
```
  "FBglob" => [["outsideTemp: ",	8, 4, "hex2int", 10],	[" flowTemp: ",		12, 4, "hex2int", 10],
	      [" returnTemp: ",		16, 4, "hex2int", 10],	[" hotGasTemp: ", 	20, 4, "hex2int", 10],
	      [" dhwTemp: ",	 	24, 4, "hex2int", 10],	[" flowTempHC2: ",	28, 4, "hex2int", 10],
	      [" evaporatorTemp: ",	36, 4, "hex2int", 10],	[" condenserTemp: ",	40, 4, "hex2int", 10],
	      [" mixerOpen: ",		45, 1, "bit0", 1],	[" mixerClosed: ",	45, 1, "bit1", 1],
	      [" heatPipeValve: ",	45, 1, "bit2", 1],	[" diverterValve: ",	45, 1, "bit3", 1],
	      [" dhwPump: ",		44, 1, "bit0", 1],	[" heatingCircuitPump: ",44, 1, "bit1", 1],
	      [" solarPump: ",		44, 1, "bit3", 1],	[" compressor: ",	47, 1, "bit3", 1],
	      [" boosterStage3: ",	46, 1, "bit0", 1],	[" boosterStage2: ",	46, 1, "bit1", 1],
	      [" boosterStage1: ",	46, 1, "bit2", 1],	[" highPressureSensor: ",49, 1, "nbit0", 1],
	      [" lowPressureSensor: ",	49, 1, "nbit1", 1],	[" evaporatorIceMonitor: ",49, 1, "bit2", 1],
	      [" signalAnode: ",	49, 1, "bit3", 1],	[" evuRelease: ",	48, 1, "bit0", 1],
	      [" ovenFireplace: ",	48, 1, "bit1", 1],	[" STB: ",		48, 1, "bit2", 1],
	      [" outputVentilatorPower: ",50, 4, "hex", 10],	[" inputVentilatorPower: ",54, 4, "hex", 10],	[" mainVentilatorPower: ",	58, 4, "hex", 10],
	      [" outputVentilatorSpeed: ",62, 4, "hex", 1],	[" inputVentilatorSpeed: ",66, 4, "hex", 1],		[" mainVentilatorSpeed: ",	70, 4, "hex", 1],
	      [" outside_tempFiltered: ",74, 4, "hex2int", 10],	[" relHumidity: ",	78, 4, "hex2int", 10],
	      [" dewPoint: ",		82, 4, "hex2int", 10],
	      [" P_Nd: ",		86, 4, "hex2int", 100],	[" P_Hd: ",		90, 4, "hex2int", 100],
	      [" actualPower_Qc: ",	94, 8, "esp_mant", 1],	[" actualPower_Pel: ",	102, 8, "esp_mant", 1],
	      [" collectorTemp: ",	4,  4, "hex2int", 10],	[" insideTemp: ",	32, 4, "hex2int", 10], 
	      [" windowOpen: ",		47, 1, "bit2", 1], 	# board X18-1 clamp X4-FA (FensterAuf): window open - signal out 230V
	      [" quickAirVent: ",	48, 1, "bit3", 1], 	# board X15-8 clamp X4-SL (SchnellLüftung): quickAirVent - signal in 230V
	      [" flowRate: ",		110, 4, "hex", 100],	# board X51 sensor P5 (on newer models B1 flow temp as well) changed to l/min as suggested by TheTrumpeter Antwort #771
	      [" p_HCw: ",		114, 4, "hex", 100],	# board X4-1..3 sensor P4 HC water pressure
	      [" humidityAirOut: ",	154, 4, "hex", 100]	# board X4-4..6 sensor B15
          ],
```

here is the complete .pm file: 
https://github.com/mhop/fhem-mirror/blob/master/fhem/FHEM/00_THZ.pm

I´m not able to convert this into working code - maybe sbdy else can do?

zimbo86, 2025-02-27:
also the current implementation of "HEIZKREISPUMPE" is not correct, compared with heatingCircuitPump from code above, HEIZKREISPUMPE always reports "on".

[image]

kr0ner, 2025-02-28:
I also have a THZ504 and for me the HEIZKREISPUMPE seems to work.

[image]

So at least the value is updated. I think your thz504_pump_hc tracks a different entity. Which ID is it mapped to?

kr0ner, 2025-02-28:
With regards to the *.pm file ...
Just converting this makes no sense. Which sensors are missing in your setup? Are you really using all of the ones above?
If there is the need for a sensor to be added, please follow the instructions in the README.md and feel free to submit a PR or provide pictures and logs so that I can add the sensor ;)

zimbo86, 2025-02-28:
> With regards to the *.pm file ... Just converting this makes no sense. Which sensors are missing in your setup? Are you really using all of the ones above? If there is the need for a sensor to be [...]

dhw circuit pump for example.
i think its hard to get these values, as these are "read only" values and for my understanding are part of an answer that contains all the values above.

I try to get familar with the code, 

p75passiveCooling	010085 0A0575 1003 => 0x0575
p13GradientHC1	01001B 0B010E 1003 => 0x010e
sGlobal			0100FC FB         1003
sDisplay		        010082 0A0176 1003
sDHW			0100F4 F3          1003

kr0ner, 2025-02-28:
There is no "part of a big answer" with CAN 😅 there are only 7 Bytes per CAN message. This must be modbus or UART or sth. like that. But any sensor can be fetched via CAN ... just not in one go.

zimbo86, 2025-03-01:
ok, forget the most i wrote above - i got an outdated communications.h and BETRIEBS_STATUS was not populated properly.
This works now, however there are still some parameters missing for me - and I dont have a clue how to get them.

For example - there is a temp_outdoor_filtered, which is calculated using temp_outdoor and the specified "DÄMPFUNG AUSSENTEMPERATUR" (12h for example).
This parameter is pretty useful as it is used to calculate the VORLAUFSOLLTEMP.
I cant find this parameter at the THZs display at all, just the "normal" temp_outdoor.

In fhem it is also part of "sGlobal"..

[image]

zimbo86, 2025-03-03:
Temporary solution: I created a HA statistics sensor which creates a mean value over the last 12h.

[image]

kr0ner, 2025-03-05:
Did you also check the entries in the display after entering the service pin? A lot of entries are only visible after the correct pin has been entered.

closed 2025-03-05 by kr0ner