kr0ner/OneESP32ToRuleThemAll #147: support for Waveshare ESP32-S3 ETH Development Board

skycryer, 2025-03-10:
Does the config support the Waveshare ESP32-S3 ETH Development Board? I tried to adapt it but getting error about the spa pin already in use?

The default ethernet config in the code sample does not help me.

kr0ner, 2025-03-10:
I don't see a reason for any ESP32 Board to not being supported. They should all work. What is the exact error message? What did you change in the yamls? Is the board supported by platformIO?

skycryer, 2025-03-10:
First error was something about the kondensatorpumpe port, after commenting that out I get

  `spi` component is using interface 'SPI2_HOST'. To use W5500, you must change the `interface` on the `spi` component.

skycryer, 2025-03-10:
Is it not possible to use it for both? Are other ethernet esp32 not using the spa pin like this one? Or how could I control the canbus module and the ethernet?

kr0ner, 2025-03-10:
In theory yes ... SPI is a bus but requires explizit slave select. Also not sure how it works with a Highspeed SPI. For i2c its different. I'd recommend to use a dedicated SPI for ethernet and can.

skycryer, 2025-03-10:
How do i do that? What should the config look like with the spa for canbus module and the ethernet config?

I just testet the ethernet config separate to see if it work, and it does. SO I "only" need to get it working together 

`spi:
  id: McpSpi
  clk_pin: GPIO18
  mosi_pin: GPIO23
  miso_pin: GPIO19

#########################################
#                                       #
#   CAN configuration                   #
#                                       #
#########################################
canbus:
  - id: !extend wp_can
    platform: mcp2515
    spi_id: McpSpi
    cs_pin: GPIO15
    bit_rate: 20kbps

#########################################
#                                       #
#   Ethernet                             #
#                                       #
#########################################
ethernet:
  type: W5500
  clk_pin: GPIO13
  mosi_pin: GPIO11
  miso_pin: GPIO12
  cs_pin: GPIO14
  interrupt_pin: GPIO10
  reset_pin: GPIO9`

kr0ner, 2025-03-10:
Ethernet is most likely hardwired so just connect the MCP to a different SPI and update the pins in the config accordingly.

skycryer, 2025-03-10:
[image]

So this is the pin layout of the ethernet board. Not sure where I can connect the spa pins here.

kr0ner, 2025-03-11:
SPI
The ESP32 S3 has four SPI interfaces which are:

SPI0 used by ESP32-S3’s cache and Crypto DMA (EDMA) to access in-package or off-package flash/PSRAM
SPI1 used by the CPU to access in-package or off-package flash/PSRAM
SPI2 is a general purpose SPI controller with its own DMA channel
SPI3 is a general purpose SPI controller with access to a DMA channel shared between several peripherals
The SPI2 (VSPI) default pins are:

gpio.12 SCK Defined as SPI0_SCK
gpio.11 MOSI Defined as SPI0_MOSI
gpio.13 MISO Defined as SPI0_MISO
gpio.10 CSO Defined as SPI_CS0
These default pins run through the IOMUX instead of the GPIO matrix, and therefore have higher
performance characteristics. The maximum frequency for IOMUX pins is 80 MHz.

SPI3 does not have default pin mappings because it can be mapped to any available gpio pins.
[ESP32-S3_Pin_Reference](http://wiki.fluidnc.com/en/hardware/ESP32-S3_Pin_Reference)

[image]

This should clear things up :)

closed 2025-03-11 by kr0ner

skycryer, 2025-03-11:
sorry for reopening, I got the connection and config right now I think. No errors while building anymore and also no error in log. I can see I receive data from canbus module but something is missing I think because in home assistant nothing is shown anymore. Everything is off and empty, not sure what I am missing. Only difference is that spi was set to interface software, but that should be fine because I receive messages or not?

Here is the log file:

[logs_esphome-web-7d2a20_logs.txt](https://github.com/user-attachments/files/19179970/logs_esphome-web-7d2a20_logs.txt)

[image]
[image]

Here is my full config 
```

esphome:
  name: esp32-poe-technik
  friendly_name: ESP32-PoE Technik
  platformio_options:
    build_flags:
      - "-std=gnu++17"
    build_unflags:
      - "-std=gnu++11"
      - "-fno-rtti"

esp32:
  board: esp32-poe
  framework:
    type: arduino

# Enable logging
logger:
  level: INFO
  baud_rate: 0
  hardware_uart: UART1

# Enable Home Assistant API
api:
  encryption:
    key: !secret api_encryption_key

ota:
  platform: esphome
  password: !secret ota_password

time:
  - platform: homeassistant
    id: homeassistant_time

substitutions:
  interval_very_fast: 15s
  interval_fast: 30s
  interval_medium: 60s
  interval_slow: 5min
  interval_very_slow: 15min
  interval_once_in_a_while: 1h

  entity_room_temperature: "sensor.durchschnittstemperatur_haus_ohne_keller"
  entity_humidity: "sensor.durchschnitt_luftfeuchtigkeit_haus"

#########################################
#                                       #
#   SPI configuration                   #
#                                       #
#########################################
spi:
  id: McpSpi
  clk_pin: GPIO14
  mosi_pin: GPIO13
  miso_pin: GPIO4

#########################################
#                                       #
#   CAN configuration                   #
#                                       #
#########################################
canbus:
  - id: !extend wp_can
    platform: mcp2515
    spi_id: McpSpi
    cs_pin: GPIO15
    bit_rate: 20kbps

#########################################
#                                       #
#   Buttons                             #
#                                       #
#########################################
button:
  - platform: restart
    name: "Restart ESP32-PoE Technik"

#########################################
#                                       #
#   Ethernet Konfiguration              #
#                                       #
#########################################
ethernet:
  type: LAN8720
  mdc_pin: GPIO23
  mdio_pin: GPIO18
  clk_mode: GPIO0_OUT
  phy_addr: 0
  power_pin: GPIO12

#########################################
#                                       #
#   Packages                            #
#                                       #
#########################################
packages:
  energymeter: !include OneESP32ToRuleThemAll/yaml/energymeter.yaml
  watermeter: !include OneESP32ToRuleThemAll/yaml/watermeter.yaml
  sensors: !include OneESP32ToRuleThemAll/yaml/sensors.yaml
# thz404: !include OneESP32ToRuleThemAll/yaml/thz404.yaml
  thz504: !include OneESP32ToRuleThemAll/yaml/thz504.yaml
# thz55eco: !include OneESP32ToRuleThemAll/yaml/thz5_5_eco.yaml
# ttf07: !include OneESP32ToRuleThemAll/yaml/ttf07.yaml
# wpl13: !include OneESP32ToRuleThemAll/yaml/wpl13.yaml
  kondensatpumpe: !include OneESP32ToRuleThemAll/yaml/kondensatpumpe.yaml

#########################################
#                                       #
#   EVU Sperre                          #
#                                       #
#########################################
switch:
  - platform: gpio
    name: "EVU Sperre"
    id: evu_sperre
    pin: GPIO5
```

kr0ner, 2025-03-11:
Are you sure that the other yamls were not touched? Somehow extended ids are used for CAN ... pretty sure that is not the default

skycryer, 2025-03-11:
> Are you sure that the other yamls were not touched? Somehow extended ids are used for CAN ... pretty sure that is not the default

Do you mean this one?

[image]

This extend is also in your code. And yes ham sure no other code has been touched

kr0ner, 2025-03-11:
no, I mean this:

[image]
Your CAN is not working

skycryer, 2025-03-11:
Ah shit I thought I sorted it out. I first had the read error message but then it was gone. Maybe the SPI config is wrong?