kr0ner/OneESP32ToRuleThemAll #149: Activation of Electric Heaters

flom89, 2025-03-12:
Hello Everyone,

Is there any possibility to activate the electric heaters? I think about a scenario to activate the heaters for generation of hot water in case of sufficiant energy from Solar PV.

Heating System is a THZ 304eco which works great with OneESP32ToRuleThemAll

kr0ner, 2025-03-12:
[image]

Should work 👍🏼 you need to navigate to Fachmann -> Manuelles Einschalten -> Ne Stufe 1 - 3. Each of them should add 3.68 kW Power consumption. The energy consumption is estimated already btw. :)

Be careful when settings this. I have no clue if this bypasses the MaxWWTemp or other safety mechanisms. I'd propose to add a timer to it so that NE can never be active for more than 30 mins without an additional trigger. 

Are you willing to implement this yourself and push a PR? 🫡

flom89, 2025-03-14:
Yeah thats the way I also thought doing it. I will try to get the CAN IDs and then work on it.

kr0ner, 2025-04-15:
I finally tried it and it does not work as expected. Turning the heater on manually is just for testing and resets very quickly. My solution was to create an automation in HA that sets the heatpump to "NOTBETRIEB", the water temp to 65° and the level for the NE to 1. For this to work there is also a "SPERRZEIT NE" which defaults to 120mins ... I reduced that one to 10 mins. So even if the script below was executed successfully it takes 10 minutes until the heater is activated

```
alias: Notheizung aktivieren um PV Überschuss loszuwerden
description: ""
triggers:
  - trigger: state
    entity_id:
      - input_boolean.pv_uberschuss_in_warmwasser_umwandeln
    for:
      hours: 0
      minutes: 0
      seconds: 15
    to: null
conditions: []
actions:
  - if:
      - condition: state
        entity_id: input_boolean.pv_uberschuss_in_warmwasser_umwandeln
        state: "on"
    then:
      - action: select.select_option
        metadata: {}
        data:
          option: Notbetrieb
        target:
          entity_id: select.esp32_poe_technik_programmschalter
      - action: number.set_value
        metadata: {}
        data:
          value: "1"
        target:
          entity_id: number.esp32_poe_technik_ne_stufe_ww
      - action: climate.set_temperature
        metadata: {}
        data:
          temperature: 65
        target:
          entity_id:
            - climate.esp32_poe_technik_hot_water_day
            - climate.esp32_poe_technik_hot_water_night
    else:
      - action: select.select_option
        metadata: {}
        data:
          option: Automatik
        target:
          entity_id: select.esp32_poe_technik_programmschalter
      - action: number.set_value
        metadata: {}
        data:
          value: "0"
        target:
          entity_id: number.esp32_poe_technik_ne_stufe_ww
      - action: climate.set_temperature
        metadata: {}
        data:
          temperature: 43
        target:
          entity_id:
            - climate.esp32_poe_technik_hot_water_day
            - climate.esp32_poe_technik_hot_water_night
mode: single

```

closed 2025-05-24 by github-actions[bot]