kr0ner/OneESP32ToRuleThemAll #150: Custom Components removed from ESPHome 2025.2.0 - 19th February 2025

flom89, 2025-03-12:
It is no longer possible to flash units since the custom components were removed from ESP Home.

If you want to flash the device you get an error:

Failed config

climate.custom: [source /config/esphome/OneESP32ToRuleThemAll/yaml/wp_base.yaml:175]
  
  The "custom" component has been removed. Consider conversion to an external component.
  https://esphome.io/guides/contributing#a-note-about-custom-components.

kr0ner, 2025-03-12:
Already fixed 3 weeks ago https://github.com/kr0ner/OneESP32ToRuleThemAll/commit/6ae5bba3d2b44af754b6092bcb19ad3e098c771f ;)

git fetch && git pull origin master --rebase

closed 2025-03-12 by kr0ner

flom89, 2025-03-12:
my fault. did not thought about the easy way via git