kr0ner/OneESP32ToRuleThemAll #151: To many request or something else blocking heater controls

skycryer, 2025-03-15:
Just saw today when I needed to change the filters that I could not see the fan settings on the device itself. Also not able to reset the filter counter on device. I had to restart the heating system before it came back to control. Restarting esp did not help. 

Not sure what it is but maybe related to to many request send to the can bis of the heater?

kr0ner, 2025-03-15:
Can you please provide some logs? And the changes you have done to the config? Otherwise it is just wild guessing. Thanks

skycryer, 2025-03-16:
I only changed the spi to use a different interface and updated the ethernet part. 

[logs_esphome-web-7d2a20_logs-3.txt](https://github.com/user-attachments/files/19269578/logs_esphome-web-7d2a20_logs-3.txt)

```

esphome:
  name: heatingpump
  friendly_name: heatingpump
  platformio_options:
    build_flags:
      - "-std=gnu++17"
    build_unflags:
      - "-std=gnu++11"
      - "-fno-rtti"

esp32:  
  board: esp32-s3-devkitc-1
  framework:
    type: esp-idf

# Enable Home Assistant API
api:
  encryption:
    key: apikey
logger:
  #level: INFO
ota:
  - platform: esphome
    password: !secret ota_password

time:
  - platform: homeassistant
    id: homeassistant_time

substitutions:
  interval_very_fast: 15s
  interval_fast: 30s
  interval_medium: 60s
  interval_slow: 5min
  interval_very_slow: 15min
  interval_once_in_a_while: 1h

  entity_room_temperature: "sensor.durschnittliche_raumtemperatur"
  entity_humidity: "sensor.durchschnitt_humidity"

#########################################
#                                       #
#   Ethernet                             #
#                                       #
#########################################
ethernet:
  type: W5500
  clk_pin: GPIO13
  mosi_pin: GPIO11
  miso_pin: GPIO12
  cs_pin: GPIO14
  interrupt_pin: GPIO10
  reset_pin: GPIO9

#########################################
#                                       #
#   SPI configuration                   #
#                                       #
#########################################
spi:
  id: McpSpi
  clk_pin: GPIO39
  mosi_pin: GPIO40
  miso_pin: GPIO41
  interface: spi3

#########################################
#                                       #
#   CAN configuration                   #
#                                       #
#########################################
canbus:
  - id: !extend wp_can
    platform: mcp2515
    spi_id: McpSpi
    cs_pin: GPIO42
    bit_rate: 20kbps

#########################################
#                                       #
#   Buttons                             #
#                                       #
#########################################
button:
  - platform: restart
    name: "Restart Heatingpump"

#########################################
#                                       #
#   Packages                            #
#                                       #
#########################################
packages:
#  energymeter: !include OneESP32ToRuleThemAll/yaml/energymeter.yaml
#  watermeter: !include OneESP32ToRuleThemAll/yaml/watermeter.yaml
  sensors: !include OneESP32ToRuleThemAll/yaml/sensors.yaml
# thz404: !include OneESP32ToRuleThemAll/yaml/thz404.yaml
#  thz504: !include OneESP32ToRuleThemAll/yaml/thz504.yaml
  thz55eco: !include OneESP32ToRuleThemAll/yaml/thz5_5_eco.yaml
# ttf07: !include OneESP32ToRuleThemAll/yaml/ttf07.yaml
# wpl13: !include OneESP32ToRuleThemAll/yaml/wpl13.yaml
  #kondensatpumpe: !include OneESP32ToRuleThemAll/yaml/kondensatpumpe.yaml

#########################################
#                                       #
#   EVU Sperre                          #
#                                       #
#########################################
switch:
  - platform: gpio
    name: "EVU Sperre"
    id: evu_sperre
    pin: GPIO5

sensor:
  - platform: uptime
    name: Uptime
```

kr0ner, 2025-03-17:
From the logs I can't see an unusual amount of messages 😉 
It might be related to the recent exchange of your ESP. The CAN receiver of the display might go into error mode, if sth. is off. This can be wrong baudrate or some voltage spike during the rewiring of the can to the MCP2515. It happened to me as well in the past(2-3 times in 2 years) .. but mostly during testing. Is everything fine again after the reset? If yes, please close the issue and watch out for those kinds of problems and reopen the issue if it appears again :)

closed 2025-04-24 by github-actions[bot]