kr0ner/OneESP32ToRuleThemAll #157: COP values

MarcoH2013, 2025-04-16:
Hi Kr0ner,

I see in the yaml files the global variable gCOP_WW_TAG and gCOP_HEIZ_TAG but I don't see any function calculating them.

Is this function implemented already?

Kind regards
Marco

kr0ner, 2025-04-16:
Hi @MarcoH2013 
Yes, they are implemented already. Might take some time after an update until the values become available 

For Thz and Ttf it is done here
https://github.com/kr0ner/OneESP32ToRuleThemAll/blob/master/yaml%2Fcommon.yaml#L212

For WPL13 these are calculated by the heatpump itself

MarcoH2013, 2025-04-16:
Thanks, I looked at the WPL13 indeed. I have a very old WPMi which I try to add to your code.
I will copy from the Thz in this case. Thanks

closed 2025-04-16 by MarcoH2013