kr0ner/OneESP32ToRuleThemAll #159: Button Synchronize Time  & HEATPUMP_DATETIME

Nebnenut, 2025-04-23:
What is the purpose of the "Synchronize Time" button? My heat pump isn't currently running at the correct time. I understand that pressing the button allows me to correct the heat pump's time.
I have a Stiebel Eltron lwz 5s plus, which is identical to the Tecalor thz 5.5 eco. That's why I use thz5_5_eco.yaml.

After I press the button, I get the following message:
[canbus:053]: send to standard id=0x6a2 failed!

`[23:14:32][I][Communication:189]: Message received: Read/Write ID 0xd0 0x3c(0x6bc) for property TAG (0x0122) with raw value: 5888
[23:14:32][I][Communication:189]: Message received: Read/Write ID 0xd0 0x3c(0x6bc) for property MONAT (0x0123) with raw value: 1024
[23:14:32][I][Communication:189]: Message received: Read/Write ID 0xd0 0x3c(0x6bc) for property JAHR (0x0124) with raw value: 6400
[23:14:32][I][Communication:189]: Message received: Read/Write ID 0xd0 0x3c(0x6bc) for property STUNDE (0x0125) with raw value: 5888
[23:14:32][I][Communication:189]: Message received: Read/Write ID 0xd0 0x3c(0x6bc) for property MINUTE (0x0126) with raw value: 2048
[23:14:32][I][Communication:189]: Message received: Read/Write ID 0xd0 0x3c(0x6bc) for property FEHLERMELDUNG (0x0001) with raw value: 0
[23:14:32][I][Communication:189]: Message received: Read/Write ID 0xd0 0x3c(0x6bc) for property BETRIEBS_STATUS (0x0176) with raw value: 1
[23:14:32][I][Communication:189]: Message received: Read/Write ID 0xd0 0x3c(0x6bc) for property PROGRAMMSCHALTER (0x0112) with raw value: 2816
[23:14:37][I][QUEUE:125]: Requesting data from Kessel for VERDICHTERDREHZAHL
[23:14:37][I][Communication:189]: Message received: Read/Write ID 0xd2 0x22(0x6a2) for property VERDICHTERDREHZAHL (0x069e) with raw value: 0
[23:14:39][I][QUEUE:125]: Requesting data from Kessel for MOTORLEISTUNG
[23:14:40][I][Communication:189]: Message received: Read/Write ID 0xd2 0x22(0x6a2) for property MOTORLEISTUNG (0x06a0) with raw value: 0
[23:14:41][I][QUEUE:125]: Requesting data from Kessel for HEIZ_KUEHL_LEISTUNG
[23:14:41][I][QUEUE:125]: Requesting data from HK1 for VORLAUFISTTEMP
[23:14:41][I][Communication:189]: Message received: Read/Write ID 0xd2 0x22(0x6a2) for property HEIZ_KUEHL_LEISTUNG (0xc0ee) with raw value: 0
[23:14:41][I][Communication:189]: Message received: Read/Write ID 0xd2 0x21(0x6a1) for property AUSSENTEMP (0x000c) with raw value: 135
[23:14:41][I][Communication:189]: Message received: Read/Write ID 0xd2 0x21(0x6a1) for property SPEICHERISTTEMP (0x000e) with raw value: 474
[23:14:41][I][Communication:189]: Message received: Read/Write ID 0xd2 0x21(0x6a1) for property VORLAUFISTTEMP (0x000f) with raw value: 221
[23:14:41][I][Communication:189]: Message received: Read/Write ID 0xd2 0x22(0x6a2) for property VORLAUFISTTEMP (0x000f) with raw value: 221
[23:14:43][I][QUEUE:125]: Requesting data from Kessel for MOTORSTROM
[23:14:44][I][Communication:189]: Message received: Read/Write ID 0xd2 0x22(0x6a2) for property MOTORSTROM (0x069f) with raw value: 0
[23:14:47][I][QUEUE:125]: Requesting data from Kessel for RUECKLAUFISTTEMP
[23:14:47][I][Communication:189]: Message received: Read/Write ID 0xd2 0x22(0x6a2) for property RUECKLAUFISTTEMP (0x0016) with raw value: 224
[23:14:50][I][Communication:189]: Message received: Read/Write ID 0xd2 0x21(0x6a1) for property AUSSENTEMP (0x000c) with raw value: 135
[23:14:50][I][Communication:189]: Message received: Read/Write ID 0xd2 0x21(0x6a1) for property VORLAUFISTTEMP (0x000f) with raw value: 219
[23:14:50][I][Communication:189]: Message received: Read/Write ID 0xd2 0x21(0x6a1) for property SPEICHERISTTEMP (0x000e) with raw value: 474
[23:14:52][I][QUEUE:125]: Requesting data from Kessel for SPEICHERISTTEMP
[23:14:53][I][Communication:189]: Message received: Read/Write ID 0xd0 0x3c(0x6bc) for property MINUTE (0x0126) with raw value: 2304
[23:14:53][I][Communication:189]: Message received: Read/Write ID 0xd2 0x22(0x6a2) for property SPEICHERISTTEMP (0x000e) with raw value: 475
[23:14:55][I][QUEUE:125]: Requesting data from Kessel for VOLUMENSTROM
[23:14:55][I][Communication:189]: Message received: Read/Write ID 0xd2 0x22(0x6a2) for property VOLUMENSTROM (0x01da) with raw value: 0
[23:14:59][I][QUEUE:125]: Requesting data from Kessel for HEIZLEISTUNG_RELATIV
[23:14:59][I][Communication:189]: Message received: Read/Write ID 0xd2 0x22(0x6a2) for property HEIZLEISTUNG_RELATIV (0x069a) with raw value: 20
[23:15:01][I][Communication:189]: Message received: Read/Write ID 0xd2 0x21(0x6a1) for property AUSSENTEMP (0x000c) with raw value: 135
[23:15:01][I][Communication:189]: Message received: Read/Write ID 0xd2 0x21(0x6a1) for property VORLAUFISTTEMP (0x000f) with raw value: 219
[23:15:01][I][Communication:189]: Message received: Read/Write ID 0xd2 0x21(0x6a1) for property SPEICHERISTTEMP (0x000e) with raw value: 473
[23:15:02][I][QUEUE:125]: Requesting data from Kessel for MOTORSPANNUNG
[23:15:02][I][QUEUE:125]: Requesting data from Kessel for SAMMLERISTTEMP
[23:15:02][I][Communication:189]: Message received: Read/Write ID 0xd2 0x22(0x6a2) for property MOTORSPANNUNG (0x06a1) with raw value: 0
[23:15:02][I][Communication:189]: Message received: Read/Write ID 0xd2 0x22(0x6a2) for property SAMMLERISTTEMP (0x000d) with raw value: 217
[23:15:11][I][Communication:189]: Message received: Read/Write ID 0xd2 0x21(0x6a1) for property AUSSENTEMP (0x000c) with raw value: 135
[23:15:11][I][Communication:189]: Message received: Read/Write ID 0xd2 0x21(0x6a1) for property VORLAUFISTTEMP (0x000f) with raw value: 220
[23:15:11][I][Communication:189]: Message received: Read/Write ID 0xd2 0x21(0x6a1) for property SPEICHERISTTEMP (0x000e) with raw value: 473
[23:15:12][I][TIMESYNC:020]: Synchronizing heat pump time to 23.4.2025 23:15
[23:15:12][I][QUEUE:147]: Scheduling request for data from Kessel for JAHR in 10 seconds
[23:15:12][I][QUEUE:147]: Scheduling request for data from Kessel for MONAT in 10 seconds
[23:15:12][I][QUEUE:147]: Scheduling request for data from Kessel for TAG in 10 seconds
[23:15:12][I][QUEUE:147]: Scheduling request for data from Kessel for STUNDE in 10 seconds
[23:15:12][W][canbus:053]: send to standard id=0x6a2 failed!
[23:15:12][I][QUEUE:147]: Scheduling request for data from Kessel for MINUTE in 10 seconds
[23:15:19][I][TIMESYNC:020]: Synchronizing heat pump time to 23.4.2025 23:15
[23:15:19][I][QUEUE:147]: Scheduling request for data from Kessel for JAHR in 10 seconds
[23:15:19][I][QUEUE:147]: Scheduling request for data from Kessel for MONAT in 10 seconds
[23:15:19][I][QUEUE:147]: Scheduling request for data from Kessel for TAG in 10 seconds
[23:15:19][I][QUEUE:147]: Scheduling request for data from Kessel for STUNDE in 10 seconds
[23:15:19][W][canbus:053]: send to standard id=0x6a2 failed!
[23:15:19][I][QUEUE:147]: Scheduling request for data from Kessel for MINUTE in 10 seconds
[23:15:20][I][TIMESYNC:020]: Synchronizing heat pump time to 23.4.2025 23:15
[23:15:20][I][QUEUE:147]: Scheduling request for data from Kessel for JAHR in 10 seconds
[23:15:20][I][QUEUE:147]: Scheduling request for data from Kessel for MONAT in 10 seconds
[23:15:20][I][QUEUE:147]: Scheduling request for data from Kessel for TAG in 10 seconds
[23:15:20][I][QUEUE:147]: Scheduling request for data from Kessel for STUNDE in 10 seconds
[23:15:20][W][canbus:053]: send to standard id=0x6a2 failed!
[23:15:20][I][QUEUE:147]: Scheduling request for data from Kessel for MINUTE in 10 seconds
[23:15:20][I][TIMESYNC:020]: Synchronizing heat pump time to 23.4.2025 23:15
[23:15:20][I][QUEUE:147]: Scheduling request for data from Kessel for JAHR in 10 seconds
[23:15:20][I][QUEUE:147]: Scheduling request for data from Kessel for MONAT in 10 seconds
[23:15:20][I][QUEUE:147]: Scheduling request for data from Kessel for TAG in 10 seconds
[23:15:20][I][QUEUE:147]: Scheduling request for data from Kessel for STUNDE in 10 seconds
[23:15:21][W][canbus:053]: send to standard id=0x6a2 failed!
[23:15:21][I][QUEUE:147]: Scheduling request for data from Kessel for MINUTE in 10 seconds
[23:15:21][I][Communication:189]: Message received: Read/Write ID 0xd2 0x21(0x6a1) for property AUSSENTEMP (0x000c) with raw value: 135
[23:15:21][I][Communication:189]: Message received: Read/Write ID 0xd2 0x21(0x6a1) for property VORLAUFISTTEMP (0x000f) with raw value: 220
[23:15:21][I][Communication:189]: Message received: Read/Write ID 0xd2 0x21(0x6a1) for property SPEICHERISTTEMP (0x000e) with raw value: 475
[23:15:21][I][TIMESYNC:020]: Synchronizing heat pump time to 23.4.2025 23:15
[... cut ...]
[23:15:22][I][Communication:189]: Message received: Read/Write ID 0xd0 0x3c(0x6bc) for property JAHR (0x0124) with raw value: 6400
[23:15:22][I][Communication:189]: Message received: Read/Write ID 0xd0 0x3c(0x6bc) for property MINUTE (0x0126) with raw value: 3840
[23:15:22][I][Communication:189]: Message received: Read/Write ID 0xd0 0x3c(0x6bc) for property FEHLERMELDUNG (0x0001) with raw value: 0
[23:15:22][I][Communication:189]: Message received: Read/Write ID 0xd0 0x3c(0x6bc) for property STUNDE (0x0125) with raw value: 5888
`

mhosse, 2025-06-28:
@kr0ner I had the same issue with sending the STUNDE value. I found out that setting the time is only possible for values less than 13. Either this is a bug in the device or the afternoon hours need to be send differently. I tried several ways but was not able to figure it out.

mhosse, 2025-06-28:
My WPL33 is running with language set to German and time format set to 24h.

kr0ner, 2025-08-20:
> I also have a problem with time synchronization (Stiebel WPC 07 Cool / Tecalor TTF 07 pump). The time format is 24H. [https://pastebin.com/jBXGM56m](url)

What exactly is not working? Which time is reported back from the heatpump via the sensor to HA?

m2teusz, 2025-08-20:
The pump does not synchronize the time after pressing the button. 

[image]

kr0ner, 2025-08-20:
When you navigate to the date/time menu in the heatpump, is the automatic switch to summer time enabled?

m2teusz, 2025-08-20:
Yes, automatic summer/winter time is set (March 25, October 25) 

[image]

kr0ner, 2025-08-20:
I really have no idea what could be the problem. Can you once manually set the time in the heatpump and record the logs while doing so?

My suspicion is that the stored time somehow involves summer time offset.
We could try using utcnow() instead of now() in util.cpp or .h and see how that affects the time

closed 2025-08-22 by kr0ner