kr0ner/OneESP32ToRuleThemAll #181: Bug in thermostat-card with ESPHome 2025.7

ste-ge, 2025-07-19:
With ESPHome 2025.7, my THZ404 no longer recognizes the status in the thermostat card. Heating and hot water are both set to OFF. 

[image]

Everything works fine with ESPHome 2025.6. Heating is set to “Nur Lüftung” and hot water to Auto or heating mode.

[image]

ste-ge, 2025-07-22:
After water heating, the thermostat card is now set to Auto.

[image]

kr0ner, 2025-07-22:
@ste-ge I can confirm this. After a while the status is correctly reflected. I have the suspicion that something changed in the setup order. Still investigating ...

closed 2025-07-23 by kr0ner