kr0ner/OneESP32ToRuleThemAll #186: error on installation process

SimonFuVo, 2025-08-07:
hello i get this error when i want to create the data in ESPHome for uploading to the NodeMCU.
Can you help me ?

INFO ESPHome 2025.7.5
INFO Reading configuration /config/esphome/esp32-poe-technik.yaml...
INFO Detected timezone 'Europe/Berlin'
INFO Detected timezone 'Europe/Berlin'
INFO Detected timezone 'Europe/Berlin'
INFO Detected timezone 'Europe/Berlin'
INFO Detected timezone 'Europe/Berlin'
INFO Detected timezone 'Europe/Berlin'
INFO Detected timezone 'Europe/Berlin'
INFO Detected timezone 'Europe/Berlin'
INFO Detected timezone 'Europe/Berlin'
INFO Generating C++ source...
INFO Compiling app...
Processing esp32-poe-technik (board: esp01_1m; framework: arduino; platform: platformio/espressif8266@4.2.1)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 1MB Flash
Dependency Graph
|-- ESP8266WiFi @ 1.0
|-- ESP8266mDNS @ 1.2
|-- SPI @ 1.0
|-- noise-c @ 0.1.10
Compiling .pioenvs/esp32-poe-technik/src/main.cpp.o
Compiling .pioenvs/esp32-poe-technik/lib67b/ESP8266WiFi/WiFiClientSecureBearSSL.cpp.o
Compiling .pioenvs/esp32-poe-technik/lib67b/ESP8266WiFi/WiFiServer.cpp.o
Compiling .pioenvs/esp32-poe-technik/lib67b/ESP8266WiFi/WiFiServerSecureBearSSL.cpp.o
In file included from src/main.cpp:448:
src/util.h:5:10: fatal error: esp_log.h: No such file or directory

*****************************************************************
* Looking for esp_log.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:esp_log.h"
* Web  > https://registry.platformio.org/search?q=header:esp_log.h
*
*****************************************************************

    5 | #include "esp_log.h"
      |          ^~~~~~~~~~~
compilation terminated.
*** [.pioenvs/esp32-poe-technik/src/main.cpp.o] Error 1

kr0ner, 2025-08-09:
esp01_1m seems way too small for this project 🫣 best is to grab some cheap esp32 from AliExpress or Amazon 😉 that will save a lot of time and frustration.

kr0ner, 2025-08-20:
@SimonFuVo so how did the ESP32-S3 work out?

kr0ner, 2025-09-10:
@SimonFuVo this looks like a CAN setup issue. Connect to the ESP32 via UART to get early logs ... from there it should be visible, if the CAN controller could be setup correctly. If not, check the pin config again. If initialized correctly check the wiring and if that is also correct, use a different ID (0x6a3 or so)