kr0ner/OneESP32ToRuleThemAll #188: wp_temperature.yaml

m2teusz, 2025-08-19:
Hi, in my case (Stiebel WPC 07 Cool / Tecalor TTF 07 pump) there was a problem with reading the room temperature (FEK2).

[image]

The solution was to add:

filters:
- filter_out: 0.0

to the wp_temperature.yaml file

[image]

b.r.
m2teusz

m2teusz, 2025-08-19:
[logs_esp32-poe-technik_run (2).txt](https://github.com/user-attachments/files/21859725/logs_esp32-poe-technik_run.2.txt) 

Attached is the log (debug) immediately after compilation. Sorry it's so large.

kr0ner, 2025-08-20:
So here we get 0
```
17:14:09][D][CAN:250]: Can message received with CANId 0x0301
[17:14:09][I][Communication:187]: Message received: Read/Write ID 0xc1 0x01(0x601) for property RAUMISTTEMP (0x0011) with raw value: 0
```
and the value here ...
```
[17:14:09][D][CAN:250]: Can message received with CANId 0x0301
[17:14:09][I][Communication:187]: Message received: Read/Write ID 0xc0 0x01(0x601) for property RAUMISTTEMP (0x0011) with raw value: 236
[17:14:09][D][sensor:104]: 'RAUMISTTEMP': Sending state 23.60000 °C with 1 decimals of accuracy
```

although I cannot see any 'RAUMISTTEMP': Sending state 0.000000 °C, this read ID got me thinking *0xc1 0x01(0x601)*
Looks like my direct mapping is wrong here
https://github.com/kr0ner/OneESP32ToRuleThemAll/blob/master/src/communication.h#L101-L105
and I have to do it the other way around ... compute the can ID from the read/write/response id and look up the can member.
My code would expect 0x8001 instead of 0xc001 which more or less do the same

kr0ner, 2025-08-20:
duplicate of https://github.com/kr0ner/OneESP32ToRuleThemAll/issues/100